neu-rah/ArduinoMenu #95: fix step/validation for non int's (floats)

sphh, 2017-04-17:
Hi,

Does your last commit "fix step/validation for non int's (floats)" (https://github.com/neu-rah/ArduinoMenu/commit/994034e603a9e0c67cef5f7fe4ec5d83a2581371) work as intended with unsigned variables?

When I wrote the `stepit(int)` function, I admit that I just had integers in mind. I started with a function just like your last commit, but when I tested it with `byte menuField` the variable `thisstep` became unsigned and had the overroll issue, because `stepit((byte) -1)` became 255 or so. Keep in mind, that I might be completely wrong!!

neu-rah, 2017-04-17:
It seems to work very well. Floats were not taken in account initially, but it was just a matter of adding a template parameter... unless someone complains, its ok for me :D thanks

sphh, 2017-04-19:
Well, I (re)tried it with the template version. For me it does not work. I use the following field to change the brightness of a LED:
```C++
    byte brightness;

    altFIELD(cancelField, brightness, "Brightness", "%", 0, 100, 10, 1, go_brightness, anyEvent, noStyle),
```
Whenever I enter the edit mode and try to decrease the value, it jumps to 100 where I get stuck. That was the reason, why I made the the `increment` parameter an integer. But you are right, that my approach makes the `thisstep` variable an integer and this does not work for any `float` variable.

I always thought that the `increment` argument is always `+1` or `-1`, because IFAIK `stepit` is only called in the function `menuField::doNav` either with `stepit(1)` or `stepit(-1)`. Mind you, in this case it could be a `boolean` parameter and we have to handle the sign and the value separately...

sphh, 2017-04-19:
Here is a version which should (hopefully) handle `float` and unsigned values:
```C++
        void stepit(int increment) {
          int sign = increment*(options->invertFieldKeys?-1:1);
          T thisstep = tunning?tune():step();
          dirty=true;
          if (sign > 0) {
            if ((high()-target()) < thisstep) {
              if (style()&wrapStyle) {
                target() = low();
              } else {
                target() = high();
              }
            } else {
              target() += thisstep;
            }
          } else {
            if ((target()-low()) < thisstep) {
              if (style()&wrapStyle) {
                target() = high();
              } else {
                target() = low();
              }
            } else {
              target() -= thisstep;
            }
          }
        }
```

neu-rah, 2017-04-19:
thank you!
you should really be pushing to the repo, those are very nice contributions and authorship is not preserver this way.

closed 2017-05-09 by neu-rah