neu-rah/ArduinoMenu #96: How to redraw the menu?

sphh, 2017-04-24:
I have a implemented a userscreen and whenever I want to enter the menu, I cannot make the menu to draw itself. This is the same issue as #12, #13 and #18 (and possibly others). It looks like the API mentioned in these issues changed and the wiki page disappeared.

Could you please be so kind to give an example? I tried to make all output devices ` dirty:
```C++
liquidCrystalOut outLCD(lcd, tops, pList);
menuOut* const outputs[] MEMMODE = {&outLCD};
outputsList out(outputs, 1);

for(int i=0; i<out.cnt; i++) {
  ((menuOut*)memPtr(out[i]))->drawn = NULL;
}
```
and do a update with
```C++
nav.doOutput();
```
, but I had no luck.

Thank you for your help!

neu-rah, 2017-04-24:
Hi!

yes, API changed on that matter from 2.0

try setting the menu as dirty, ex:

```c++
mainMenu.dirty=true;
nav.doOutput();
```

sphh, 2017-04-24:
Hi,

Thanks. That works!!

closed 2017-04-24 by sphh