neu-rah/ArduinoMenu #97: Issues building under Visual Studio + Visual Micro plugin

flamestring, 2017-04-26:
Hallo

This Menu library is amazing and works fine for me when compiling in Arduino IDE. However, I'm getting errors when building using Visual Studio and Visual Micro plugin. Intellisense is shouting errors on macros like TOGGLE and build is reporting type 'result' as unknown:

ketch1.ino: 12:1: error: 'result' does not name a type
   result doAlert(eventMask e, navNode& nav, prompt &item, Stream &in, menuOut &out)

Build log attached.

The very same file compiles just fine in Arduino IDE. Did anybody had same issue and solved it?

[Visual Micro - menu build log.txt](https://github.com/neu-rah/ArduinoMenu/files/959847/Visual.Micro.-.menu.build.log.txt)

neu-rah, 2017-04-27:
Hi!

I do not have VisualStudio, I'm using linux+ Atom or Arduino IDE

"result" is defined on menuBase.h and by the error messages it seems not that its not being included somehow.

can you isolate the error on a smaller sketch or an example sketch?

also you can try to locate the menuBase.h file and include it

usually files included with "" are local file and files included with <> are searched on library path

the referred file is included (or it should have been) from src/menu.h and src/shadows.h

it might help changing the "" to <> on both, but its just a guess, having the same error on a shorter sketch or an example would help... but the ideal would using VisualStudio to check it...
at this moment i'm downloading it but it might take a while until its up and running (if it would)

please let me know of example compilation results on visualstudio

neu-rah, 2017-04-27:
Hi

I'm running linux and using both Arduino IDE and Atom Platformio...

looking at the log  i would expect a missing include file error message before that error, but there was none.

I've also installed visual studio but this is new to me and i'm having an hard time installing that Visual Micro plugin from a .VSIX file on linux.

does this error also happens when compiling a simple (included) example?

flamestring, 2017-04-27:
Wow, I didn't expect that much effort in helping me that quickly!

The original problem is with I2C example. Now I tried Blink, which is shorter. Intellisense shouted about calling MENU macros (this declaration has no storage class or type specifier), but compiled fine.
I played with libraries inclusions already, but to no success. I'll try later in the evening.

Thanks a lot for your support.

neu-rah, 2017-04-27:
:D

burt still this one might be though because its out of my platform, hope that someone with VisualStudio can drop in and shed some light on this.

the "this declaration has no storage class or type specifier" might be more clear if you try this code that uses no macros (code is part of my dev branch and i'm not sure yet if it compiles on on master branch, will try it later)

```c++
#include <Arduino.h>

#include <menu.h>
#include <menuIO/serialOut.h>

using namespace Menu;

//define "Op 1"
void op1Func();
const char op1Text[] PROGMEM="Op 1";
const promptShadowRaw op1InfoRaw PROGMEM={(callback)op1Func,_noStyle,op1Text,enterEvent,noStyle};
const promptShadow& op1Info=*(promptShadow*)&op1InfoRaw;
//or just this line on non PROGMEM devices like teensy or esp8266 instead of the above three
//promptShadow op1Info("Op 1",(callback)op1Func,enterEvent);
prompt op1(op1Info);

//define "Op 2"
void op2Func();
const char op2Text[] PROGMEM="Op 2";
const promptShadowRaw op2InfoRaw PROGMEM={(callback)op2Func,_noStyle,op2Text,enterEvent,noStyle};
const promptShadow& op2Info=*(promptShadow*)&op2InfoRaw;
//or just this line on non PROGMEM devices like teensy or esp8266 instead of the above three
//promptShadow op2Info("Op 2",(callback)op2Func,enterEvent);
prompt op2(op2Info);

//define the menu
prompt* const menuData[] PROGMEM={&op1,&op2};
//or just prompt* menuData[]={&op1,&op2}; on non avr devices
const char menuTitle[] PROGMEM="Main menu";
const menuNodeShadowRaw menuInfoRaw PROGMEM={
  doNothing,
  (systemStyles)(_menuData|_canNav),
  menuTitle,
  noEvent,
  wrapStyle,
  2,
  menuData
};
const menuNodeShadow& menuInfo=*(menuNodeShadow*)&menuInfoRaw;
//or just this line on non PROGMEM devices like teensy or esp8266 instead of the above three
//menuNodeShadow menuInfo("Main menu",2,menuData,(callback)doNothing,noEvent,wrapStyle);
menuNode mainMenu(menuInfo);

#define MAX_DEPTH 1

//define output device
idx_t serialTops[MAX_DEPTH]={0};
serialOut outSerial(Serial,serialTops);

//define outputs controller
menuOut* const outputs[] MEMMODE={&outSerial};//list of output devices
outputsList out(outputs,1);//outputs list controller

//define navigation root and aux objects
navNode nav_cursors[MAX_DEPTH];//aux objects to control each level of navigation
navRoot nav(mainMenu, nav_cursors, MAX_DEPTH-1, Serial, out);

//implement the menu actions
void op1Func() {Serial.println("Op 1 executed");}
void op2Func() {Serial.println("Op 2 executed");}

#include <Streaming.h>

/////////////////////////////////////////////////////////////////////////////
// arduiin osketch
void setup() {
  Serial.begin(115200);
  while(!Serial);
  Serial.println("Menu initialized without macros.");
  Serial<<"active:";
  Serial<<*(prompt*)&nav.active();
  Serial<<endl;
  Serial<<"maxDepth:"<<nav.maxDepth<<endl;
  Serial<<"Selected:";
  Serial<<(*(prompt*)&nav.selected());
  Serial<<endl;
  Serial<<"active dirty:"<<nav.active().dirty<<endl;
  // Serial<<":"<<nav.()<<endl;
}

void loop() {
  nav.poll();
}
```

you are welcome.

flamestring, 2017-04-27:
Hallo
This code compiles well after I changed #include <Streaming.h> to <Stream> (Streaming.h not found anywhere, Stream.h is part of my Arduino default libs.)

There is intellisense (design time) arror for "doNothing", but it does not prevent successful build:

"Error (active)		a value of type "Menu::result (*)()" cannot be used to initialize an entity of type "Menu::callback"	Sketch2	c:\Users\Radek\Documents\Visual Studio 2015\Projects\Arduino\Sketch2\Sketch2\Sketch2.ino	31"

neu-rah, 2017-04-28:
Streaming.h is https://github.com/scottdky/Streaming but i use it only for debug, a seen here: https://github.com/neu-rah/ArduinoMenu/blob/master/src/menu.h#L24

but if VS defines DEBUG somewhere on some option then it will be required

doNothing might have a warning as the function does nothing... but the "error" is not clear

I would expect this code to have a similar error as before, but this seems not to be the case... so I'm kinda lost here.

here iI've managed to install Visual Studio Code (https://code.visualstudio.com/) and learned enough to know its not the same thing as the "Visual Studio". The Visual Micro plugin seems to be incompatible with VSCode, But I managed to install Arduino plugin (from microsoft) and have it working on linux after some setup instruction that i've found here on github. The blink compiled and uploaded ok, bug again this is not the same as Visual Studio neither I'm using the same plugin.

BTW: what board are you using? I've compiled for Arduino Nano with 328

I will try to have access to a windows machine.. but that might take a while.

flamestring, 2017-04-28:
I'm using Arduino Uno.
Will try with Microsoft plugin as well (IoT, right?).

neu-rah, 2017-04-28:
this one
Arduino for Visual Studio Code
vsciot-vscode.vscode-arduino

flamestring, 2017-04-28:
Partial success: the problem with 'result' as function return type was that Visual Micro automatically creates function prototypes and places them on top of file, that is even before type definition. Fortunately, this feature can be turned off in Project Properties: Generate Prototypes = false.

I was able to advance in building to another error:
`ld.exe: pcint.cpp.o: plugin needed to handle lto object

Error linking for board Arduino/Genuino Uno
 
ccstc1bh.ltrans3.ltrans.o*: In function PCattachInterrupt
pcint.h:107: undefined reference to PCattachInterrupt(unsigned char, mixHandler, unsigned char)
pcint.h:107: undefined reference to PCattachInterrupt(unsigned char, mixHandler, unsigned char)
Debug build failed for project 'Menu_Test_02'
`
Will investigate further.

neu-rah, 2017-04-29:
Thanks for your patience :D
 yup, prototype generation brings weird errors, and i was far from considering that.

about PCattachInterrupt, I suppose from the PCINT lib (https://github.com/neu-rah/PCINT), its only used for the encoder driver, are you using one? if not you can discard it.
And there is also an alternative encoder driver clickEncoderIn.h, that you might consider.

note: menu inputs are just data streams, so if you find another encoder driver of your choice we can turn it into a stream and use it with the menu system.

flamestring, 2017-04-29:
The thing is your code builds and works perfectly when compiled from Arduino IDE. It's just Visual Micro plugin under MS Visual Studio where it has problems, so I want to find the correct setting for it to work there as well. Might be usefull for others as well.

I made my own classes for menu and encoder management and completed my project with it (barndoor stepper control for night sky photography). But your library is much better and more generic (and also more advanced in terms of programming skills), so I'd like to build new version using your library.

I'll take a look into that PCattachInterrupt problem tomorrow.

Thanks a lot!

neu-rah, 2017-04-29:
PCINT library is in fact nice because it does not skips encoders steps on a busy MCU, however some of the problems may come from it trying to be generic in terms of platforms, I'm sure we can code it for a more specific platforms it there is need to.
If the changes made to it might help others and you are willing to share them, I would be most pleased to accept a PR.

Speaking of new versions, at this moment I'm exploring new degrees of abstraction using haskell (my fav. language), Just feel free to talk about menus or whatever code you feel deserves discussing.

There is a chat room here https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link

flamestring, 2017-05-01:
Hallo again, it took me hell of a time, but I managed to solve the problem and make your library compile and build under Visual Studio with Visual Micro plugin. 

To solve the issue of **undefined reference to PCattachInterrupt**, it is necessary to move it's definition and related declarations from pcint.cpp to pcint.h. It's this code in particular:

```
	static int PCintMode[3][8];
	static HANDLER_TYPE PCintFunc[3][8];
	static bool PCintLast[3][8];//?we can use only 3 bytes... but will use more processing power calculating masks

	void PCattachInterrupt(uint8_t pin, HANDLER_TYPE userFunc, uint8_t mode) {
		volatile uint8_t *pcmask = digitalPinToPCMSK(pin);
		if (!pcmask) return;//runtime checking if pin has PCINT, i would prefer a compile time check
		uint8_t bit = digitalPinToPCMSKbit(pin);
		uint8_t mask = 1 << bit;
		uint8_t pcicrBit = digitalPinToPCICRbit(pin);
		PCintMode[pcicrBit][bit] = mode;
		PCintFunc[pcicrBit][bit] = userFunc;
		//initialize last status flags
		PCintLast[pcicrBit][bit] = (*portInputRegister(digitalPinToPort(pin)))&digitalPinToBitMask(pin);
		// set the mask
		*pcmask |= mask;
		// enable the interrupt
		PCICR |= (1 << pcicrBit);
	}

```

The second needed change is to turn off Genrate Prototypes in project properties (Solution explorer pane): Generate prototypes = false.

These 2 modifications allowed me to compile your example I2C under VS/VM. I did not yet test the functionality on Arduino itself, will try later today. Hope this helps to other users.

flamestring, 2017-05-01:
P.S. Here are more details that directed me to the fix: [http://stackoverflow.com/questions/10632251/undefined-reference-to-template-function](url)

flamestring, 2017-05-01:
I was happy too early. The library compiles and uploads, but single click of encoder does not work. When the same compiled and uploaded from Arduino IDE, it works well. I'm lost :)

neu-rah, 2017-05-01:
At least it compiles.. that is a great advancement, I get a bit frustrated here because all that stuff is on windows and I've no windows machine around.
Now the focus is on PCINT library, it has a single simple example and a mapping example that i think it does not apply here unless we want to check if the new building environment still maps the same as the original arduino. The mapping sketch will print out a C++ line that should in principle be exactly equal to the one already in the lib.
The normal example just blinks a led with a push of a button, its a good diagnostic for PCINT.

Thanks for your your efforts, I can see that you've been battling a lot.

flamestring, 2017-05-03:
I was unable to make your libs work well under VS/VM. I installed PlatformIO IDE + Clang now and all works like a breeze. I will stick with this environment for a while for sure :)

neu-rah, 2017-05-04:
HI :)
I use platformio also, just didn't suggest it because its a big change and because there might be some others using your former platform, but we will deal with it when they appear.

I was also considering to add another encoder driver to the library, found this one very interesting https://github.com/zacsketches/Encoder

at first read it also uses templates and meta-programming, so it might result in the same problem, but it would be valuable as an alternative encoder driver using interrupts.

glad, you found a suitable platform, and unless you want to add something please close the issue

thanks, and sorry I was not able to help you more.

flamestring, 2017-05-06:
For my current project Platformio IDE is fine. But VS has more advanced features, mainly in terms of code navigation and refactoring. In the future I might try one more time to investigate how to use it for projects created in Platformio/Clang.

Untill now, I was using this encoder library, you might be interested. It has implemented doubleclick and speed rotation: [https://github.com/0xPIT/encoder](https://github.com/0xPIT/encoder).

Anyway, thanks a lot for your library, it's beautifully written! :)

closed 2017-05-06 by flamestring

neu-rah, 2017-05-06:
Let me know when the time for VS comes, if it comes.

It happens that https://github.com/christophepersoz/encoder is based on 0xPIT encoder library, so that library is already adapted by  https://github.com/christophepersoz

About the lib, I tried my best, thank for your appreciation. Still there is space for improvements (many).
And you can still add us a star :D