neu-rah/ArduinoMenu #98: OP with confirmation

sphh, 2017-05-16:
I tried to write an `altOP` which ask me to confirm the activation of this menu item. The idea is that whenever I "press" `enterCmd` on a menu item the menu will respond with a "`Press [ENT] to confirm!". If I "press" `enterCmd` again, the menu will execute the function, if I "press" `escCmd`, it will return to the menu item.

Here is the code I have so far:
```C++
void go_shutdown() {
  // Shut down the system
}

// Use 'navTarget', because it has 'doNav()'
class confirmPrompt:public navTarget {
protected:
  bool confirm;
public:
  confirmPrompt(const promptShadow& shadow): navTarget(shadow), confirm(false) {}
  idx_t printTo(navRoot &root, bool sel, menuOut& out, idx_t idx, idx_t len) override {
    if (confirm) {
      return out.printRaw("Press [ENT] to confirm!", len);
    } else {
      return navTarget::printTo(root, sel, out, idx, len);
    }
  }
  void doNav(navNode& nav, navCmd cmd) {
    if (cmd.cmd == enterCmd && !confirm) {
      confirm = true;
      dirty = true;
      return;
    } else if (cmd.cmd == escCmd && confirm) {
      confirm = false;
      dirty = true;
      return;
    }
    confirm = false;
    navTarget::doNav(nav, cmd);
  }
};

MENU(system_menu, "System", doNothing, noEvent, wrapStyle,
  altOP(confirmPrompt, "Shutdown!", go_shutdown, enterEvent)'
  );

// ...
```

My idea was that `doNav` is called and if it has to display the confirmation message (or return to the prompt of the menu item) it makes the menu dirty. Unfortunately this does not work as I hoped: It simply executes `go_shutdown` without asking for confirmation. This just shows that the internals of the menu is is way above my head! Can you be so kind to point me in the right direction (again)?

Thanks!

neu-rah, 2017-05-17:
you need only to derive an alternative prompt if you wish it to be non-blocking (keep looping).
if you do not mind of blocking, then the op handler can do all sort of thing prior to return.

still need non-blocking?

neu-rah, 2017-05-18:
P.S. am I over-simplifying it? please let me know if you still want the derived class.. because it can be done.

sphh, 2017-05-19:
Hi,

Thanks for your answer. To do the confirmation in the op handler is certainly a possibility I have not thought of. But in my case the Arduino should be able to react on other events, so a non-blocking version would be preferable, if it can be done. Can it be done?

neu-rah, 2017-08-25:
what kind of confirmation? can it be a submenu with yes/no? being yes the action and no the quit (exit menu)

sphh, 2017-08-25:
The idea was that

1. I navigate to the menu item by the usual means, let's say `Shutdown`.
2. I press `enterCmd`, but instead of activating the shutdown immediately, it asks me `Do you really want to shutdown?`
3. I press `enterCmd` to activate the shutdown process or `escCmd` to cancel.

I guess this functionality can be done by showing a submenu with the two items `No` and `Sure, shut down`, but this requires an additional step: scrolling down one item. Thinking about it, a submenu with just `Confirm to shutdown` would do the job, because the user can always backstep with `escCmd`. Is that right?

neu-rah, 2017-08-25:
even on a 2 options menu user can escape provided it has an escape button or he is using a console, however users with and encoder (single button) can not.

i think that a menu must have a minimum of 2 options

we can use menu title to do the question.

if we were going to implement some sort of dialog it would involve 2 options [Ok][Cancel] to honor the encoder input case.

sphh, 2017-08-26:
Using a 2 options menu (YES/NO) is perfectly feasible and does actually do what I want to do. There is only one (cosmetic) glitch: If I name the submenu "Shutdown" it shows up in the menu structure as "Shutdown" and when entering it it's still "Shutdown" whereas I would like to have it "Shutdown?" with the two menu items "Yes" and "No". If I name it "Shutdown?" it would show in the menu structure as "Shutdown?". Is there a way to have a submenu with two titles, one for showing as menu item and one for showing as title when the user is in it? (I hope that you can understand what I mean!!)

neu-rah, 2017-08-28:
Added an example (v3.1.11) of exit confirmation with menu lock where sub-menu prompt can differ from sub-menu title by customizing the submenu draw, much like a custom prompt. The only difference is on the menu definition macro that uses altMENU macro to wire the customized class.

sphh, 2017-08-29:
Thanks. That's great.

Now, if I have not only one menu item, which needs confirmation, I have to write an `class confirmExit` for each menu item, e.g. for `Reboot`, `Shutdown`, ... That is certainly possible, but is there a way to code the two prompts in a more generic way, such as

    otheraltMENU(confirmExit,subMenu,"Exit?","Exit",doNothing,anyEvent,noStyle
      ,OP("Yes",systemExit,enterEvent)
      ,EXIT("Cancel")
    );

neu-rah, 2017-08-29:
yes, thats a problem with that solution. Adding extra arguments to the class construction would be easy by that is not accepted by the macros. So, we can either:

1 - Use a menu without macros (see nomacros example) and make the text an constructor argument
2 - Add a special confirmation object to the library with the respective macros (kind of a pain)

I've not explored option 2, but I'm convinced its exequible.

are you willing to try the no macros case?

sphh, 2017-08-29:
Now I start to appreciate your macros even more ;-)

I don't mind writing an `altMenu` for every menu item needing confirmation. I just wondered, if a more generic route would save some bytes?

neu-rah, 2017-08-29:
i've fight for every RAM byte.. and maybe too much. As you noticed I did not care much about flash memory. That made sense back then for AVR's but now with teensy, esp and arm's ram is not so important.

I do HATE those macros and i would rather pass without them. But the static menu construction and the way C allocates pointer arrays (except for char*) requiring decomposition of all parts and separate allocation to aggregate all in the end, that along with the FLASH memory macros and the fact that C++ (unlike C) does not support this memory overlap used by AVR's (FLASH and RAM share the same address) converged all to that great mess.

Anyway i was thinking on the alt macros limitations, I've latter introduced them to allow usage of alternative objects. The problem was that i've used the same parameters (with only one additional initial parameter the class name), this makes the alt macros very limiting because the object construction is done inside the macro and therefor with the same arguments.

I think that i can make an extra set of macros (lets say objPROMPT, objMENU,...) that can use external allocated objects instead of allocating themselves the obj). That would allow more flexibility with parameters and object construction and would allow casual mix of more customizable objects with  the macro allocated ones... and therefor not having to go for a situation of all or nothing posed by the current macros thing.

Gota explore this... and make an essay to check if there is no corner case that would inviabilize this.

After all this journey i've started to appreciate more C instead of C++, so I might some day come up with a more clean thing, backed up by all obstacles i've found on C++ and by my recent haskell experience. But that is a far day for sure.

neu-rah, 2017-10-20:
V4 dynamic example might be of interest for you in this area
https://github.com/neu-rah/ArduinoMenu/blob/master/examples/dynamic/dynamic/dynamic.ino
its available for non AVR's by default and for AVR's under conditional compile.

you are welcome to join us at https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link

closed 2017-10-20 by neu-rah