neu-rah/ArduinoMenu #99: Quadencoder not decoding

Redferne, 2017-06-24:
Thanks for this library, it really helps cutting down development time :smile_cat: 

However, I cannot get the encoder to work properly. I can see that both interrupts are firing in PCINT when the encoder is turned, but the position only moves erratically in one direction (down). I'm using the Moteino Mega (1284p). Please advice.

neu-rah, 2017-06-26:
Hi!
mega has been a trouble with pcint maps because there are many mappings around. The PCINT library has a sketch to map your particular pcint schema.
Also there are 2 encoder libraries available for the menu, you can try an alternative.
Let me know if you find any difficulties.

Redferne, 2017-06-26:
I had to change the decoding into:
```
    template<uint8_t pinA,uint8_t pinB>
    void encoderIn<pinA,pinB>::encoderInIrq(class encoderIn<pinA,pinB> *e) {
      int8_t abNew = (digitalRead(pinA) << 1) | digitalRead(pinB);
      enum { upMask = 0x66, downMask = 0x99 };
      int8_t newTick = abNew ^ e->oldState;
      if (newTick == 1 || newTick == 2)
      {
        if (upMask & (1 << (2 * e->oldState + abNew / 2)))
          e->pos++;
        else e->pos--;
      }
      e->oldState = abNew;
      e->irq = abNew;
    }
```
Now it works beautifully :smile_cat:

closed 2017-06-26 by Redferne

neu-rah, 2017-06-27:
can we have a PR?

Redferne, 2017-06-27:
Of course! Let me clean it up and make sure it works first :+1:

aerodolphin, 2017-07-15:
When I use the encoder the selection of the menu sometimes jumps more than one item. Is there a way to change the sensitivity of the encoder?

neu-rah, 2017-07-15:
yes...

```c++
  // Encoder /////////////////////////////////////
  #define encA 5
  #define encB 6
  //this encoder has a button here
  #define encBtn 4

  encoderIn<encA,encB> encoder;//simple quad encoder driver
  #define ENC_SENSIVITY 4
  encoderInStream<encA,encB> encStream(encoder,ENC_SENSIVITY);// simple quad encoder fake Stream
```