neu-rah/ArduinoMenu #101: Not compiling on Arduino Dua

aerodolphin, 2017-07-10:
`Arduino: 1.8.3 (Windows 7), Board: "Arduino Due (Programming Port)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\aerodolphin\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\aerodolphin\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\aerodolphin\Documents\Arduino\libraries -fqbn=arduino:sam:arduino_due_x_dbg -ide-version=10803 -build-path C:\Users\AERODO~1\AppData\Local\Temp\arduino_build_4860 -warnings=all -build-cache C:\Users\AERODO~1\AppData\Local\Temp\arduino_cache_931787 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Users\aerodolphin\Desktop\Slider Project 2017\TFT\tft menu\tft2\tft2.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\aerodolphin\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\aerodolphin\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\aerodolphin\Documents\Arduino\libraries -fqbn=arduino:sam:arduino_due_x_dbg -ide-version=10803 -build-path C:\Users\AERODO~1\AppData\Local\Temp\arduino_build_4860 -warnings=all -build-cache C:\Users\AERODO~1\AppData\Local\Temp\arduino_cache_931787 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Users\aerodolphin\Desktop\Slider Project 2017\TFT\tft menu\tft2\tft2.ino
Using board 'arduino_due_x_dbg' from platform in folder: C:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11
Using core 'arduino' from platform in folder: C:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11
Detecting libraries used...
"C:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10803 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/libsam" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/CMSIS/Include/" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/Device/ATMEL/" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x" "C:\Users\AERODO~1\AppData\Local\Temp\arduino_build_4860\sketch\tft2.ino.cpp" -o "nul"
"C:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10803 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/libsam" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/CMSIS/Include/" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/Device/ATMEL/" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\libraries\SPI\src" "C:\Users\AERODO~1\AppData\Local\Temp\arduino_build_4860\sketch\tft2.ino.cpp" -o "nul"
"C:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10803 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/libsam" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/CMSIS/Include/" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/Device/ATMEL/" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\libraries\SPI\src" "-IC:\Users\aerodolphin\Documents\Arduino\libraries\Adafruit_GFX_Library" "C:\Users\AERODO~1\AppData\Local\Temp\arduino_build_4860\sketch\tft2.ino.cpp" -o "nul"
"C:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10803 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/libsam" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/CMSIS/Include/" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/Device/ATMEL/" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\libraries\SPI\src" "-IC:\Users\aerodolphin\Documents\Arduino\libraries\Adafruit_GFX_Library" "-IC:\Users\aerodolphin\Documents\Arduino\libraries\Adafruit_ST7735_Library" "C:\Users\AERODO~1\AppData\Local\Temp\arduino_build_4860\sketch\tft2.ino.cpp" -o "nul"
"C:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10803 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/libsam" "-IC:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/CMSIS/Include/" "-IC:\Users\aerodolphin\AppData\Loc [... cut ...]
"C:\Users\aerodolphin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -MMD -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10803 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\aerodolphin\AppData\Local\Ar

aerodolphin, 2017-07-10:
I'm getting a lot of errors. Anyone have a clue of what is wrong?

neu-rah, 2017-07-12:
what was the installation method? noting that some libs are in
C:\Users\aerodolphin\Documents\Arduino\libraries\...
while the menu is at C:\Program Files (x86)\Arduino\libraries\menu (legacy)
legacy?
please install from the arduino IDE library manager and be sure to clean-up any manual installation.

aerodolphin, 2017-07-12:
I installed "menu" and "pcint" libraries manualy because they don't appear in the IDE library manager.

aerodolphin, 2017-07-12:
I saw that the correct way is to install the libraries in zip files. I will try it. I think this will solve the issues.

neu-rah, 2017-07-12:
I've installed ide 1.8.3 and both libs are listed on lib manager, please use the lib manager. key search (pcint, menu) to install both. The libs are conformant with https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#15-library-format-rev-21

and are both registered with library manager, so they should be listed, if not please let me know

aerodolphin, 2017-07-12:
Thank you. I installed the zip files of the libraries and now I can compile.

But nothing happens when I run the code on Due. Only a white screen. When I run the example "graphictest" the screen works.

I connected all iaw that example and made the change here in the code of tft:

```
#define TFT_CS   10
#define TFT_DC   8
#define TFT_RST  9
Adafruit_ST7735 gfx(TFT_CS, TFT_DC, TFT_RST);
```

I connected also the TFT to the SPI SCK and MOSI. VCC to 3.3V, GND to GND, LED- to GND and LED + to 3.3V.

Any idea?

neu-rah, 2017-07-13:
too vague... 
can you print to screen (ie on setup)?
are you calling poll on loop?

aerodolphin, 2017-07-13:
I will make a schematic of the connections.

aerodolphin, 2017-07-13:
The encoder will not work on version 3?

aerodolphin, 2017-07-13:
Yes, I have the next line in the loop:

`  nav.poll();//this device only draws when needed`

aerodolphin, 2017-07-13:
The example "tft" was unchanged. I only changed the part of the code related to the pins of the CS, D/C and RESET. They were initially to analog and I changed to digital pins.

aerodolphin, 2017-07-13:
Is there anyway to test what is wrong?

aerodolphin, 2017-07-13:
Maybe can be related to this issue:
[Issue #81](https://github.com/neu-rah/ArduinoMenu/issues/81)

neu-rah, 2017-07-13:
> The example "tft" was unchanged. I only changed the part of the code related to the pins of the CS, D/C and RESET. They were initially to analog and I changed to digital pins.

you can check if its really this change by changing back... and if it is then you can try to select other pins.

PCINT only works on ports B, C, D (because it was ported from Atmega328p)

if there is any problem with due + encoder please let me know... preferably on another issue to keep things easy for future reference

aerodolphin, 2017-07-13:
Yesterday I tried also on the analog ports (unchanged). Same issue, a blankscreen.

Regarding to the PCINT. I don't understand what are the ports B, C and D.

About encoder, I will have to test when I saw the menu on the screen.

aerodolphin, 2017-07-13:
The ports B, C and D are the microprocessor ports? What are the corresponding ones in the arduino Due?

```
Board | Digital Pins Usable For Interrupts
Uno, Nano, Mini, other 328-based | 2, 3
Mega, Mega2560, MegaADK | 2, 3, 18, 19, 20, 21
Micro, Leonardo, other 32u4-based | 0, 1, 2, 3, 7
Due | all digital pins
101 | all digital pins
```

neu-rah, 2017-07-13:
Not using interrupts but using pin change interrupts, PinChangeInterrupts are available for many pins.
http://www.robgray.com/temp/Due-pinout-WEB.png
pale yellow (legend: port/pin) tells the needed Port

aerodolphin, 2017-07-13:
Ok. Din't know about PinChangeInterrupts.

Only the pins of the encoder have to be connected to those ports (B, C or D)? Or also the ports for the CS, D/C and RST?

Pins from the encoders are ok (D2 PIN - B.25, D3 - C.28 and D4, B.25).
If CS, D/C and RST has to be connected to those ports. If we use A0, A1 and A2 they will be connected to the A ports.

So, I think the problem is in another place

neu-rah, 2017-07-13:
port like [B.17] or [C.12] where the letter indicates the port and number. on ATMega328 there are 8 pins per port max. the same limit applies to due even if ports have more than 8 pins (because of the demultiplexing tables).... the PCINT library example can print a pin map.

encoders can use an alternative driver (clickEncoder) that uses timers instead of PCINT (have to install click encoder and timer1 libs)

aerodolphin, 2017-07-14:
`void setup() {
  options=&myOptions;
  pinMode(LEDPIN,OUTPUT);
  Serial.begin(115200);
  while(!Serial);
  Serial.println("menu 3.0 test");
  Serial.flush();
  nav.idleTask=idle;//point a function to be used when menu is suspended
  mainMenu[1].enabled=disabledStatus;
  //outGfx.usePreview=true;//reserve one panel for preview?
  //nav.showTitle=false;//show menu title?

  //pinMode(encBtn, INPUT_PULLUP);
  encButton.begin();
  encoder.begin();

  SPI.begin();
  gfx.initR(INITR_BLACKTAB);
  gfx.setRotation(3);
  gfx.setTextSize(textScale);//test scalling
  gfx.setTextWrap(false);
  gfx.fillScreen(ST7735_BLACK);
  gfx.setTextColor(ST7735_RED,ST7735_BLACK);
  gfx.println("Menu 3.x test on GFX");
  delay(1000);
}`

In void setup() I removed the two lines of the encoder:
```
  //encButton.begin();
  //encoder.begin();
```

And now I can see the menu on display.
So, the problem is in the encoder part.

I changed the pins for the encoder to:

```
// rotary encoder pins
#define encA    36 // C.4
#define encB    34 // C.2
#define encBtn  30 // D.9
```

Now the ports are in the (B, C or D) and the numbers are less than 8. It's correct, right?

I will try to make some more tests with the encoder.

aerodolphin, 2017-07-14:
I tried the next example and I can't make it work.

```
#include <pcint.h>

#define led 13
#define btn 30

void setled() {
  digitalWrite(led,digitalRead(btn));
}

void setup() {
  pinMode(led,OUTPUT);
  pinMode(btn,INPUT_PULLUP);
  PCattachInterrupt<btn>(setled,CHANGE);
  setled();//initial led status
}

void loop() {}

```

aerodolphin, 2017-07-14:
I tried several pins and the LED didn't turn on. I also put a pull up resistor and nothing.

Any ideas why this PCINT simple example don't work with Arduino Due?

neu-rah, 2017-07-14:
on atmega328 the led pin is also used by spi... but the looking at due pinout it does not seem so... i will try with my due.

aerodolphin, 2017-07-14:
Ok. Thanks for the help. I will wait your test.

neu-rah, 2017-07-14:
i can confirm it does not work on due, even with no other devices attached or using some other btn pins like 11, this looks like a lib bug... investigating

aerodolphin, 2017-07-14:
ok

neu-rah, 2017-07-14:
fixed PCINT bug on due, resulting on call of uninitialized function call
https://github.com/neu-rah/PCINT/commit/6b4df143d659fe394223785481fb149de6fbb703#diff-af8a6206c4eff4ac03c1b8ad758fe971R105
 was able to run the btnled example

no restriction for ports or pins (sorry those mentioned restrictions were for atmega1284 and 2560)
for due pcint is implemented as normal interrupts because due supports them on any pin.

unfortunately there was a bug, missing an initialization... hope its fixed now.

thanks for you patience and help by reporting it :D

aerodolphin, 2017-07-15:
Thanks. It's working now.

closed 2017-07-15 by neu-rah