neu-rah/ArduinoMenu #103: Sub-Menu problem

aerodolphin, 2017-07-18:
When I run the code tft on Arduino Due I get for every submenu allways the same submenu. Please check the video bellow:

[https://youtu.be/VNVPiauAuiM](url)

Another problem that appear also in the video is the use of an encoder. It jumps more than one menu item. I tried to change the sensibility without success. My encoder has a detent. Don't know if the problem is that.

neu-rah, 2017-07-20:
you can rule out encoder problems by disconnecting it and using serial navigation (use chars +-*/ enter)

aerodolphin, 2017-07-20:
I will buy a new encoder.
Do you think the problem of the submenu is linked to the encoder issue?

neu-rah, 2017-07-20:
its strange, i'm uploading a simpler (serialio) to due to do basic checking, if anything related to encoder it might be lack of debouncing... but still hard to believe it would make all that alone, so do not throw away the encoder :D

neu-rah, 2017-07-20:
serialio example seems work well on due (need click to activate menu as the example starts with suspended menu) but the options respond as expected, so it seems that menu system (core) is ok for due... next we need to check for encoder problems... you can use the serial console to do navigation with keys, that would test the tft example (i've not loaded tft because it would require wiring it here by removing my tft from its current assembly) but it should be trivial to test.

by the way, are you initializing encoder by calling its begin function? as in
https://github.com/neu-rah/ArduinoMenu/blob/master/examples/adafruitGfx/tft/tft/tft.ino#L218

neu-rah, 2017-07-20:
problem with tft example, move encoder to other pins rather than 2 & 3, because serial is using them, I'm updating examples...

neu-rah, 2017-07-20:
examples updated and submitted to arduino ide lib manager... should auto-update soon on next start

aerodolphin, 2017-07-20:
Yes. I'm initializing the encoder with its begin function. I will try to use a trigger schmitt with the encoder to filter the signal. I try to remove debounce problems with resistors and capacitors.

I think I'm using one of that pins for the encoder. Will try on another.

It's normal the submenu appear in the right? Should not be better to appear in another screen?

neu-rah, 2017-07-20:
submenu preview and open on next menu panel.

panels are defined here: https://github.com/neu-rah/ArduinoMenu/blob/master/examples/adafruitGfx/tft/tft/tft.ino#L186

,{0,0,14,8},{14,0,14,8})

minimal 1 panel (full or partial device covering)

submenu will open on sub panel while available

previews will show on sub-panel if any to spare and config defined

neu-rah, 2017-07-29:
solved?

aerodolphin, 2017-07-31:
Not yet. Will try with the new encoder when arrive.

neu-rah, 2017-12-26:
please reopen, create new or ask on gitter if you still need help with this.

closed 2017-12-26 by neu-rah