neu-rah/ArduinoMenu #106: Save 38 bytes program storage space

sphh, 2017-08-27:
Since the operator `<<` is only used in debugging mode (at least I think so), I managed to save 38 bytes program storage space by applying the patch which conditionally comments out all references to `<<`. I hope it does not break anything (it works for me).

```diff
--- ArduinoMenu_library/src/menu.cpp
+++ ArduinoMenu-3.1.10/src/menu.cpp
@@ -86,12 +86,10 @@
   }
 }
 
-#ifdef DEBUG
 menuOut& menuOut::operator<<(const prompt& p) {
   print_P(*this,p.getText());
   return *this;
 }
-#endif
 
 void menuOut::clearChanged(navNode &nav) {
   //if (nav.target->dirty) Serial<<"clear dirty "<<*(prompt*)nav.target<<" sz:"<<nav.sz()<<endl;
--- ArduinoMenu_library/src/menu.h
+++ ArduinoMenu-3.1.10/src/menu.h
@@ -40,10 +40,8 @@
     // Streams
     //////////////////////////////////////////////////////////////////////////
     //template<typename T> inline Print& operator<<(Print& o, T t) {o.print(t);return o;}
-#ifdef DEBUG
     Print& operator<<(Print& o,prompt const &p);
     inline String& operator<<(String &s,prompt &p);
-#endif
     //template<typename T> HardwareSerial& operator<<(HardwareSerial& o,T t) {o.print(t);return o;}
     //template<typename T> inline menuOut& operator<<(menuOut& o,const T x) {return o.operator<<(x);}
 
@@ -342,12 +340,10 @@
         inline idx_t maxY(idx_t i=0) const {return panels[i].h;}
         inline idx_t& top(navNode& nav) const;
         idx_t printRaw(const char* at,idx_t len);
-#ifdef DEBUG
         virtual menuOut& operator<<(prompt const &p);
         #ifdef ESP8266
         template<typename T> menuOut& operator<<(T o) {(*(Print*)this)<<(o);return *this;}
         #endif
-#endif
         virtual menuOut& fill(
           int x1, int y1, int x2, int y2,char ch=' ',
           colorDefs color=bgColor,
@@ -748,9 +744,7 @@
 
     idx_t& menuOut::top(navNode& nav) const {return tops[nav.root->level];}
 
-#ifdef DEBUG
     inline String& operator<<(String &s,prompt &p) {return s+=p.getText();}
+  }//namespace Menu
+
 #endif
-  }//namespace Menu
-
-#endif
--- ArduinoMenu_library/src/menuBase.cpp
+++ ArduinoMenu-3.1.10/src/menuBase.cpp
@@ -29,12 +29,10 @@
   return noCmd;
 }*/
 
-#ifdef DEBUG
 Print& Menu::operator<<(Print& o,prompt const &p) {
   print_P(o,p.getText());
   return o;
 }
-#endif
 
 const navCodesDef Menu::defaultNavCodes={
   {noCmd,(char)0xff},
```

neu-rah, 2017-08-27:
how do I apply this to preserve your authorship?

sphh, 2017-08-28:
It looks like I have to learn how to do pull requests, but this time, don't worry about the authorship! I'm happy to help.

neu-rah, 2017-08-28:
pull request are very easy to do. I would be happy to have your collaboration authored... so think again, it cant hurt and we can always move back and forth if its not ok the first time :D

sphh, 2017-08-28:
Do you know of a good tutorial?

neu-rah, 2017-08-28:
nope, just the wiki (and it needs some polishing)

sphh, 2017-08-29:
Yes, that wiki needs certainly some polishing (the first time I looked at it I just gave up). But with the help of two other tutorials I managed to make my first pull request #107 ;-)

neu-rah, 2017-08-29:
merged :D
thanks for this

closed 2017-08-29 by neu-rah