neu-rah/ArduinoMenu #108: use with spi ram?

papillon2k, 2017-08-30:
Hi, nice library. I am using it with arduino pro-mini and 128x64 OLED (u8g2) but I am running out of RAM (and flash too). I have a couple of 23K256 spi Sram chips that I would like to use, would it be posible to modify or extend the library so the data that is now being saved on progmem, be saved on the RAM ic instead? I have taken a look at the library code but im pretty noob at arduino/c/c++ code and dont even know where or what should I look for...  in menu.h I see around line 305 it uses memcpy_P() and I guess this is where it gets the data from progmem to be display on the oled, but that's all I have been able to find lol... maybe you could consider implementing it as a new feature? 
thanks for a great library anyway

neu-rah, 2017-08-30:
Hi! Thanks for using the library :D
memcpy_P is rarely used because we always try to use progmem directly (even for print).
To be honest right now I have no clue what is needed to work with another type of memory... but i will investigate. I was just that kind of thing that I've never thought of.
Main memory hacks are on shadows.h just in case you become curious.. however its a file that even I avoid touching ;)

neu-rah, 2017-09-01:
I think you need a more powerful microcontroller like teensy, STM, Due or ESP8266. Some can be about the same price as a pro-mini (especially esp's) and they use the arduino environment.
Swapping menu definitions would be more suitable for a dynamic menu (that already works on RAM), this one is a static menu, flash stored, and flash can not be written at runtime.
Redirecting all the machinery that currently reads from flash is a profound change and it would not for soon.

Still, if you are using similar submenus we can workout to reuse a common sub.menu and maybe use it to serve multiple cases. This would be a smaller change.

papillon2k, 2017-09-01:
Hi Neu-rah, thanks for the replies.  
I have just tested my spi ram chip on arduino (just a simple test to write/read something) with code from https://ucexperiment.wordpress.com/2013/02/23/interfacing-microchip-23k256-32kb-sram-spi-memory-to-the-arduino/ and it seems to work ok with the functions defined there to read and write from spi ram. 
Also I had located a block of code in your macros.h where you define some macros that wrap calls to pmg_read_byte, pgm_red_ptr etc (those macros are called in shadow.h)  I was hoping it could be a matter of replacing those macros so they call the functions that reads from the spi ram instead of the pgm_ functions... but for your comments i guess its much more complicated. ouch...

anyway, I am pasting here my sketch, as you can see its not a complex menu... the sketch just creates the menu and reads a button ladder to navigate it. As it is, arduino ide says its 29426bytes (95%) of program space, and 1399 bytes (68%) of RAM. are these numbers coherent with the sketch, or am I doing something wrong that is taking up more progmem or ram that needed?
(note -  I *think* am using u8g2 in page-buffer mode, or at least that was my intention)
(note2 - the file FWBpinUnico.h manages the buttons, but its fairly simple and I guess irrelevant in terms of flash or ram usage)
(note3 - dont know why but the code formatting seems broken when pasted here...)

`#include <Arduino.h>
#include <menu.h>
#include <menuIO/u8g2Out.h>
using namespace Menu;

// BUTTONS -------------------------------------------------------------
#include "FWBpinUnico.h"
#define PIN_BUTTONS A0
FWB botonera;
#define BTN_NONE 0
#define BTN_RIGHT 1
#define BTN_UP 2
#define BTN_DOWN 3
#define BTN_LEFT 4
#define BTN_SELECT 5
//button push types
#define CLICK 1
#define DBLCLICK 2
#define LONGPRESS 3
#define VLONGPRESS 4

uint8_t lastButton = 0;
uint8_t lastPushType = 0;

// global vars
#define RIGHT 0
#define LEFT  1
uint16_t duracion = 60;
uint8_t longitud = 100;
bool direccion = RIGHT;     
uint8_t delayProceso = 5;
boolean enProceso = false;
uint8_t posicionHome = -1;

//instancia el u8g2 (para el oled)
U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, SCL, SDA, U8X8_PIN_NONE);

result zZz() {Serial.println("zZz");return proceed;}

result actionStartProceso() {
    Serial.println(F("starting process..."));
    return quit;
}

TOGGLE(direccion,setDireccion,"Dir. ",doNothing,noEvent,noStyle
  ,VALUE("Dcha",RIGHT,doNothing,noEvent)
  ,VALUE("Izqda",LEFT,doNothing,noEvent)
);
TOGGLE(delayProceso,setDelay,"Del. ",doNothing,noEvent,noStyle
  ,VALUE("0s",0,doNothing,noEvent)
  ,VALUE("3s",3,doNothing,noEvent)
  ,VALUE("5s",5,doNothing,noEvent)
);
MENU(menuEmpezar,"Empezar",doNothing,noEvent,noStyle
  ,OP("Si", actionStartProceso,enterEvent)
  ,EXIT("Cancel")
);
MENU(menuOpciones,"Opciones",doNothing,noEvent,noStyle
  ,FIELD(duracion,"Dur. ","s",0,15000,60,1,doNothing,noEvent,wrapStyle)
  ,FIELD(longitud,"Lon. ","%",0,100,10,1,doNothing,noEvent,wrapStyle)
  ,SUBMENU(setDireccion)
  ,SUBMENU(setDelay)
  ,EXIT("<<<")
);
MENU(mainMenu,"Main menu",zZz,noEvent,noStyle
  ,SUBMENU(menuEmpezar)
  ,SUBMENU(menuOpciones)
  ,EXIT("<<<")
);

//each color is in the format:
//  {{disabled normal,disabled selected},{enabled normal,enabled selected, enabled editing}}
// this is a monochromatic color table
const colorDef<uint8_t> colors[] MEMMODE={
    {{0,0},{0,1,1}},//bgColor
    {{1,1},{1,0,0}},//fgColor
    {{1,0},{1,0,0}},//valColor
    {{1,1},{1,0,0}},//unitColor
    {{0,1},{0,0,1}},//cursorColor
    {{0,0},{1,1,1}},//titleColor
};
#define fontName u8g2_font_6x10_tf
#define fontX 6
#define fontY 10    //
#define offsetX 8
#define offsetY 4
#define MAX_DEPTH 2

MENU_OUTPUTS(out,MAX_DEPTH
    //,U8G2_OUT(u8g2,colors,fontX,fontY,offsetX,offsetY,{0,0,64/fontX,48/fontY})
    ,U8G2_OUT(u8g2,colors,fontX,fontY,offsetX,offsetY,{0,0,128/fontX,64/fontY})
    //,SERIAL_OUT(Serial)
    ,NONE
);
NAVROOT(nav,mainMenu,MAX_DEPTH,Serial,out);

//when menu is suspended
result idleMenu(menuOut& o,idleEvent e) {
    o.clear();
    switch(e) {
        case idleStart:o.println("suspending menu!"); break;
        case idling:o.println("suspended..."); break;
        case idleEnd:o.println("resuming menu."); break;
    }
    return proceed;
}

void setup() {
    //init buttons
    botonera.Configure(PIN_BOTONERA);
    botonera.OnClick = OnClick;
    
    u8g2.begin();
    u8g2.setFont(fontName);
    //flashscreen
    u8g2.firstPage();
    do {
        u8g2.drawStr(30,30,"CamSlider");
        u8g2.drawFrame(20,10,100,40);        
    } while(u8g2.nextPage());
    delay(2000);
    
    //menus
    nav.idleTask=idleMenu;  //point a function to be used when menu is suspended
}

void loop() {
    botonera.CheckBP();	    //check if button pressed

    if(enProceso) {
        //process ongoing
        Serial.println("en proceso");
    }

    switch(lastButton) {
        case BTN_DOWN: nav.doNav(upCmd); break;
        case BTN_UP: nav.doNav(downCmd); break;
        case BTN_LEFT: nav.doNav(escCmd); break;
        case BTN_RIGHT: nav.doNav(enterCmd); break;
        case BTN_SELECT: nav.doNav(enterCmd); break;
    }
    lastButton=0;
    lastPushType=0;
        
    if (nav.changed(0)) {//only draw if menu changed for gfx device
        //change checking leaves more time for other tasks
        u8g2.firstPage();
        do {
            nav.doOutput();
        } while(u8g2.nextPage());
    }
}

//functions for buttons
int findButtonPressed(int adcIn) {
    if (adcIn > 900) return BTN_NONE;
    else if (adcIn < 50) return BTN_LEFT; 
    else if (adcIn < 200) return BTN_UP;
    else if (adcIn < 400) return BTN_RIGHT;
    else if (adcIn < 600) return BTN_DOWN;
    else if (adcIn < 1000) return BTN_SELECT;
    else return BTN_NONE; 
}

void OnClick(uint8_t pin, uint8_t tipo, int val) {
    lastButton = determinaBoton(val);
    lastPushType = tipo;
}`

neu-rah, 2017-09-02:
yup, you sure have an unusual high flash usage, i've been able to fit much larger menus into an atmega328, so i've done a code split to try to understand each part.

first i've done some abbreviation of code because i have no idea where that buttons lib is, so i've just cutted out the buttons part, therefore the numbers below are just a relative comparison (not absolute values)

then i've removed menu nav and output device, cutting your nice code driven from buttons.

after that i've focused my attention into the menu structure with the following results

-no menu
9906 30.2%

-just menu def
21694 66.2% +12k <-menu data + menu code of parts used

- setDirecction cutted
20370 62.2%

- setDelay cutted
18754 57.2%

-menuEmpezar cutted
18524 56.5%

-menuOpciones
13526 41.3%

41% was the best after cutting all submenus...
then I've focused on fields

------------------------
-just menu def
21694 66.2%

- no FIELD's
16834 51.4%

-all fields uint16_t
20120 61.4%

we can cut on some flash usage just by cutting on data types

because those fields are templates, it means that every time you use the same field or toggle or whatever refers to your data, c++ compiles a version for each data type.

meaning: using an int8_t FIELD and an int16_t field would result on expanding the code twice, one for each type

therefor you can save flash by using both field of int16_t even if it takes a bit more of ram.

the same for all data referring structures like TOGGLE, SELECT, etc... having the same datatype for each struct will void multiple template instantiation.

same for bool's, you can use int's if there is also the same structure using an int.

try it with

```c++
uint16_t duracion = 60;
uint16_t longitud = 100;
uint8_t direccion = RIGHT;
uint8_t delayProceso = 5;
```
 this should get you a bit more of flash

neu-rah, 2017-09-02:
and please let me know if this helps you because you caught my attention to this problem (thanks) and FLASH memory issues. I was focused on RAM usage minimization and never worried about flash.. but now that you've mentioned it, I've started doing some tests by spliting the menuField code into 2 classes, a base that holds generic data and stuff and leaving everything that is template related to the menuField and I've got some flash usage improvements, the same has to be done to all templates... this "fight" will result in flash usage reduction.

because your problem is not with the menu structure itself, its more a problem of code generated by menu structure usage, the external flash would not help too much, because we can not store code on external flash.

neu-rah, 2017-09-03:
dev branch contains a version with template expansion optimization that can cut almost 1.5Kb of flash with your original field types.

papillon2k, 2017-09-03:
great! just by changing those 4 lines it went from:

 Sketch uses 27708 bytes (90%) of program storage space. 
 Global variables use 1305 bytes (63%) of dynamic memory

to:

 Sketch uses 24904 bytes (81%) of program storage space. 
 Global variables use 1226 bytes (59%) of dynamic memory

then i switched to using the dev version (still using the modified field types): 

 Sketch uses 25258 bytes (82%) of program storage space. 
 Global variables use 1219 bytes (59%) of dynamic memory

then I changed the variable declarations to the original ones, while still using the dev version:
 Sketch uses 26558 bytes (86%) of program storage space.
 Global variables use 1292 bytes (63%) of dynamic memory

So it seems I have now room for the rest of my project (adding accelStepper library to control a motor). 
thank you! great library!

neu-rah, 2017-09-03:
and now I can make the lib even better :D
I was expecting C++ to optimize that template expansion (because they could) and I didn't even bother to check.

Thanks a lot for bringing this question... it end up having nothing to do with SPI ram hehe...

I've learned something new (I like this!)

Gracias!

It would be great to signal that to others by giving us a star :))

closed 2017-09-07 by neu-rah

neu-rah, 2017-10-23:
Hi again!
Sorry to add to this closed issue. Version 4.x of the menu inherited all the optimizations referred above and was subject of a revision, saving some extra 1.4K of flash and having the ability to allocate the menu in RAM. Its is however a different version of the menu but eventually it will result on some extra savings for your case.
If it happens that you try the new version I would be pleased to have feedback on the flash memory consumption in comparison with V3.x.
I can say that the revision applied to v4 is also possible on v3 if you are interested.
Thanks
RA