neu-rah/ArduinoMenu #109: Preserve memory space using unique char* declaration

christophepersoz, 2017-09-12:
Hi Rui,

I'm trying for an another project to use the v3.1.2 (I'm not sure that the bug is related to this version). I have the same char* string that appears in different places (menu choices and main display). I would like to declare the strings only once and use them in the menu and during the display. It seems that there is a problem when I'm doing this because the compilation fails on the string in the menu declaration.
Here what I would like to do (I'm still working on Teensy so there is some variations like PROGMEM and so)

[code]
// before setup and loop
typedef struct {
    uint8_t fxValue;
    const char fxName[11];
    const uint8_t minSpd;   
} _cEffect; 

_cEffect _FX[5] =
{
    { 0, "SOFT NOISE", 16 },
    { 1, "SPARKLES", 32 },
    { 2, "WAVE", 28 },
    { 3, "STILL", 0 }, 
    { 4, "STROBE", 75}
};

// if I declare the in the sub-menu this way it does work, but it occupied twice the size of the strings because I will use _FX[x].fxName on the main display when I'm outside the menu.
    CHOOSE( _fx, _mFX, "ANIMATION ", doNothing, noEvent, noStyle,
        VALUE( "SOFT NOISE", 0, doNothing, noEvent ),
        VALUE( "SPARKLES ", 1, doNothing, noEvent ),
        VALUE( "WAVE", 2, doNothing, noEvent ),
        VALUE( "STILL ", 1, doNothing, noEvent ),
        VALUE( "STROBE", 2, doNothing, noEvent )
    );

// if I declare the CHOOSE the way, the compilation fail due to the following errors 
// error: initializer fails to determine size of 'valueLabel2'
// error: array must be initialized with a brace-enclosed initializer

CHOOSE( effect, _mFX, "ANIMATION ", doNothing, noEvent, noStyle,
        VALUE( _FX[0].fxName, 0, doNothing, noEvent ),
        VALUE( "SPARKLES ", 1, doNothing, noEvent ),
        VALUE( "WAVE", 2, doNothing, noEvent ),
        VALUE( "STILL ", 1, doNothing, noEvent ),
        VALUE( "STROBE", 2, doNothing, noEvent )
    );
[/code]

In the rest of the program the use of _FX[x].fxName does not create any issue. It seems that the builder does not interpret the struct before creating the menu...

Is there a workaround to avoid this ?

Thanks a lot!
Christophe

neu-rah, 2017-09-12:
I've changed _FX to const PROGMEM to load on nano. The menu expects data to be PROGMEM.
however this might not be what you want, specially if you intend to change some data dynamically.

```c++
const _cEffect _FX[5] PROGMEM =
{
    { 0, "SOFT NOISE", 16 },
    { 1, "SPARKLES", 32 },
    { 2, "WAVE", 28 },
    { 3, "STILL", 0 },
    { 4, "STROBE", 75}
};
``` 
this is the expansion of your problematic macro on an AVR with progmem:

```c++
const char _mFX2_text[] PROGMEM="ANIMATION ";
const char anim_valueLabel0[] PROGMEM=_FX[0].fxName;
const PROGMEM Menu::menuValueShadowRaw<typeof(effect)> anim_choice0ShadowRaw={ (Menu::callback)doNothing, Menu::_noStyle, anim_valueLabel0, noEvent, Menu::noStyle, 0 };
const Menu::menuValueShadow<typeof(effect)>& anim_choice0Shadow= *(menuValueShadow<typeof(effect)>*)&anim_choice0ShadowRaw;
Menu::menuValue<typeof(effect)> anim_menuValue0(anim_choice0Shadow);
const char anim_valueLabel1[] PROGMEM="SPARKLES ";
const PROGMEM Menu::menuValueShadowRaw<typeof(effect)> anim_choice1ShadowRaw={ (Menu::callback)doNothing, Menu::_noStyle, anim_valueLabel1, noEvent, Menu::noStyle, 1 };
const Menu::menuValueShadow<typeof(effect)>& anim_choice1Shadow= *(menuValueShadow<typeof(effect)>*)&anim_choice1ShadowRaw;
Menu::menuValue<typeof(effect)> anim_menuValue1(anim_choice1Shadow);
const char anim_valueLabel2[] PROGMEM="WAVE";
const PROGMEM Menu::menuValueShadowRaw<typeof(effect)> anim_choice2ShadowRaw={ (Menu::callback)doNothing, Menu::_noStyle, anim_valueLabel2, noEvent, Menu::noStyle, 2 };
const Menu::menuValueShadow<typeof(effect)>& anim_choice2Shadow= *(menuValueShadow<typeof(effect)>*)&anim_choice2ShadowRaw;
Menu::menuValue<typeof(effect)> anim_menuValue2(anim_choice2Shadow);
const char anim_valueLabel3[] PROGMEM="STILL ";
const PROGMEM Menu::menuValueShadowRaw<typeof(effect)> anim_choice3ShadowRaw={ (Menu::callback)doNothing, Menu::_noStyle, anim_valueLabel3, noEvent, Menu::noStyle, 3 };
const Menu::menuValueShadow<typeof(effect)>& anim_choice3Shadow= *(menuValueShadow<typeof(effect)>*)&anim_choice3ShadowRaw;
Menu::menuValue<typeof(effect)> anim_menuValue3(anim_choice3Shadow);
const char anim_valueLabel4[] PROGMEM="STROBE";
const PROGMEM Menu::menuValueShadowRaw<typeof(effect)> anim_choice4ShadowRaw={ (Menu::callback)doNothing, Menu::_noStyle, anim_valueLabel4, noEvent, Menu::noStyle, 4 };
const Menu::menuValueShadow<typeof(effect)>& anim_choice4Shadow= *(menuValueShadow<typeof(effect)>*)&anim_choice4ShadowRaw;
Menu::menuValue<typeof(effect)> anim_menuValue4(anim_choice4Shadow);
Menu::prompt* const _mFX2_data[] PROGMEM={ &anim_menuValue0, &anim_menuValue1, &anim_menuValue2, &anim_menuValue3, &anim_menuValue4, };
const PROGMEM Menu::menuVariantShadowRaw<typeof(effect)> _mFX2ShadowRaw={ (Menu::callback)doNothing, ((systemStyles)(Menu::_menuData|Menu::_canNav|Menu::_isVariant)), _mFX2_text, noEvent, noStyle, sizeof(_mFX2_data)/sizeof(prompt*), _mFX2_data, &effect };
const PROGMEM Menu::menuVariantShadow<typeof(effect)>& _mFX2_Shadow=*(menuVariantShadow<typeof(effect)>*)&_mFX2ShadowRaw;
Menu::choose<typeof(effect)> _mFX2 (_mFX2_Shadow);
```

this is the problematic line (right on top):
```c++
const char anim_valueLabel0[] PROGMEM=_FX[0].fxName;
```
giving this errors:

`initializer fails to determine size of 'anim_valueLabel0'`

`array must be initialized with a brace-enclosed initializer`

the line is an allocation line and expects a braced initializer to determine the length, however the data is already initialized.

changing to:

```c++
// const char anim_valueLabel0[] PROGMEM=_FX[0].fxName;
const PROGMEM Menu::menuValueShadowRaw<typeof(effect)> anim_choice0ShadowRaw={ (Menu::callback)doNothing, Menu::_noStyle, _FX[0].fxName, noEvent, Menu::noStyle, 0 };
```

compiles ok, however i'm not sure if this is what you want, it looks like you want to generate choice values from a structure and reuse them, right? what about the right-most values? are they dynamic?

if you are happy with the above i can translate it to due or teensy.

neu-rah, 2017-09-12:
this version seems to compile on teensy and runs ok on due

```c++
_cEffect _FX[5] =
{
    { 0, "SOFT NOISE", 16 },
    { 1, "SPARKLES", 32 },
    { 2, "WAVE", 28 },
    { 3, "STILL", 0 },
    { 4, "STROBE", 75}
};

char _mFX2_text[]="ANIMATION ";
// char anim_valueLabel0[]=_FX[0].fxName;
Menu::menuValueShadowRaw<typeof(effect)> anim_choice0ShadowRaw={ (Menu::callback)doNothing, Menu::_noStyle, _FX[0].fxName, noEvent, Menu::noStyle, 0 };
Menu::menuValueShadow<typeof(effect)>& anim_choice0Shadow= *(menuValueShadow<typeof(effect)>*)&anim_choice0ShadowRaw;
Menu::menuValue<typeof(effect)> anim_menuValue0(anim_choice0Shadow);
char anim_valueLabel1[]="SPARKLES ";Menu::menuValueShadowRaw<typeof(effect)> anim_choice1ShadowRaw={ (Menu::callback)doNothing, Menu::_noStyle, anim_valueLabel1, noEvent, Menu::noStyle, 1 };
Menu::menuValueShadow<typeof(effect)>& anim_choice1Shadow= *(menuValueShadow<typeof(effect)>*)&anim_choice1ShadowRaw;
Menu::menuValue<typeof(effect)> anim_menuValue1(anim_choice1Shadow);
char anim_valueLabel2[]="WAVE";Menu::menuValueShadowRaw<typeof(effect)> anim_choice2ShadowRaw={ (Menu::callback)doNothing, Menu::_noStyle, anim_valueLabel2, noEvent, Menu::noStyle, 2 };
Menu::menuValueShadow<typeof(effect)>& anim_choice2Shadow= *(menuValueShadow<typeof(effect)>*)&anim_choice2ShadowRaw;
Menu::menuValue<typeof(effect)> anim_menuValue2(anim_choice2Shadow);
char anim_valueLabel3[]="STILL ";Menu::menuValueShadowRaw<typeof(effect)> anim_choice3ShadowRaw={ (Menu::callback)doNothing, Menu::_noStyle, anim_valueLabel3, noEvent, Menu::noStyle, 3 };
Menu::menuValueShadow<typeof(effect)>& anim_choice3Shadow= *(menuValueShadow<typeof(effect)>*)&anim_choice3ShadowRaw;
Menu::menuValue<typeof(effect)> anim_menuValue3(anim_choice3Shadow);
char anim_valueLabel4[]="STROBE";Menu::menuValueShadowRaw<typeof(effect)> anim_choice4ShadowRaw={ (Menu::callback)doNothing, Menu::_noStyle, anim_valueLabel4, noEvent, Menu::noStyle, 4 };
Menu::menuValueShadow<typeof(effect)>& anim_choice4Shadow= *(menuValueShadow<typeof(effect)>*)&anim_choice4ShadowRaw;
Menu::menuValue<typeof(effect)> anim_menuValue4(anim_choice4Shadow);
Menu::prompt* _mFX2_data[]={ &anim_menuValue0, &anim_menuValue1, &anim_menuValue2, &anim_menuValue3, &anim_menuValue4, };Menu::menuVariantShadowRaw<typeof(effect)> _mFX2ShadowRaw={ (Menu::callback)doNothing, ((systemStyles)(Menu::_menuData|Menu::_canNav|Menu::_isVariant)), _mFX2_text, noEvent, noStyle, sizeof(_mFX2_data)/sizeof(prompt*), _mFX2_data, &effect };Menu::menuVariantShadow<typeof(effect)>& _mFX2_Shadow=*(menuVariantShadow<typeof(effect)>*)&_mFX2ShadowRaw;
Menu::choose<typeof(effect)> _mFX2 (_mFX2_Shadow);

```

christophepersoz, 2017-09-12:
Oh! What I do want is to avoid a numéros char* declaration for static string more than once to save SRAM and also it seems ridiculous to declare several times the same string in different vars just you want to use them in different places. 
Can we have this ?
Thanks a lot

neu-rah, 2017-09-12:
yes its a macro limitation
macro is doing allocation. because allocating all that PROGMEM is tedious and generates too many names and dependent chains, most of them to be used only once... not this case thou.

on dev branch I have a macro that allows mixing macro allocated and user allocated menu items, probably will come up with something similar for values.

christophepersoz, 2017-09-12:
Hmmm... but I can override the MACRO limitation by using a non Macro mode ?... look gonna be though but...

(A part of that, is there a way to avoid the double editing on FIELD items ? coarse and fine value into one edit value ?)

Thanks

neu-rah, 2017-09-12:
yup, the non macro version is the code pasted above

```c++
//static part, can go to PROGMEM
Menu::menuValueShadowRaw<uint8_t> anim_choice0ShadowRaw={ (Menu::callback)doNothing, Menu::_noStyle, **_FX[0].fxName**, noEvent, Menu::noStyle, 0 };

//cast to C++ object (c++ cant create objects on progmem, just structs. no constructors can be called => agregation initialization)
Menu::menuValueShadow<uint8_t>& anim_choice0Shadow= *(menuValueShadow<uint8_t>*)&anim_choice0ShadowRaw;

//c++ object refering  static data and with possible dynamic data
Menu::menuValue<typeof(effect)> anim_menuValue0(anim_choice0Shadow);
```

all this to allocate one 'choose' value, and then some code at end to group them all

```c++
//options data
Menu::prompt* _mFX2_data[]={ &anim_menuValue0,...};

//static struct (progmem)
Menu::menuVariantShadowRaw<typeof(effect)> _mFX2ShadowRaw={ (Menu::callback)doNothing, ((systemStyles)(Menu::_menuData|Menu::_canNav|Menu::_isVariant)), _mFX2_text, noEvent, noStyle, sizeof(_mFX2_data)/sizeof(prompt*), _mFX2_data, &effect };

//cast static struct to c++ object (an hack)
Menu::menuVariantShadow<typeof(effect)>& _mFX2_Shadow=*(menuVariantShadow<typeof(effect)>*)&_mFX2ShadowRaw;

//full c++ object with dynamic data and virtual functions
Menu::choose<typeof(effect)> _mFX2 (_mFX2_Shadow);
```

to be linked to menu as

`,SUBMENU(_mFX2)`

neu-rah, 2017-09-20:
Hi! :D

too make things easy, dev branch contains a VALUEOBJ macro that allows link in user defined values mixed with VALUE macro allocated ones.

```c++
typedef struct {
    uint8_t fxValue;
    const char fxName[11];
    const uint8_t minSpd;
} _c_fx;

constMEM _c_fx _FX[5] MEMMODE=
{
    { 0, "SOFT NOISE", 16 },
    { 1, "SPARKLES", 32 },
    { 2, "WAVE", 28 },
    { 3, "STILL", 0 },
    { 4, "STROBE", 75}
};

uint8_t _fx=1;

//define a value
constMEM menuValueShadowRaw<typeof(_fx)> fx0ShadowRaw MEMMODE={ (callback)doNothing, _noStyle, _FX[0].fxName, noEvent, noStyle, 0 };
menuValueShadow<typeof(_fx)>& fx0Shadow= *(menuValueShadow<typeof(_fx)>*)&fx0ShadowRaw;
menuValue<typeof(_fx)> fxValue0(fx0Shadow);

// mix user allocated value with macro allocated ones
CHOOSE( _fx, _mFX1, "ANIMATION", doNothing, noEvent, noStyle,
    //VALUE( "SOFT NOISE", 0, doNothing, noEvent ),
    VALUEOBJ(fxValue0),
    VALUE( "SPARKLES", 1, doNothing, noEvent ),
    VALUE( "WAVE", 2, doNothing, noEvent ),
    VALUE( "STILL", 3, doNothing, noEvent ),
    VALUE( "STROBE", 4, doNothing, noEvent )
);
```

neu-rah, 2017-09-20:
still you can make a macro for your allocation if you going to use many, also note the **constMEM** and **MEMMODE** on the _FX array declaration.

this macros will expand to nothing on non AVR MCUs, so they should not affect you, on AVR's they will put the structure on flash memory because that's where the menu expects the strings to be... that would affect the access to the structure when using AVR's

christophepersoz, 2017-09-26:
I'm going to have a look on this, I will keep you in touch. I'm a bit busy to finish some other parts of the dev before ;)
Thanks a lot

neu-rah, 2017-09-26:
no rush :D

neu-rah, 2017-10-20:
Version 4 dynamic menu should make this easy
see https://github.com/neu-rah/ArduinoMenu/blob/master/examples/dynamic/dynamic/dynamic.ino
or reopen if its the case
thank you :D

closed 2017-10-20 by neu-rah