neu-rah/ArduinoMenu #110: new encoder function

paulosilver, 2017-09-19:
Hello, is there a way to include is encoder function to be used as input?

`
// Interrupt on A changing state
void doEncoderA() {

// debounce
  if ( rotating ) delayMicroseconds(250);  // wait a little until the bouncing is done
  
  // Test transition, did things really change?
    if ( bitRead(PIND, 2) != A_set ) { // debounce once more
      A_set = !A_set;
  
  // adjust counter + if A leads B
    if ( A_set && !B_set )
      encoderPos += 1;
      rotating = false;  // no more debouncing until loop() hits again
    }
}

// Interrupt on B changing state, same as A above
void doEncoderB() {

  if ( rotating ) delayMicroseconds(250);
    if ( bitRead(PIND, 3) != B_set ) {
      B_set = !B_set;
  //  adjust counter - 1 if B leads A
    if ( B_set && !A_set )
      encoderPos -= 1;
      rotating = false;
    }
}
`

neu-rah, 2017-09-19:
its possible, but there are 2 encoder drivers already on the lib (one normal and other with accel.), this one does software debounce. This kind of things tend to slow down the menu because some delay has to be observed before considering a change, and therefore augmenting the menu poll time-slice.
Another consideration, the code seems adequate for polling but not for interrupts, because even waiting to skip bouncings the interrupt will still be recalling the functions for every bounce, right?
As alternative we can change the _encoderIn_ driver to ignore changes that occur in less than a delta time of the previous change. This last one will return between bounces shrinking the menu time-slices.

neu-rah, 2017-09-19:
p.s. also a small capacitor between encoder pins can do an hardware debounce.

neu-rah, 2017-10-20:
anyone wants to modify current encoder driver to include software debounce?

lucky62, 2018-05-26:
note that any delays in interrupt routines are unwanted.

neu-rah, 2018-05-26:
enhancement, as I would like to support more input drivers.. not necessarily the exact code but some encoder with software debounce maybe, I'm open to sugestions...

neu-rah, 2018-05-26:
@paulosilver its now possible to use code ctrl, so virtually any input method can be indirectly used as long as your code translates that to menu commands

lucky62, 2018-05-27:
>  its now possible to use code ctrl, so virtually any input method can be indirectly used as long as your code translates that to menu commands
@neu-rah, it is perfect solution.

Regarding the encoder - I think some library with debouncing already exists in the world - so only some kind of wrapper must be created.

neu-rah, 2018-05-27:
yes, menu input/output drivers are usually that, just wrappers. Historically they would be streams and would emit characters, now they wrap an existing driver and emit menu commands.

neu-rah, 2018-05-29:
because we have codeCtrl menu commands and because this concrete code uses `delay`, and also because it is open for too long... I'm closing this.
Feel free to join us at https://gitter.im/ArduinoMenu/Lobby fir further discussion

closed 2018-05-29 by neu-rah