neu-rah/ArduinoMenu #111: stepper motors

paulosilver, 2017-09-19:
When you say "it's a non-blocking" that means if using stepper functions and libs like AccelS, the stepper motor does not pause for doing another thing like updating the LCD?
Thank you

neu-rah, 2017-09-19:
Yes, as you noticed. When I say non-blocking its a polled system that will return as soon as possible if there is nothing to be done. Its enough for running non time critical task in parallel.
For time critical task I would recommend a timer interrupt to keep stepper running. That would be the case in all systems, even if some lighter menus may make it less noticeable. Because the output has to be updated and input has to be read, that will for sure interfere with stepper timing.

neu-rah, 2017-09-19:
I had a similar problem when updating some car monitor displays. In my assembly the sensors read and display is independent from the menu, and i was only using the menu for setups like calibrations, set alarm levels, etc...
Still the menu polling was causing a slight flicking of the 3 7-segment led display that has to be actively maintained by the mcu.
my solution was to start with a suspended menu, displays can flick when i'm doing the setup and using the menu but then I exit and every thing runs smooth.
The menu has a suspended state for that, but it can also be done externally by not calling the menu polling functions on suspended state.
This might not be your case but if it is then steppers can roll free if you don't need to actively change parameters when steppers are working.
I'm also open to developing AVR timer drivers because I have some experience with it. (sem compromissos :D )

neu-rah, 2017-10-03:
did the "suspended menu" strategy solved it for you?

neu-rah, 2017-10-20:
please reopen if anything new, thanks.

closed 2017-10-20 by neu-rah