neu-rah/ArduinoMenu #112: Time and Date fields

martinmarcos, 2017-09-26:
Hi. I've seen that it's possible to extend menu object classes to customize it way the look and behave and I was wondering if it was possible to extend the menuField class to get editable time and date fields. I was thinking that these fields could be hooked up to 3 variables, in the time case to hours, minutes and seconds and in the date case to day, month and year. They could be displayed with separating characters, for example **00:00:00** for time and **00/00/0000** for date. To edit them, when you select the field you would start editing one of the members(day for example), selecting again would let you edit the second member(month), selecting a third time would let you edit the last member(year) and selecting a fourth time would finish editing. Data validation would have to be done through the callback function to check errors like entering day 31 for June or something like that.

Do you think this would be possible? Do you think it would be to difficult to implement?

Regards.
Martin

neu-rah, 2017-09-26:
Hi, this is getting heavy for AVR's :D
but of course the idea crossed my mind, still is very vague how to implement it, but if we can implement it in a way that it wont be included unless its used (and probably we can) the it worth giving it a try.

several starting lines:

1) Just use existing text fields they can restrict the characters entered on each position... no date/time validation and / position would be editable but only with one valid character '/' <-- its done no need for extra code (version 4.x on dev branch)
2) implement multiple numeric fields... need to draw them inline, maybe 'asPad' style can help (experimental on v4)... there would be some validation problems because you can enter day 31 and then edit the month making it invalid... still the validator could correct it.
3) a single custom look field, click to edit year, then click again to edit day/month as a single field (menu supports 2 levels of edit)

this is too vague and maybe inaccurate, eventually API might have to be extended to support some corner case... if so then its not zero cost. It wont be a problem for mighty non-avrs, put the lib size for AVR's is getting critical

neu-rah, 2017-10-12:
Version 4.0 released supporting user format fields by composition of multiple fields in a single line.

non AVR example: https://github.com/neu-rah/ArduinoMenu/blob/master/examples/dynamic/dynamic/dynamic.ino

neu-rah, 2017-10-20:
Custom formats are achieved by composing multiple option in a submenu and use them in-place.
This way full menu power is available to define custom formats, including events to validate the inputs during of after edit.
you are welcome to join us at https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link

neu-rah, 2017-10-28:
please reopen if needed

closed 2017-10-28 by neu-rah