neu-rah/ArduinoMenu #114: Menu EXIT

manaioussema, 2017-09-29:
Hello
first i would thank you for this great library.
i have a question about main menu EXIT, what happen when the main menu exit ? where goes the code ? and thanks.

neu-rah, 2017-09-29:
Hi!
Thanks for using the library :D

if you do not specify nothing an internal dummy function is used

you can point your own function to be used instead, by changing your navRoot object idleTask member to point to it.

```c++
nav.idleTask=myFunct;//can do this on setup or when apropriate
```

you can force menu to a sleep state by doing:

```c++
nav.sleeOn(myFunct);//next pool will be redirect to myFunct
```
if used on setup then the menu will 'start' on suspended state (click/select to leave)

myFunct is of type `idleFunct`, that is:

```c++
result (*idleFunc)(menuOut& o,idleEvent);
```

your function will be called for each of your output devices (passed it at first parameter) and with the following events:

```c++
enum idleEvent {idleStart,idling,idleEnd};
```

so, it will be called at least 3 times, on entry with `idleStart`, on pool with `idling`, on exit with idleEnd

on some devices (like serial) the pool=>idling is only called once, on others (requiring redraw) it will be called for each poll.

if I misspelled something or if something is not working please let me know.

manaioussema, 2017-09-29:
i have tried what you said but i didn't get what i wanted. In fact, what i want to do is to make a monitor screen that displays some informations (that are updated like time) when i'm not using the menu and when i need it it shows back, i have also read the other issues and the different ideas to do this purpose but the idle state seemed to be a good solution. If you have other idea, i would be thankful to hear from you. Best regards.

neu-rah, 2017-09-29:
on the above answer i've already spotted an error and there might be others (edited the answer), because I was away from my computer and rushing when I wrote that. Sorry for the inconvenience.

The idle function is not good for updatable info on serial monitor. For that I would use something like:

```c++
void loop() {
  nav.poll();
  if (nav.sleepTask) {
    //do your update here...
  }
}
```

**info**
using the idle function will only be called 3 times (entry,idle,function) on some devices (like serial monitor).

manaioussema, 2017-09-30:
thanks for your replies, i already tried to set a variable in my idle function and then test it in the loop to be able to make the update but the `if (nav.sleepTask)` is better and cleaner. Another little problem i'm facing is the encoder glitching, what i'm mean is when i'm turning it the selection jumps many items upward and downward with a strange behavior , i've tried to change the sensitivity and steps but with no good results (i'm using the clickEncoderIn.h). Thank you again

neu-rah, 2017-09-30:
there is also encoderIn.h
clickEncoderIn has accel. while encoderIn has not, don't know if its related. Also encoderIn will require a call to begin in setup to initialize the pins and an extra library PCINT https://github.com/neu-rah/PCINT.
encoderIn uses pin change interrupts.
Similar issues might also be causes by bouncing. you can debounce the encoder with small capacitors between the encoder pins.
you're welcome :D

neu-rah, 2017-10-03:
solved?

manaioussema, 2017-10-03:
yes by changing the encoder and i'm tweaking the sensitivity parameter. Thank you.

closed 2017-10-03 by neu-rah