neu-rah/ArduinoMenu #115: Re-entering Menu after exit do not redraw

christophepersoz, 2017-09-29:
Hi Rui,

I noticed a bug yesterday and I do not find a workaround to solve it. 
I have a LED system running on the loop and I use a rotary encoder to manage everything. When I'm doing a long press, the Menu is called and I can change the options. 
The bug is, when I'm exiting the menu, back to the main loop, if I'm making a long press again to call back the menu a second time (it happens any time you are calling the menu), the display remain blank. I have to press the switch one more time to make it appear. Have you notice that before ?
The problem is only on u8g2 output, it seems that the serial output is not affected by it. 
I tried to force a redraw or a poll but it is not working. 

Do you need the code or does it ring a bell to you ?

PS: your menu l'un is really amazing. I love it!

Thanks a lot

neu-rah, 2017-09-29:
Hi!

thank for having the patience of testing this things, this kind of contribution is essential :D 

(maybe) it rings a bell... the driver might be sending an escape for the long press and therefor not entering the menu, however the display (u8g) should not be cleared.

I think I can emulate it here, will let you know if code needed.

btw, is it master or dev branch?

christophepersoz, 2017-09-29:
Great ! It is the master one ;) I haven't find the time yet to test the dev.

Thanks !

neu-rah, 2017-09-29:
I don't recommend testing the dev yet... Its quite unstable, it might be nice to appreciate a feature but would be a pain to test because I'm still constantly updating it

neu-rah, 2017-10-03:
I was unable to reproduce this on U8G2, have any code?

exit or long press exits the menu and prints my "suspended" message, then long press wont get back to menu, but the menu does not clear "suspended" is still on display, have to normal click to enter menu again.

This is the normal behavior, but maybe esc should get us back to menu. Is that what you refering?
and it makes sense.

from your text i have the feeling that even first normal click on suspended menu does not move back to the menu, right?

christophepersoz, 2017-10-03:
Hi Rui, 

Sorry for the answer's delay.
Here a video that is showing the issue. You will see that on the second enter menu after a long presse the blank screen. I also need to rotate the encoder then click to make it appears, because if I only click nothing is happening and the screen remain blank.

[MENU_Exit_issue.mov.zip](https://github.com/neu-rah/ArduinoMenu/files/1353218/MENU_Exit_issue.mov.zip)

Please note that I do not use any exit() event, is that related to ?

-----

I also discovered an another bug related to setFont with the U8G2 output. if seems that the
#define fontName xxxx 
is never call inside u8g2Out.h which can cause some visuals issues if you are using an another font just before calling the menu.
In u8g2Out.h I tried to change this,
```
) :gfxOut(resX,resY,t,p,menuOut::redraw),gfx(gfx),colors(c) {
					**gfx.setFont(fontName);**
					gfx.setFontPosBottom(); // U8Glib font positioning
					this->offsetX=offsetX;
					this->offsetY=offsetY;
			}
```
but it create an error on build, due to fontName missing, even if it does exist. replacing it by a real u8g2 font name, does not change anything. At this time it seems that the only workaround is to initiate setFont() in the display loop() like this :

```
 if (inMENU)
    {
	_U8G2.setFont(fontName);	
        pauseAnimation = true;
        if (nav.changed(0))
        {
            _U8G2.clearBuffer();
            nav.doOutput();
            _U8G2.sendBuffer();
        }
    }
```

neu-rah, 2017-10-03:
Hi!

Thanks for all info, from the video i can see that its quite strange, the screen seems to clean even before the long press release...
what encoder driver are you using?

christophepersoz, 2017-10-03:
Hi!

I'm using ClickEncoderIn.h because I need to use the acceleration functionality ;))

And about the #define fontName, I though that I should put them before making the include, that makes sense - I'm quite sure that is the problem.

christophepersoz, 2017-10-03:
For your info, placing the #define fontName before the include u8g2Out.h does not solve the issue ^^

neu-rah, 2017-10-03:
I will try with clickencoder too, don't think its the culprit but to have long click consistency.

Because the menu does not hold any font info and if using multiple fonts then it makes sense to set the font just before calling the menu functions (inside the loop). But still that should result on a menu being drawn with another font and things overlapping and not a blank screen... that blank screen that needs a position change to redraw looks a lot like a missing `dirty` flag setting from the u8g.

does you screen work with adafruitGfx? I can imagine its quite hard to test this because all the non-menu part of code being tied to U8G primitives :\

still I have to do some testing with the code you've sent me, will keep in touch

P.S. '/' key also produces an `escape` and it can exit a menu even without an Exit option... the lonk click should also produce an `escape`, does it have the same behavior when using '/' key on suspended menu?

christophepersoz, 2017-10-03:
I'm going to try the long click to escape and see what's going on. 
On the U8G2 side. The display used is based on a SSD1306 set in Full Buffer mode with an I2C interface (which is fast enough for this dev).

I can try to remove all the code and extract it LT the menu part but it will takes time because I have some other stuff to manage aside ;)

Thanks!

neu-rah, 2017-10-03:
I imagine... an alternative is to try an example and see if it shows the same behavior.

neu-rah, 2017-10-03:
for future reference: we concluded (on gitter) that this was due the usage of an external logic to decide if the menu should be executed or not. The problem was that when resuming (start calling the menu again) the menu system had no idea of the intromission and therefor didn't draw himself again.

A possible solution would involve signaling the main menu that a drawn was needed, but we ended up using the menu internal logic instead by checking/setting `navRoot::sleepTask`...

see https://github.com/neu-rah/ArduinoMenu/wiki/Idling

christophepersoz, 2017-10-03:
Thanks a lot !

closed 2017-10-03 by christophepersoz