neu-rah/ArduinoMenu #116: Non editable field type to display a value

christophepersoz, 2017-09-30:
Hi Rui,

Is there a way to display a value inside the menu, which is non editable, and which display a value that can be the result of a calculation for example ?

Thanks

neu-rah, 2017-09-30:
yes, make it read only, it will still update

christophepersoz, 2017-09-30:
Oh great! I thought that disable will hide it from the menu. Perfect! Thanks a lot

closed 2017-09-30 by christophepersoz

christophepersoz, 2017-09-30:
Sorry, how do you determine the index value to hide it ?

neu-rah, 2017-09-30:
its zero indexed and static, so no chance of changing
[0] is the first option

the syntax is "alice in wonderland" inspired.. :D have a note to improve it.

christophepersoz, 2017-09-30:
OK, lets go for the trial and error experiments ! ;)

neu-rah, 2017-09-30:
why, you don't know your menus?
its zero indexed for each menu, and all menus have names so...

mainMenu[1];//its main menu's second option
subMenuX[3];//is subMenuX forth option

christophepersoz, 2017-09-30:
Oh, so if I have a structure like this, 

```
MENU(hwdSMENU, ">HARDWARE", doNothing, noEvent, noStyle,
    SUBMENU( _mDISP ),
    SUBMENU( _mWRAP ),
    SUBMENU( _mLEDS ),
    FIELD( _LEDW, "LED COLS ", "", _MINLEDW, _MAXLEDW, 1, 0, _updPOWERDRAW, noEvent, wrapStyle ),
    FIELD( _LEDH, "LED ROWS ", "", _MINLEDH, _MAXLEDH, 1, 0, _updPOWERDRAW, noEvent, wrapStyle ),
    FIELD( POWERDRAW, "POWER DRAW ", "mA", 0, _MAXPOWERDRAW, 0, 0, _updPOWERDRAW, noEvent, wrapStyle ),
    EXIT( "< BACK" )
    );
```

I guess that POWERDRAW is hwdSMENU[5] // fifth option of the submenu hwdSMENU.

I thought that I had to count from mainMENU ^^

neu-rah, 2017-09-30:
exact, zero indexed for each menu
plus on your code you now have a hwdSMENU variable that is of type: 

menu - https://github.com/neu-rah/ArduinoMenu/blob/master/src/menu.h#L206
menuNode - https://github.com/neu-rah/ArduinoMenu/blob/master/src/menu.h#L193
navtarget - https://github.com/neu-rah/ArduinoMenu/blob/master/src/menu.h#L101
prompt - https://github.com/neu-rah/ArduinoMenu/blob/master/src/menu.h#L55

inheriting all those method and data from the above hierarchy, the data is not very interesting because its only a shadow pointer to somewhere on flash

by indexing on it you have a prompt object
https://github.com/neu-rah/ArduinoMenu/blob/master/src/menu.h#L197

and if you know the type you can cast it to the correct type and have more usefull info, ie:
((menuField*)&hwdSMENU[3])->... //this gives you the forth field object, and you can access its members or call methods

christophepersoz, 2017-09-30:
Awesome! That's quite a lot of changes since the v2

neu-rah, 2017-09-30:
And look there are are enable/disable method for prompts :D
Sometimes I get surprised too

neu-rah, 2017-09-30:
Guess I have to highlight those things on wiki

christophepersoz, 2017-09-30:
Good idea!