neu-rah/ArduinoMenu #117: Conflict between menu library and Modbus library

manaioussema, 2017-10-03:
Hello again,
i have faced another problem when i've tried to implement the Modbus protocol in my project. I'm using a library that relays on the Hardware serial library and uses Stream functions, so i had a suspect that the menu input system and the modbus library had conflict since it uses Stream. This is only a guess and i'm requesting a confirmation. The library is at : [https://github.com/angeloc/simplemodbusng](url)
Thank you.

neu-rah, 2017-10-03:
your menu has any other IO alternative to Serial? because you can discard the menu serial IO to free it for another purpose.
what MCU are you using?

manaioussema, 2017-10-04:
My menu input is a rotary encoder not Serial, but from what i understood from the library that it relays on Stream and i thought it would make a conflict. I'm using the MSP430FR5969 MCU from TI.

neu-rah, 2017-10-04:
Hi!

_some basic stuff_
1) be sure to not use serial as part of menu input or output then.
2) be sure not to be using any pins that are also for serial communications
3) I think that **encoderIn** is not compatible with software serial (at least on AVR's)

encoderIn uses PCINT (pin change interrupts), it might interfere with **software serial**, but there is also clickEncoderIn driver that does not use PCINT (but uses timer1 library)

you can try to disable encoder for a quick check.

I have no reports of problems with hardware serial, but this might be the first, please let me know.

thanks

manaioussema, 2017-10-04:
i'm sure for the second option that serial pins are not common with anything else, but i don't how can i make sure that serial is not part of the menu input or output. i'm defining this for the menu input and output : `MENU_OUTPUTS(nokia_lcd,MAX_DEPTH
  ,ADAGFX_OUT(display,colors,fontX,fontY,{0,0,84/fontX,48/fontY})
  ,NONE//must have at least 2 entryes
);` and `NAVROOT(nav,mainMenu,MAX_DEPTH,enc,nokia_lcd);`

neu-rah, 2017-10-04:
assuming that `enc` is just the encoder, and that you are not compiling with DEBUG (not likely, it would be noticeable) I'm inclined to the third option, because software serial must use some sort of interrupt and that is known to cause problem on AVR's because _encoderIn_ also uses them, not sure about you MCU. Therefor we need to check that.

1) simple test. disable encoder input
2) use alternative encoder driver

note: avr's pcint control can not be shared between libraries.

about using **Stream**, its a C++ class and therefor just a specification. Its only a manifest of hardware on the **Serial** or similar **objects**, so we can use Stream as long as not from same Hardware.
Yes menu uses streams for all hardware IO, encoders, keyboards, even output hardware.

also, can you past your MENU_INPUTS lines?

manaioussema, 2017-10-05:
i've disabled the enc and nothing changed, i'm not using MENU_INPUTS i'm directly passing the enc to the NAVROOT.

neu-rah, 2017-10-06:
then if you do not call any menu functions like poll or doInput/doOutput the the menu is not active

encoderIn interrupts will only be active when touching the encoder, you should also not call encoder .begin function on setup and avoid even constructing the menu nav root object, that would completely remove the menu from the picture

neu-rah, 2017-10-06:
don't call encoder begin function on setup and if possible do not even construct nav root menu object or don't call its functions (like poll, doInput or do Output).

The menu will be completely removed from the scene

fell free to join us at:
https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link

manaioussema, 2017-10-06:
even when the menu is completely removed nothing changed., still same problem

neu-rah, 2017-10-06:
what other parts does your project have?

On this cases I use to subdivide the code in large parts, either by removing from the original or by starting with a simple case and add parts until discover the culprit of the combination of parts that causes the problem

we were discussing a similar case on gitter and tracked the problem down to serial port buffers on teensy -> solved

neu-rah, 2017-10-10:
> even when the menu is completely removed nothing changed., still same problem

removed even the includes?
removed all structure?

manaioussema, 2017-10-13:
sorry for not replying for long time.
yes the problem is solved but as you said it was not a problem from the encoder library, it was a wrong pin definition when i ported the library from the Arduino to the MSP. Thanks again for all your support.

closed 2017-10-13 by manaioussema