neu-rah/ArduinoMenu #118: AnsiSerial example bootloop on esp8266

reaper7, 2017-10-10:
I got bootloop on esp8266 with AnsiSerial example (serialio example works good)

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
```

this is very strange, because when I comment out all code inside setup and loop, bootloop still exists!

neu-rah, 2017-10-10:
master branch i assume, checking

reaper7, 2017-10-10:
Currently I play with dev branch but this causes on master too

neu-rah, 2017-10-10:
confirmed

neu-rah, 2017-10-10:
fixed changed a few things (https://github.com/neu-rah/ArduinoMenu/commit/f5a0e1a2dbc30c1e27ee66f76c0091f3ed0bf033)

including the LEDPIN definition, also removed an unused function (cosmetics)

got it running on my esp12e

neu-rah, 2017-10-10:
weirdness, same on dev branch, also corrected but I took some time to pinpoint it to an unused menu experimental structure that was left on the code and not causing problems on AVR's

```c++
TOGGLE((mainMenu[1].enabled),togOp,"Op 2:",doNothing,noEvent,noStyle
  ,VALUE("Enabled",enabledStatus,doNothing,noEvent)
  ,VALUE("disabled",disabledStatus,doNothing,noEvent)
);
```

also pushed correction to dev branch

thanks for reporting it :D

reaper7, 2017-10-10:
both, master and dev starts OK, tnx!

Still about the same example from dev branch:
Positions 1-3,5-9 works very well but when I enter to **[4] Sub-Menu**, 
then keys / and * works as expected but keys + and - causes reset

```
Exception (28):
epc1=0x40202818 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3fff04c0 end: 3fff0770 offset: 01a0

>>>stack>>>
3fff0660:  00000000 00000000 3ffef164 40202897  
3fff0670:  00000000 0000000d 00000000 00000007  
3fff0680:  00000006 00000001 3ffef264 00000001  
3fff0690:  00000003 000000ff 3ffef164 402049a8  
3fff06a0:  00000006 00000000 00000004 00000007  
3fff06b0:  00000003 00766cff fffffffc 01000000  
3fff06c0:  00000000 feefeffe feefeffe 00000008  
3fff06d0:  3ffef158 3ffef3f4 00000001 3ffef73c  
3fff06e0:  40204a10 3ffef254 3ffef164 40204a21  
3fff06f0:  00000000 3ffef254 3ffef164 40204d38  
3fff0700:  00000003 00766cff 3ffef540 40205586  
3fff0710:  3ffe9730 3ffef540 3ffef164 402042ac  
3fff0720:  00000003 00766cff 3ffef138 40205556  
3fff0730:  00000001 3ffef188 3ffef138 40204348  
3fff0740:  3fffdad0 00000000 3ffef138 40202cb4  
3fff0750:  3fffdad0 00000000 3ffef734 40205260  
3fff0760:  feefeffe feefeffe 3ffef750 40100108  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
```
Exception decoder for this dump:
```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 14 results
0x40202818: operator(Print&, Menu::prompt&) at C:\PROGRAMY\arduino_projekty\libraries\ArduinoMenu\src/menuIO/serialOut.h line 24
0x40202897: showEvent(Menu::eventMask, Menu::navNode&, Menu::prompt&) at C:\PROGRAMY\arduino_projekty\libraries\ArduinoMenu\src/menuIO/serialOut.h line 24
0x402049a8: Menu::action::operator()(Menu::eventMask, Menu::prompt&) const at C:\PROGRAMY\arduino_projekty\libraries\ArduinoMenu\src/menuBase.h line 140 (discriminator 1)
:  (inlined by) Menu::prompt::operator()(Menu::eventMask, Menu::prompt&) const at C:\PROGRAMY\arduino_projekty\libraries\ArduinoMenu\src/menu.h line 109 (discriminator 1)
:  (inlined by) Menu::navNode::doNavigation(Menu::navCmd) at C:\PROGRAMY\arduino_projekty\libraries\ArduinoMenu\src/menu.cpp line 526 (discriminator 1)
0x40204a10: Menu::navTarget::doNav(Menu::navNode&, Menu::navCmd) at C:\PROGRAMY\arduino_projekty\libraries\ArduinoMenu\src/menu.cpp line 450
0x40204a21: Menu::navTarget::doNav(Menu::navNode&, Menu::navCmd) at C:\PROGRAMY\arduino_projekty\libraries\ArduinoMenu\src/menu.cpp line 453
0x40204d38: HardwareSerial::read() at C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\cores\esp8266/HardwareSerial.cpp line 171
0x40205586: Menu::serialIn::read() at C:\PROGRAMY\arduino_projekty\libraries\ArduinoMenu\src/menuIO/serialOut.h line 24
0x402042ac: Menu::navTarget::parseInput(Menu::navNode&, Menu::menuIn&) at C:\PROGRAMY\arduino_projekty\libraries\ArduinoMenu\src/menu.cpp line 553
0x40205556: Menu::serialIn::available() at C:\PROGRAMY\arduino_projekty\libraries\ArduinoMenu\src/menuIO/serialOut.h line 24
0x40204348: Menu::navRoot::doInput(Menu::menuIn&) at C:\PROGRAMY\arduino_projekty\libraries\ArduinoMenu\src/menu.cpp line 564
0x40202cb4: Menu::navRoot::doOutput() at C:\PROGRAMY\arduino_projekty\libraries\ArduinoMenu\src/menuIO/serialOut.h line 24
:  (inlined by) Menu::navRoot::poll() at C:\PROGRAMY\arduino_projekty\libraries\ArduinoMenu\src/menu.h line 750
:  (inlined by) loop at C:\PROGRAMY\arduino_projekty\libraries\ArduinoMenu\examples\Serial\ansiSerial\ansiSerial/ansiSerial.ino line 317
0x40205260: loop_wrapper at C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 57
0x40100108: cont_norm at C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266\cores\esp8266/cont.S line 109
```

neu-rah, 2017-10-10:
AVR ok, checking ESP...
you're welcome to join us on https://gitter.im/rust-lang/rust?utm_source=share-link&utm_medium=link&utm_campaign=share-link if you wish

P.S.yes it does so.

neu-rah, 2017-10-10:
> .../menuIO/serialOut.h line 24

same example? 
just to confirm, because ansiSerialOut includes it also

reaper7, 2017-10-10:
yes, same example from dev branch...

I see, exception decoder show strange information :) I don't know why!

neu-rah, 2017-10-10:
got it, the problem was the handler (deactivated it)
that handler was showing events and invalid item is being passed -> BUG!

works ok on simple event handlers

pushed error circumvent version to dev... still need to trace the origin.

event handler parameters are a current revision topic for v4

neu-rah, 2017-10-11:
reviewed handler calls, regression from (eventMask,prompt&) back to (eventMask,navNode&,prompt&)

reuse example needed them to call handlers without a navNode, the example is now recuring to local variable to obtain them.

this was a change on v4

reaper7, 2017-10-11:
tnx too!
I found Your libs yesterday (AnsiStream lib too - very usefull for me!).
It's a really big lib and has a lot of potential.
I understand how hard it is to create lib for many cpu's :)

BTW I must read wiki how to connect keys via PCF8574 as input (this is important for esp with a small number of pins :) )

neu-rah, 2017-10-11:
yes, very difficult, thanks for your interest on the lib and for having the patience to report dev branch issues.

you will probably want to see the codeCtrl example to implement custom input.

I have some ESP01 :+1: 

my VirtualPins idea was about making pins accessible as local pins, no mater if they were over I2C or SPI.... and eventually others because the medium support was a driver. But that would have to be accepted by Arduino Guys... we sure had less libraries to write if done so.

reaper7, 2017-10-11:
👍

neu-rah, 2017-10-12:
closing... feel free to reopen if not solved or open another issue.

closed 2017-10-12 by neu-rah