neu-rah/ArduinoMenu #121: esp8266 compilation error (arduino 1.8.5)

reaper7, 2017-10-19:
today I update my local copy of this lib from 4.0.1 to current 4.0.6
and I got this error msg:
```
xtensa-lx106-elf-gcc: error: C:\tmp\arduino_build_229612\libraries\ArduinoMenu\ArduinoMenu.a: No such file or directory
``` 
I found that problematic changes have been added with commit https://github.com/neu-rah/ArduinoMenu/commit/5198bb0fed7199b0cf92c9ac2945ec03f194ed7b
more precisely, change inside _library.properties_ -> _dot_a_linkage_ from **false** to **true**

when I change this property to false everything is compiled without errors

neu-rah, 2017-10-19:
thanks for the feedback, I wonder if the same could be achieved by cleaning up the project...
but anyway I will set it back to false

reaper7, 2017-10-20:
solved 👍

closed 2017-10-20 by reaper7