neu-rah/ArduinoMenu #122: buttons navigation question

reaper7, 2017-10-21:
I play with oled example extended with physical buttons.

Technically everything works fine. 
Buttons are linked to the nav commands like on image:
[image]

now, what is logical for me:
in position "**Test**", I can increase value by pressing KEY_UP or KEY_RIGHT
and decrease by pressing KEY_DOWN or KEY_LEFT.
In this application the logic of the keys is obvious 👍 

now, what is not obvious for me:
navigation in menu... for e.g.
looking at displayed menu (attached image), I expect that if I press KEY_DOWN, then selected menu item will move from "**Test**" to "**Sub-Menu**" but in fact it moves UP (of course the KEY_UP works in the same way).

So...time to ask :)
Is there a way to personalize the keys? And I'm not talking about reverse assignment (like KEY_UP to downCmd etc.), it would solve navigation in menu but broke logically increase / decrease value.
Maybe any option for reverse functionality in menu/submenu ?
Or reverse functionality for KEY_UP / KEY_DOWN for navigation and KEY_LEFT / KEY_RIGHT for change value?

neu-rah, 2017-10-21:
yes, this was a known problem, and even if you go to code and make things inverse for fields, they will work ok for keys but will look reverse to encoder :D (the default setup is for encoder)
earlier versions had a global option to inverse fields, that might work for your case, I suppose you are using  only one input method.
Because we can coalesce inputs from multiple sources a global option was not adequate either, humans have inverse expectations from keys vs encoders, the class menuIn was created on version 4 and the option is now intrinsic to each input source (as stated in the README.md, V4 changes).
Still, your case, using nav cmds and having only one input method would be easily solved by the global option :S (missed that case) instead of having to create a menuIn derived class. My fault.
I will bring the global option again :D

this was done on commits culminating on https://github.com/neu-rah/ArduinoMenu/commit/2c2301ffe0a29f9e86ea75180bcf47a8677094c3

I promise to be quick

neu-rah, 2017-10-21:
P.S. are you using dev or master branch?

reaper7, 2017-10-21:
now on master

neu-rah, 2017-10-21:
Ok pushed v4.0.8 to master with invertFieldKeys option

can get/set with `options->invertFieldKeys`

to define a complete new set of options with something like:
```c++
config myOptions('*','-',defaultNavCodes,false);//global

void setup() {
    options=&myOptions;
    //...
}
```

options class at: https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuBase.h#L174
default options: https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuBase.cpp#L46

neu-rah, 2017-10-21:
BTW, you have a nice project picture there, would you mind if i use it?

reaper7, 2017-10-21:
ok, I will test it

about pic, yes of course, no problem!

i2c button shield by **deshipu** from https://www.tindie.com/products/deshipu/x-pad-20-buttons-shield-for-d1-mini/

I prepared a temporary menuIO -> PCF8574KeyIn.h for it.
If you want to refine it and use it...
```
/* -*- C++ -*- */

/**************

***/

#ifndef __PCF8574KeyIn_h__
  #define __PCF8574KeyIn_h__

  #include "../menu.h"
  #include <Wire.h>

  namespace Menu {

    #include "keyMapDef.h"

    #ifndef BOUNCE_TICK
    #define BOUNCE_TICK 30
    #endif

    //if you hold/repeat a key for this ammount of time we will consider it an escape
    #ifndef ESCAPE_TIME
    #define ESCAPE_TIME 1500
    #endif
    //emulate a stream keyboard, this is not using interrupts as a good driver should do
    // AND is not using a buffer either!
    template <int N, int _dev=0x20, int _sda=SDA, int _scl=SCL>                 //default pcf8574 address=0x20
    class PCF8574KeyIn:public menuIn {
    public:
      keyMap* keys;
      int lastkey;
      TwoWire iicWire;
      unsigned long pressMills=0;
      PCF8574KeyIn<N, _dev, _sda, _scl>(keyMap k[]):keys(k),lastkey(-1) {}
      void begin() {
        iicWire.begin(_sda, _scl);
      }
      int available(void) {
        int ch=peek();
        if (lastkey==-1) {
          lastkey=ch;
          pressMills=millis();
        }
        else if (ch == -1 && millis()-pressMills < BOUNCE_TICK)
        {
          lastkey = -1;  //released = it's bounce. reset lastkey
          return 0;
        }
        else if (ch != -1 && millis()-pressMills > BOUNCE_TICK) return 1;
        else if (ESCAPE_TIME&&millis()-pressMills>ESCAPE_TIME) return 1;

        if (ch==lastkey) return 0;

        return 1;
      }
      int peek(void) {
        //Serial<<"peek"<<endl;

        iicWire.requestFrom(_dev, 1);
        uint8_t val = iicWire.read();

        for(int n=0;n<N;n++) {
          int8_t pin=keys[n].pin;
          if (((val)&1<<pin)==0) {
            //Serial<<"key "<<pin<<" pressed"<<endl;
            return keys[n].code;
          }
        }
        return -1;
      }
      int read() {
        //Serial<<"read"<<endl;
        int ch=peek();
        if (ch==lastkey) return -1;
        int tmp=lastkey;
        bool longPress=ESCAPE_TIME&&millis()-pressMills>ESCAPE_TIME;
        //Serial<<"read lastkey="<<lastkey<<" ch="<<ch<<endl;
        //Serial<<"down time:"<<millis()-pressMills<<endl;
        pressMills=millis();
        lastkey=ch;
        return longPress?options->getCmdChar(escCmd):tmp;//long press will result in escape
      }
      void flush() {}
      size_t write(uint8_t v) {return 0;}
    };

  }//namespace Menu

#endif
```

neu-rah, 2017-10-21:
:D excellent, can you PR that?

reaper7, 2017-10-21:
It will be better if You do it :)

I'm not a programmer, hard coded TwoWire inside...I don't know if that's a good idea :)
but I don't know how to pass TwoWire reference from main sketch to template or/and whether it is necessary. 

So...clean and Your additional comment inside code is needed.

neu-rah, 2017-10-21:
bother not, the PR is not definitive we can review it and change it... but I would be pleased to have your authorship preserved.

we can add PCINT to that very easy, but when you use a PCINT manager it tends not to be compatible with others, and sometimes causing incompatibilities with final user sketches. Wonder if Arduino already has official PCINT manager.
I use my own PCINT library for encoderIn, but ofcoz there's an alternative encoder driver that does not use it.

neu-rah, 2017-10-21:
"I'm not a programmer" well I can say you are a damn quick programmer, pulled out an input driver in the time of a couple of posts here :D damn!

reaper7, 2017-10-21:
this is only few lines difference between this and Your KeyIn :D

neu-rah, 2017-10-21:
but that implies the ability to find it, read it, understand it.. and improve it

neu-rah, 2017-10-21:
thank you!

you can also use gitter for a more fluent talking chanel if you wish

I'm closing this, reopen if needed

closed 2017-10-21 by neu-rah

reaper7, 2017-10-21:
if I set:
config myOptions('>','-',defaultNavCodes,**false**);
then inc/dec value works ok, navigation is reversed like in my first post.

if I set:
config myOptions('>','-',defaultNavCodes,**true**);
then inc/dec value works reversed but navigation is still reversed

Maybe I'm doing something wrong?

neu-rah, 2017-10-21:
reverse you navigation

reaper7, 2017-10-21:
self facepalm :)
👍 

BTW I'm still discovering your library, 
tell me, is there any possibility to disable some buttons depending on the current operation?
for e.g.: 
- navigation on menu -> enabled only UP/DOWN (+ENTER / ESC) and disabled LEFT/RIGHT
- changing value -> enabled only LEFT/RIGHT (+ENTER/ESC) and disabled UP/DOWN

neu-rah, 2017-10-21:
well not yet, but its an interesting idea, current state of 2d nav left/right navigation is to do the same as up/down as you have noticed. The keys were there for future expansion, guess the time has come.
We should think on a way that this could be configurable, I mean without having to derive fields of menu class.
Thinking about axis, up/down, left/right, enter/esc and assignment/mapping of each axis to either
nav - select or leave menu
browse - move between menu options
value - change numeric or enumerated value
hmmm... :thinking:

reaper7, 2017-10-21:
separated axis seems promising

neu-rah, 2017-10-21:
yup, still i will try to implement it in a way that does not get more code into the core, because the core is already too heavy for AVR's. menuIn seem the logical choice they were created on V4 for this kind of stuff, until v3 we were using normal streams, hence the strange usage of characters to mean cmd's, now we have a struct to clean that from the core and let menuIn object emit cmd's.

but we getting off topic :D
please join gitter, its the place for this kind of discussion... exchanging ideas and exploring possibilities.

neu-rah, 2017-10-21:
Hi!
I don't know if you are receiving private msgs from gitter, so I will post it here.
All needed resources to implement the keys behavior that you describe are already available on menuIn class API. By default the base class provides only dumb functionality and stores no internal state, to provide functionality for invariant input devices (the ones that do not need inversion).
Some like analogAxisIn implement an internal state that is completely device specific.
The menuIn API (got to add this to the wiki).

```c++
        //menuIn API
        virtual void setFieldMode(bool) {}//to set field mode true=entering edit field, false= leaving edit field
        virtual bool fieldMode() const {return false;}//eventually return internal state, if available
        //aux functions
        inline void fieldOn() {setFieldMode(true);}//entering edit field
        inline void fieldOff() {setFieldMode(false);}//exiting edit field
```

based on this info and your internal state you can sen Up/Down events based on either down/up keys or left/right ;)