neu-rah/ArduinoMenu #126: keyMap initialize error

niijyeni, 2017-10-23:
if initialize keyMap with this code ( reference from example u8g2)
`keyMap btn_map[]={
  {-4, options->getCmdChar(enterCmd)}
  };`
btn_map[0].code will be error

but do like this all ok
`keyMap btn_map[]={
  {-4, '*'}//options->getCmdChar(enterCmd)}
  };`
-------------------------
please help me

neu-rah, 2017-10-23:
Try this, I've not compiled it, so if not working please let me know
```c++
keyMap btn_map[]={ {-4, options->navCodes[enterCmd} };
```

this is **for sure** and area to be subject of revision on next version.

niijyeni, 2017-10-23:
doesn't work, same error like above code.

debug ouput enterCmd:55

niijyeni, 2017-10-23:
this is my define code:
keyMap btn_map[]={
  {-6, options->navCodes[downCmd].ch},
  {-5, options->navCodes[upCmd].ch},
  {-4, options->navCodes[enterCmd].ch}
  };

debug code:
  Serial << "downCmd:" << btn_map[0].code << endl;
  Serial << "upCmd:" << btn_map[1].code << endl;
  Serial << "enterCmd:" << btn_map[2].code << endl;

debug output:
downCmd:57
upCmd:0
enterCmd:55

neu-rah, 2017-10-23:
ohh, sorry u thought the error was was on first line, and still I've failed
so, you want to access the defined code, right?

`btn_map[0].cmd` will give you the associated command code
`btn_map[0].ch` will give you the associated character

struct defined here: https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuBase.h#L170

niijyeni, 2017-10-23:
no, i don't want to access the defined code.
i want to define keyMap for keyIn, but `options->getCmdChar(xxxCmd)` return error char, so input does not correctly handle

neu-rah, 2017-10-23:
hmm interesting...

this is the definition of default nav keys
https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuBase.cpp#L33

what country or codepage are you in?

niijyeni, 2017-10-23:
china

neu-rah, 2017-10-23:
this is strange because if `keyMap btn_map[]={ {-4, '*'}` is ok on main file it should be the same on menuBase.cpp

please use `keyMap btn_map[]={ {-4, '*'}` for now, i will investigate more

niijyeni, 2017-10-23:
ok, thank for your help

neu-rah, 2017-10-23:
and your IDE is?
and sorry for taking so long to understand the problem, i was divided between tasks here :D

niijyeni, 2017-10-24:
arduino 1.8.2

neu-rah, 2017-12-24:
this was related to C++ initialization order problem, so we can not call that function on startup.

should be ok when replacing this:
```c++
 options->getCmdChar(enterCmd)
```
by this:
```c++
 defaultNavCodes[enterCmd].ch
```
pushed new version to master with corrected examples

closed 2017-12-24 by neu-rah