neu-rah/ArduinoMenu #127: choose show garbled

niijyeni, 2017-10-24:
arduino due / ide 1.8.2 / ssd1306 i2c

#define fontX 6
#define fontY 13
#define offsetX 0
#define offsetY 0
#define U8_Width 128
#define U8_Height 64
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

int chooseTest=-1;
CHOOSE(chooseTest,chooseMenu,"Choose",doNothing,noEvent,noStyle
  ,VALUE("First",1,doNothing,noEvent)
  ,VALUE("Second",2,doNothing,noEvent)
  ,VALUE("Third",3,doNothing,noEvent)
  ,VALUE("Last",-1,doNothing,noEvent)
);

u8g2.enableUTF8Print();
u8g2.setFont(u8g2_font_wqy12_t_gb2312b);

enter choose menu will see show garbled char on title, please help me

neu-rah, 2017-10-24:
Hi
I assume there a main menu where that.
Can you try with simple ASCII characters? If not its not important.
I think this is caused by you using a character set that has more than 1 byte per character, If so I see a reason for them to be garbled. 
Its a bug on current version.

neu-rah, 2017-10-24:
just to confirm, are you using this library version?
https://github.com/larryli/u8g2_wqy

niijyeni, 2017-10-24:
yes，I‘m try simle ASCII font (like u8g2_font_6x12_mf), all fine

neu-rah, 2017-10-24:
I can see a motive for this but its not easy for me to setup a test case, still struggling on how to use the alternative library. Have you inserted/changed the include into menuIO/u8g2Out.h?

please help me :D

neu-rah, 2017-10-24:
also can you provide an example of a correct print with that font without using the menu?

neu-rah, 2017-10-24:
the bug is located here: https://github.com/neu-rah/ArduinoMenu/blob/master/src/io.cpp#L10

this function should use U8G2 drawGlyph or similar function to print it correctly. being this a generic function that uses multiple base drivers the API would have to be extended so that each driver could provide this function with the correct print mechanism.

still i think that some define activates this kind of printing, maybe U8G2_USE_LARGE_FONTS, but I'm not sure if this define is specific to the alternative lib. version.

niijyeni, 2017-10-24:
#include <Arduino.h>
#include <Wire.h>

#include <U8g2lib.h>
#include <U8g2wqy.h>

U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

void setup(void) {
  u8g2.begin();
  u8g2.enableUTF8Print();		// enable UTF8 support for the Arduino print() function
}

void loop(void) {
  u8g2.setFont(u8g2_font_wqy12_t_gb2312b);  // use chinese2 for all the glyphs of "你好世界"
  u8g2.setFontDirection(0);
  u8g2.firstPage();
  do {
    u8g2.setCursor(0, 12);
    u8g2.print("For testing purposes this is a long text.");
    u8g2.setCursor(0, 24);
    u8g2.print("这是一段用于测试超长的文字。");    // Chinese "Hello World" 
  } while ( u8g2.nextPage() );
  delay(1000);
}

neu-rah, 2017-10-24:
I'm missing something

5:21: fatal error: U8g2wqy.h: No such file or directory

niijyeni, 2017-10-24:
https://github.com/larryli/u8g2_wqy
download and install this library

neu-rah, 2017-10-24:
did, but even with include <U8g2_wqy.h> or <U8g2wqy.h> I have the same error... checking my system.

neu-rah, 2017-10-24:
my fault :D
```c++
#include <u8g2_wqy.h>
```
case sensitive

neu-rah, 2017-10-24:
:thinking: still missing the font

25: undefined reference to `u8g2_font_wqy12_t_gb2312b'

neu-rah, 2017-10-24:
I can see that the lib has some shell script to generate files ie:
https://github.com/larryli/u8g2_wqy/blob/107f444677c7883b3770bd3dd49a555ccd0d895f/build.sh#L8

refering the font. I do not have a windows system, and even if i did running an executable is not something i would do without a sandbox...

is there any other source for the file? or am I missing something?

niijyeni, 2017-10-24:
https://raw.githubusercontent.com/larryli/u8g2_wqy/master/src/u8g2_font_wqy.c
how to send a source file to you?
2,932KB

neu-rah, 2017-10-24:
Hi niijyeni!

I've created a new branch (**wqy**) with a version that might solve your problem, I was able to test it using an extended European font on the arduino nano with U8G2.

https://github.com/neu-rah/ArduinoMenu/tree/wqy

This version does not check boundaries, but that should not be a problem for your kind of menu (using all device).

niijyeni, 2017-10-25:
choose menu title still does not display correctly
if fontx set to 6 main menu title does not display all content
```
#define fontX 6 // 必须小于字符宽度的一半（当时使用中文字体的时候，即定义：U8G2_USE_LARGE_FONTS）
#define fontY 12
#define offsetX 0
#define offsetY 2
#define U8_Width 128
#define U8_Height 64
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

// ... hide code

MENU(mainMenu,"main menu 菜单 test output", doNothing,noEvent,noStyle
  ,SUBMENU(setLed)
  ,SUBMENU(subMenu)
  ,OP("LED On",ledOn,enterEvent)
  ,OP("LED Off",ledOff,enterEvent)
  ,SUBMENU(selMenu)
  ,SUBMENU(chooseMenu)
  ,OP("Alert test",doAlert,enterEvent)
  ,EXIT("<Exit")
);

// ... hide code

MENU_OUTPUTS(out,MAX_DEPTH
  ,U8G2_OUT(u8g2,colors,fontX,fontY,offsetX,offsetY,{0,0,U8_Width/fontX,U8_Height/fontY})
  ,NONE//SERIAL_OUT(Serial)
);
```
[main menu photo](https://share.weiyun.com/5989db0e1cd424b4ac3d902125cf6573)
[choose menu photo](https://share.weiyun.com/ae2bb561873c96c22f5947198455224a)

neu-rah, 2017-10-25:
is this using the version from wqy branch, right?

niijyeni, 2017-10-25:
sorry, my fault.
main menu displayed correctly.
but choose menu still garbled

neu-rah, 2017-10-25:
Ok, a step in the right direction, I will check fields print on my test case.
thanks for helping me test it, my test case is a bit limited, nor using the same font.

neu-rah, 2017-10-25:
yes, confirmed... I'm on top of it

niijyeni, 2017-10-25:
this is my choose menu definiton code
```
int chooseTest=-1;
CHOOSE(chooseTest,chooseMenu,"Choose",doNothing,noEvent,noStyle
  ,VALUE("First",1,doNothing,noEvent)
  ,VALUE("Second",2,doNothing,noEvent)
  ,VALUE("Third",3,doNothing,noEvent)
  ,VALUE("Last",-1,doNothing,noEvent)
);
```

neu-rah, 2017-10-25:
other fields like `select`, `toggle` and `field` working ok?

niijyeni, 2017-10-25:
toggle, select works fine
field not used yet

neu-rah, 2017-10-25:
choose bug fixed, that value should had not been there either. pushed changes to wqy branch (and master)

niijyeni, 2017-10-25:
choose title does not show current value now? 
[see choose photo](https://share.weiyun.com/bc42da077dcb68cbb92f81d997f7fa72)

neu-rah, 2017-10-25:
nope, it was a bug. because it shows selected below and in front of title JUST when choose is closed.
if not, please tell me.

neu-rah, 2017-10-25:
is it ok for you?
if having other non related problems feel free to open a new issue.

niijyeni, 2017-10-25:
found no other problems at the moment, it seems to work very well

neu-rah, 2017-10-25:
Thank you, for helping to make this better. In case you find some more problem please open an issue or reach me at gitter.
The patch that was implemented for to print Chinese characters is not definitive because it does not respect margins. But that should only affect when not using the full screen and if the text exceeds the area. As soon it is solved, this correction will be joining the `master` branch.
Closing this.

closed 2017-10-25 by neu-rah