neu-rah/ArduinoMenu #129: idle function does not update idle screen data

niijyeni, 2017-10-26:
'''
//when menu is suspended
result idle(menuOut& o,idleEvent e) {
  o.clear();
  switch(e) {
    case idleStart:o.println("suspending menu!");break;
    case idling:
      o.setCursor(0,0);
      o << "Voltage: " << voltage_bat_1 << " V";
      o.setCursor(0,1);
      o << "Voltage: " << current_bat_1 << " V";
      
      break;
    case idleEnd:o.println("resuming menu.");break;
  }
  return proceed;
}
'''
Not refresh the screen after menu enters an idle state, making it impossible to display real-time status data, please help me!

neu-rah, 2017-10-26:
hi

there are multiple ways of doing this update see here: https://github.com/neu-rah/ArduinoMenu/wiki/Idling

there is also a way to signal the need for a redraw from the idle function. The idle function is intended as a way of updating multiple devices n parallel, following the same usage as the menu. That is most of the time not necessary to use with a single display. 

Most users use this method: https://github.com/neu-rah/ArduinoMenu/wiki/Idling

neu-rah, 2017-10-26:
added this text to 'idle' wiki page as it was the found solution (on gitter)

```c++
  //...
  //on u8g2 or other devices that uses doInpu/doOutput instead of poll
  nav.doInput()
  if (nav.sleepTask) {
    u8g2.firstPage();
    do {
      u8g2.setCursor(0, 15);
      u8g2.print("suspended");
    } while ( u8g2.nextPage() );
  } else {
    if (nav.changed(0)) {
      u8g2.firstPage();
      do nav.doOutput(); while(u8g2.nextPage());
    }
  }
  //...
```

closed 2017-10-26 by neu-rah