neu-rah/ArduinoMenu #130: Crash when navigating with numbers and menu index doesn't exist

martinmarcos, 2017-10-31:
I made a rudimentary driver that streams digits pressed on a 4x4 matrix keypad. The keypad has numbers 0-9, *,# and capital letters A-D. I am able to navigate through the menu system using the numbers of the keypad. It behaves exactly the same as if numbers are input into the serial. 

The problem is the menu system becomes unresponsive when entering a number for which there isn't a menu entry. For example if the menu being displayed has 4 entries and you press numbers 1-4, it will do an enter command on the corresponding entry, on the other hand if numbers greater than 4 are entered the menu system freezes.

I am using version v3.1.13. I've looked through the issues to see if the bug was fixed on newer versions but was not able to find it.

By the way, I tried checking out the newest version and compiling and I get errors. So I've rolled back to v3.1.13. What would I need to modify in order to migrate to the newest version?

Regards.

neu-rah, 2017-10-31:
This is the first report. Checking...
Thanks.

neu-rah, 2017-10-31:
confirmed, pushed correction to branch 3.x

neu-rah, 2017-10-31:
also made this correction into v4.x

solved?

martinmarcos, 2017-10-31:
Great!

> solved?

Will let you know as soon as I can test it.

martinmarcos, 2017-11-07:
Confirmed. Bug was fixed.

closed 2017-11-07 by neu-rah