neu-rah/ArduinoMenu #132: Data/Time File as pad is very hard to see

niijyeni, 2017-11-08:
https://share.weiyun.com/23289dad20f34ae57dabb58efbfc43e2

neu-rah, 2017-11-08:
Hi has a similar look on u8g2 when line not being properlly cleared (because coordinates of clearing rect were not correct) in combination with some u8g2 font fill style... however this can arise from the choice of colors.
Can you paste your color table?
This asPad feature being the newest is probably messing something

niijyeni, 2017-11-08:
```
// 色彩定义
const colorDef<uint8_t> colors[] MEMMODE={
  {{0,0},{0,1,1}},//bgColor
  {{1,1},{1,0,0}},//fgColor
  {{1,1},{1,0,0}},//valColor
  {{1,1},{1,0,0}},//unitColor
  {{0,1},{0,0,1}},//cursorColor
  {{0,0},{1,1,1}},//titleColor
};
```

neu-rah, 2017-11-08:
this is the same color table I had, and i can confirm a very hard read.

it improved with `u8g2.setBitmapMode(0);` but i think there can be a better choice of colors

niijyeni, 2017-11-08:
There is no need to extend the color table for the time being, just need to differentiate between aspad rendering methods

neu-rah, 2017-11-08:
I've observed some glitches when using u8g2 and asPad style menus, probably some clearing react being drawn out of place

neu-rah, 2018-06-13:
merged branch wqy with master, and noticed the exact same problem on asPad menu entries.

solved it be using a different font (mode).
was using u8g2_font_7x13_te and changed it to u8g2_font_7x13_mf

so, this is related to colors and background clearing. the `t` at font name end mean transparent and therefore it becomes very hard to read

was able to use u8g2_font_7x13_me, monospace and extended characters

opened a new issue with more specific info https://github.com/neu-rah/ArduinoMenu/issues/185

closed 2018-06-13 by neu-rah