neu-rah/ArduinoMenu #133: Not compiling using ClickEncoder

Triscus, 2017-11-13:
Hello,

my first try was to adapt the ucglib [example ](https://github.com/neu-rah/ArduinoMenu/blob/master/examples/UCGLib/UCGLib/UCGLib.ino), so I can use it with the SSD1351 Display and a click Encoder.

The display works fine with SerialIn, but I am struggling to get the sketch running with a clickEncoder, any help is appreciated.

I am using a Teensy 3.2 and the Encoder Library from here : [https://github.com/0xPIT/encoder](https://github.com/0xPIT/encoder)

Find the sketch and the relevant debug Output below:

`/********************
Arduino generic menu system
UCGLib menu example
UCGLib: https://github.com/olikraus/Ucglib_Arduino

Sep. 2017 Rui Azevedo - ruihfazevedo(@rrob@)gmail.com

menu on UCGLib device
output: ST7735
input: Serial
mcu: nano328p
*/

#include <SPI.h>
#include <Ucglib.h>
#include <menu.h>
#include <menuIO/UCGLibOut.h>
#include <menuIO/chainStream.h>
#include <menuIO/clickEncoderIn.h>
#include <ClickEncoder.h>

using namespace Menu;

#define UC_Width 128
#define UC_Height 128

#ifdef ESP8266
#define UC_CS   10
#define UC_DC   9
#define UC_RST  6
#else
#define UC_CS   10
#define UC_DC   9
#define UC_RST  6
#endif

//font size plus margins
#define fontX 6
#define fontY 11

Ucglib_SSD1351_18x128x128_FT_HWSPI ucg(UC_DC, UC_CS, UC_RST);

const char* constMEM hexDigit MEMMODE = "0123456789ABCDEF";
const char* constMEM hexNr[] MEMMODE = { "0","x",hexDigit,hexDigit };
char buf1[] = "0x11";//<-- menu will edit this text

int chooseTest = -1;
CHOOSE(chooseTest, chooseMenu, "Choose", doNothing, noEvent, noStyle
	, VALUE("First", 1, doNothing, noEvent)
	, VALUE("Second", 2, doNothing, noEvent)
	, VALUE("Third", 3, doNothing, noEvent)
	, VALUE("Last", -1, doNothing, noEvent)
);

MENU(mainMenu, "Main menu", doNothing, noEvent, wrapStyle
	, OP("Op1", doNothing, noEvent)
	, EDIT("Hex", buf1, hexNr, doNothing, noEvent, noStyle)
	, SUBMENU(chooseMenu)
	, EXIT("<Back")
);
[... 319 lines cut ...]
   #define XFOR_EACH_1(what, params, x, ...) what(params,x)
baseMacros.h:59: note  in expansion of macro XFOR_EACH_1
   XFOR_EACH_1(what, params,  __VA_ARGS__)
baseMacros.h:7: note  in expansion of macro XFOR_EACH_2
   #define CONCATENATE2(arg1, arg2)  arg1##arg2
baseMacros.h:112: note  in expansion of macro XFOR_EACH_
   #define XFOR_EACH(what, params , x, ...) XFOR_EACH_(FOR_EACH_NARG(x, __VA_ARGS__), what, params, x, __VA_ARGS__)
macros.h:161: note  in expansion of macro XFOR_EACH
   XFOR_EACH(CALL,WITH(REF,id,maxDepth),__VA_ARGS__)
UCGLib_test.ino:86: note  in expansion of macro MENU_OUTPUTS
   MENU_OUTPUTS(out, MAX_DEPTH
 
UCGLib_test.ino: 93:37: error: no matching function for call to 'Menu::ClickEncoderStream::ClickEncoderStream(ClickEncoder&, int)
   ClickEncoderStream encStream(qEnc, 1);                                       \\ Pass reference to ClickEncoderStream library
 
UCGLib_test.ino:17: In file included from
clickEncoderIn.h:43: note  candidate  Menu  ClickEncoderStream  ClickEncoderStream()
     class ClickEncoderStream*: public menuIn {
clickEncoderIn.h:43: note    candidate expects 0 arguments, 2 provided
clickEncoderIn.h:43: note  candidate  constexpr Menu  ClickEncoderStream  ClickEncoderStream(const Menu  ClickEncoderStream&)
clickEncoderIn.h:43: note    candidate expects 1 argument, 2 provided
clickEncoderIn.h:43: note  candidate  constexpr Menu  ClickEncoderStream  ClickEncoderStream(Menu  ClickEncoderStream&&)
clickEncoderIn.h:43: note    candidate expects 1 argument, 2 provided
Build failed for project 'UCGLib_test'
`

neu-rah, 2017-11-13:
Hi!

I was able to go a bit further with the following includes
```c++
#include <SPI.h>
#include <Ucglib.h>
#include <menu.h>
#include <menuIO/UCGLibOut.h>
#include <menuIO/chainStream.h>
#include <ClickEncoder.h>
#include <menuIO/clickEncoderIn.h>
#include <menuIO/serialOut.h>
```

**note:** _somehow ClickEncoder is defined to be included only on some avrs, probably causing problem with some other mcu, however i know it works with teensy. here fixing the problem by including `ClickEncoder.h` before `menuIO/clickEncoderIn.h`

your code was also missing `#include <menuIO/serialOut.h>`

however still have some ucglib errors...

Triscus, 2017-11-13:
Ah ok, I already tried changing the orders, but had no luck.  I added the interrupt stuff for the encoder and now it's working like charm. 

Thank you very much for the support

closed 2017-11-13 by Triscus