neu-rah/ArduinoMenu #134: SSD1306Ascii Example causes Arduino to restart

LuckyLuzz, 2017-11-16:
Hey,
long time no see, you implemented quite a lot new features since version 2! Great :) 

I just tried a SSD1306 with that SSD1306Ascii example you provided, after uploading it just shows "Hello World" and than the arduino nano restarts. No Menu to see. 

Keep up that good work!

neu-rah, 2017-11-16:
Hi! :D
i will check that, thanks for reporting it.

neu-rah, 2017-11-16:
oops, i cant check that, have no SSD1306.
but that display also works with U8G2 and U8x8 and i guess adafruitGfx too

LuckyLuzz, 2017-11-16:
Ok will try with the other libs.

Christmas is close, may I donate you one display?

neu-rah, 2017-11-16:
of coz :D
you are also welcome to join us at https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link

LuckyLuzz, 2017-11-17:
The "ArduinoMenu_library\examples\LCDs\Malpartida\I2C\I2C" is not debugging.

libraries\ArduinoMenu_library\src/menuIO/lcdOut.h:12:21: fatal error: LCD.h: No such file or directory

     #include <LCD.h>

I find it a bit hard to work myself into the lib again, it got more complexe.

LuckyLuzz, 2017-11-17:
There is a line missing in the PCF8574 example

lcd.setBacklight(255);

neu-rah, 2017-11-17:
for malpartida's LCD you have to install the libs, url is on top of example i think
about the missing line, a PR would be very welcome

LuckyLuzz, 2017-11-17:
PR?

neu-rah, 2017-11-17:
PR=Pull-Request

LuckyLuzz, 2017-11-17:
PR Yeah will try. So now I got the PCF8574 example running, 16x2 shows the menu, but as I press "Enter" with the rotary encoder or by serial it shows "suspending menu!". The Hex Number thing also doesn't work.

neu-rah, 2017-11-17:
checked example here seem to work ok, serial and encoder

changed something on it?

LuckyLuzz, 2017-11-17:
yeah I am sorry, user related problem... :(   All good with your code ;)

neu-rah, 2017-11-17:
not all good, hex is not working, dunno why
also pad menus broken.. checking last updates

LuckyLuzz, 2017-11-17:
my flault was that I used A3 as Button pin for the rotary encoder and somehow that was leading to ("suspending menu!"). 

As you said the SELECT and EDIT fields doesn't work.

LuckyLuzz, 2017-11-17:
PCF8574 example:
As I uncomment: 
`const panel panels[] MEMMODE={{0,0,16,2}};
navNode* nodes[sizeof(panels)/sizeof(panel)];
panelsList pList(panels,nodes,1);
idx_t tops[MAX_DEPTH];
lcdOut outLCD(&lcd,tops,pList);//output device for LCD
menuOut* outputs[]={&outLCD};//list of output devices
outputsList out(outputs,1);//outputs list with 2 outputs`

and comment:
`MENU_OUTPUTS(out,MAX_DEPTH
  ,LCD_OUT(lcd,{0,0,16,2})
  ,NONE
);
*/`

it causes the Arduino Nano to restart after void setup. So same behavior as with the SSD1306.

neu-rah, 2017-11-17:
it looks that that part of code is outdated... checking

LuckyLuzz, 2017-11-19:
I am gone wait till you fix the macros ^^  You got the skills and I got the time (/basic prog skills) ;)

neu-rah, 2017-11-24:
Sorry for the long delay, been busy on some parallel projects.
yes, that part of code was a test for AVR's on dynamic mode, corrected it to work with progmem.
Pushed to master repo.

LuckyLuzz, 2017-11-25:
Hey,
loaded the updated master. The PCF8574 example  works exept the HEX thing. The SSD1306Ascci Example still reboots the arduino.

`MENU_OUTPUTS(out,MAX_DEPTH
  ,LCD_OUT(lcd,{0,0,16,2})
  ,NONE
);`
What would that line be for the SSD1306 Example?

neu-rah, 2018-01-25:
Hi!
back to this, i've uploaded the SSD1306Ascii example into adafruit nrf52 board using the SSD1306 Oled.

there's a glitch on the last option of the menu (gota check why) but not rebooted yet

what board are you using? arduino nano?

cbpdk, 2018-01-25:
Hi,
I have a SH1106 which is supported by the SSD1306Ascii. Looking for a 'simple' menu system supporting oled + rotary input, I came across ArduinoMenu.

However, I had the above mentioned problem of always rebooting using an Arduino Uno when I tested the SSD1306Ascii example.

After looking through other examples and modifying, I have one solution.
The original piece of code giving problems:
`//describing a menu output device without macros
//define at least one panel for menu output
constMEM panel panels[] MEMMODE={{0,0,128/fontW,64/fontH}};
navNode* nodes[sizeof(panels)/sizeof(panel)];//navNodes to store navigation status
panelsList pList(panels,nodes,1);//a list of panels and nodes
idx_t tops[MAX_DEPTH];//store cursor positions for each level
SSD1306AsciiOut outOLED(&oled,tops,pList);//oled output device menu driver
menuOut* outputs[]={&outOLED};//list of output devices
outputsList out(outputs,1);//outputs list`

And the modified which works:
`/describing a menu output device without macros
//define at least one panel for menu output
const panel panels[] MEMMODE={{0,0,128/fontW,64/fontH}};
navNode* nodes[sizeof(panels)/sizeof(panel)];//navNodes to store navigation status
panelsList pList(panels,nodes,1);//a list of panels and nodes
idx_t tops[MAX_DEPTH]={0,0};//store cursor positions for each level
SSD1306AsciiOut outOLED(&oled,tops,pList);//oled output device menu driver
menuOut* constMEM outputs[] MEMMODE={&outOLED};//list of output devices
outputsList out(outputs,1);//outputs list`

Hope this helps.

cbpdk, 2018-01-26:
Hi again,
I too had a problem with the SSD1306Ascii.ino menu behaving oddly.
After having modified the menuIO/SSD1306Ascii.h to use another oled driver, OzOled, and also modifying the SSD1306Ascii.ino, I noticed the same odd behaviour at the bottom of the display.
I had forgotten to change the font size to 8x8. After doing that, it worked!
Thinking that the 5x7 font used in the SSD1306Ascii.ino might need an extra pixel as padding, I changed line 104:
`#define fontH 7;`
to
`#define fontH 8;`
and no more jumping lines.
I have not investigated if the fontW needs to be changed.

neu-rah, 2018-01-26:
excellent :+1: 
I would be pleased to accept a pull-request with the new driver and corrections if you feel inclined to.

You are welcome to join us at 
https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link

neu-rah, 2018-02-09:
I'm running this example ok on nrf52 with ssd1306 with the fontH and other corrections (thanks @cbpdk), can we close this?

closed 2018-02-09 by neu-rah