neu-rah/ArduinoMenu #136: Display only variable?

jamesarm97, 2017-11-17:
I looked and couldn't figure out how to display a label and a value from a variable without being able to edit it. Basically I am looking for a FIELD option more like LABEL or something that can pass in FIELD("Current Time", curTime, .....) and just show it (and update when it changes) without editing it.

neu-rah, 2017-11-17:
this menu does just that, if your variable is visible (on display) the polling system will check for changes and update the displayed value... if it doesn't then its a bug, because it should.

jamesarm97, 2017-11-17:
Correct (as in I know it can be done), question is, what type do i use that will not be able to be edited? 

As in the MENU() definition below. I see FIELD (editable), EDIT (editable), Submenus and such. I see a TextField eluded to in the code but couldn't find an example.

If you can show me what to use I would appreciate it. Something like:

MENU(mainMenu,"Main menu",zZz,noEvent,wrapStyle
  **,TEXT("Label", variable)**
}

MENU(mainMenu,"Main menu",zZz,noEvent,wrapStyle
  ,OP("Op1",action1,anyEvent)
  ,FIELD(test,"Test","%",0,100,10,1,doNothing,noEvent,wrapStyle)
  ,SUBMENU(subMenu)
  ,SUBMENU(setLed)
  ,OP("LED On",ledOn,enterEvent)
  ,OP("LED Off",ledOff,enterEvent)
  ,SUBMENU(selMenu)
  ,SUBMENU(chooseMenu)
  ,OP("Alert test",doAlert,enterEvent)
  ,EDIT("Hex",buf1,hexNr,doNothing,noEvent,noStyle)
  ,SUBMENU(birthDate)
  ,EXIT("<Back")
);

neu-rah, 2017-11-17:
oh sorry, didn't catch that part. you can set the field to read-only (or disabled)
example on that menu i can disable the toggle field (submenu setLed)
```c++
//on setup()
mainMenu[4].disable();
```

the field will reflect the changes done by the following options "LED On"/"LED Off"

neu-rah, 2017-11-17:
p.s. there's a feature request to make readonly => like disabled but showing with normal color

neu-rah, 2017-11-17:
Completely agree, there's also a request for that, will try to reach them on next release.
But thanks for your interest and suggestions. They are welcome.

jamesarm97, 2017-11-17:
Sorry to ask here (no forum), is there a way to output text via code that goes to all the configured outputs? Say I call an action from a menu and want to output some data to the LCD and Serial. Currently I am doing it directly using my lcd.write() and consoleSerial.write. Is there a menu.write or similar that would take the string and send it to all configured output devices?

closed 2018-02-09 by neu-rah