neu-rah/ArduinoMenu #137: Update issues

AlexFundorin, 2017-11-26:
Hi. Can't tell exactly, which was the last working version for me, but I've decided to modify my sketch a little and updated all libraries. This (and other errors, related to this) is what I get when trying to compile the sketch: https://i.imgur.com/HIBOIMJ.png
Please, tell me, what should be changed in the code to fix this issue?

P.S. Rolled back to 3.1.12 and the code comped fine.

neu-rah, 2017-11-26:
hi!
on version 4.x
https://github.com/neu-rah/ArduinoMenu/blob/master/examples/Serial/serialio/serialio/serialio.ino#L77

fell free to ask here if you encounter more obstacles or join us at https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link

neu-rah, 2017-11-30:
solved?

AlexFundorin, 2017-12-01:
Kinda. I think that it might be a good idea to add a workaround to the FAQ, in case, if someone else would have the same issues.

closed 2017-12-01 by AlexFundorin