neu-rah/ArduinoMenu #139: Some problems when editing a text string.

ur7hfo, 2017-12-04:
Good afternoon. I tried to use your library ArduinoMenu.

Some problems when editing a text string.
Everything works, but during editing, the cursor _ does not move to edit the next character.

I used an example from your documentation.

char * const hexDigit PROGMEM = "0123456789ABCDEF"; // allowed hexadecimal digits
char * const hexNr [] PROGMEM = {"0", "x", hexDigit, hexDigit}; // validators
char buf1 [] = "0x11"; // the edit target buffer
...
EDIT ("Hex", buf1, hexNr, doNothing, noEvent, noStyle)
 
Thank you.

neu-rah, 2017-12-04:
Hi! what is your menu output? Serial? Lcd? ...

ur7hfo, 2017-12-04:
LСD 20x4 with using the library  https://github.com/mathertel/LiquidCrystal_PCF8574

daniel-leonard-robinson, 2017-12-04:
Hi I tried this, too. 4.11 was a little bit better than 4.12.

Output device Serial works fine, but LiquidCrystal does not move, just like @ur7hfo using `LiquidCrystal_PCF8574`. 

I thought I was doing something wrong my side, but the `idx_t x`, and `idx_t y` cursor positions, or `l` and `line`, they do not move. They are stuck at `8` and `0`, respectively.

daniel-leonard-robinson, 2017-12-04:
Sorry, output device Serial also doesn't work in 4.12

neu-rah, 2017-12-05:
Hi! published v4.0.13 to address this, hope its solved, please let me know if not.
And yes last fix on 4.0.12 broken something on the textField operation, thanks @daniel-leonard-robinson for the clue ;)

daniel-leonard-robinson, 2017-12-05:
Awesome, it works perfectly. Thanks for the quick response @neu-rah 👍

closed 2017-12-05 by neu-rah

ur7hfo, 2017-12-05:
Editing the string works fine in the new version of the library. Thank you very much, for your work! Best wishes! Thank you !