neu-rah/ArduinoMenu #141: Bug: code freezes when putting numeral 0x30 -> 0x39 in `FIELD`.

daniel-leonard-robinson, 2017-12-06:
Hi Rui,

As far as I remember this used to work? This is for `LiquidCrystal` and `Serial` as output devices. When editing a `FIELD`, the code freezes and requires a `reset`.

Version 4.13

neu-rah, 2017-12-06:
checking...

neu-rah, 2017-12-06:
Hi Daniel,
finally got a breadboard with LCD HD44780 LCD wired, running the example... hex edits ok.
what is that character code detail?
can you show field initialization code?

daniel-leonard-robinson, 2017-12-07:
Hi Rui,

Sure. These are snippets, but should give you the general idea. The line of interest is:

`FIELD(motor.motor_speed,"Speed","",0,9,1,0,setSpeed,noEvent,wrapStyle)`

It works when using up and down keys, or '+' and '-', but not when entering a number. Hmm, seems to be my side then, if it works for you. Thanks for checking.

```
MENU(manualControl_submenu,"Manual Control",showEvent,anyEvent,noStyle
  ,SUBMENU(chooseMotorMenu)
  ,SUBMENU(enableMotors)
  ,FIELD(motor.motor_speed,"Speed","",0,9,1,0,setSpeed,noEvent,wrapStyle)
  ,SUBMENU(setClockwise)
  ,EXIT("<Back")
);

MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  ,SUBMENU(manualControl_submenu)
  ,SUBMENU(events_submenu)
  ,EXIT("<Back")
);

MENU_OUTPUTS(out, 3
  ,LIQUIDCRYSTAL_OUT(lcd,{0,0,20,4})
  ,SERIAL_OUT(Serial)
  ,NONE
);
NAVROOT(nav,mainMenu,MAX_DEPTH,in,out);
```

In setup()
```
lcd.begin(20,4);
nav.idleTask=idle;
nav.showTitle=false;
```
In loop()
```
if (!enter_bt.check()) nav.doNav(enterCmd);
else if (!back_bt.check()) nav.doNav(escCmd);
else if (!up_bt.check()) nav.doNav(upCmd);
else if (!down_bt.check()) nav.doNav(downCmd);
nav.poll();
```

neu-rah, 2017-12-07:
wait, i was testing other thing, your code shows much better the problem, i will try it

neu-rah, 2017-12-07:
hi!

i'm using the LiquidCrystal example (not the PCF8574, if not please tell me, because it can be example specific) as a base, glued-in your code with minimal changes, just cut out non-relevant parts.

made it work with following code:

```c++
struct Motor {
  int motor_speed;
} motor;

//here showEvent,anyEvent was being used... changed to doNothing, noEvent
MENU(manualControl_submenu,"Manual Control",doNothing,noEvent,noStyle
  // ,SUBMENU(chooseMotorMenu)
  // ,SUBMENU(enableMotors)
  // here ,setSpeed,noEvent -> setSpeed wont be called because mask is noEvent (not the cause of problem)
  ,FIELD(motor.motor_speed,"Speed","",0,9,1,0,doNothing,noEvent,wrapStyle)
  // ,SUBMENU(setClockwise)
  ,EXIT("<Back")
);

MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  ,SUBMENU(manualControl_submenu)
  // ,SUBMENU(events_submenu)
  ,EXIT("<Back")
);

#define MAX_DEPTH 2

MENU_OUTPUTS(out, 3
  ,LIQUIDCRYSTAL_OUT(lcd,{0,0,16,2})
  ,SERIAL_OUT(Serial)
  ,NONE
);
NAVROOT(nav,mainMenu,MAX_DEPTH,in,out);

void setup() {
  pinMode(encBtn,INPUT_PULLUP);
  pinMode(LEDPIN,OUTPUT);
  Serial.begin(115200);
  while(!Serial);
  Serial.println("Arduino Menu Library");Serial.flush();
  encoder.begin();
  lcd.begin(16,2);
  nav.showTitle=false;
  lcd.setCursor(0, 0);
  lcd.print("Menu 4.x LCD");
  lcd.setCursor(0, 1);
  lcd.print("r-site.net");
  delay(2000);
}

void loop() {
  nav.poll();
  delay(100);//simulate a delay as if other tasks are running
}
```

please note the handler functions and events/masks changes

also i'm not using the key handling schema on loop. You can let the menu do that by declaring a keyboard, much like the definition of an encoder button on the encoders examples... please let me know if you need help defining a keyboard, it associates buttons/pins with menu commands.

awaiting feedback

daniel-leonard-robinson, 2017-12-07:
It's weird, I tried this code, and the numbers do update after about 1 second. But my full code just makes it freeze. Does this code update instantly for you?

```
#include <menu.h>
#include <menuIO/liquidCrystalOut.h>
#include <menuIO/serialOut.h>
#include <menuIO/serialIn.h>
#include <menuIO/keyIn.h>
#include <menuIO/chainStream.h>
using namespace Menu;

#include <LiquidCrystal.h>

LiquidCrystal lcd(44, 45, 46, 47, 48, 49);

struct Motor {
  int motor_speed;
} motor;

serialIn serial(Serial);
menuIn* inputsList[]={&serial}; // more inputs can be added here.
chainStream<1> in(inputsList); // 1 is the current number of inputs

//here showEvent,anyEvent was being used... changed to doNothing, noEvent
MENU(manualControl_submenu,"Manual Control",doNothing,noEvent,noStyle
  // ,SUBMENU(chooseMotorMenu)
  // ,SUBMENU(enableMotors)
  // here ,setSpeed,noEvent -> setSpeed wont be called because mask is noEvent (not the cause of problem)
  ,FIELD(motor.motor_speed,"Speed","",0,9,1,0,doNothing,noEvent,wrapStyle)
  // ,SUBMENU(setClockwise)
  ,EXIT("<Back")
);

MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  ,SUBMENU(manualControl_submenu)
  // ,SUBMENU(events_submenu)
  ,EXIT("<Back")
);

#define MAX_DEPTH 2

MENU_OUTPUTS(out, 3
  ,LIQUIDCRYSTAL_OUT(lcd,{0,0,16,2})
  ,SERIAL_OUT(Serial)
  ,NONE
);

NAVROOT(nav,mainMenu,MAX_DEPTH,in,out);

void setup() {
//  pinMode(encBtn,INPUT_PULLUP);
//  pinMode(LEDPIN,OUTPUT);
  Serial.begin(115200);
  while(!Serial);
  Serial.println("Arduino Menu Library");Serial.flush();
//  encoder.begin();
  lcd.begin(16,2);
  nav.showTitle=false;
  lcd.setCursor(0, 0);
  lcd.print("Menu 4.x LCD");
  lcd.setCursor(0, 1);
  lcd.print("r-site.net");
  delay(2000);
}

void loop() {
  nav.poll();
  delay(100);//simulate a delay as if other tasks are running
}
```

neu-rah, 2017-12-08:
no, not instantly
my monitor console does not need enter to send characters, it is sending as i'm typing
some consoles are like that.
checked on both arduino and platformio monitor console.
also arduino console has the option for not sending the enter, so menu code waits, if no more numbers then it converts the value. just because we cant relly on an enter terminating the string :\

daniel-leonard-robinson, 2017-12-08:
Thanks Rui,

I'm gonna close it. Really appreciate your help. I think the library is fine; just something I'm doing wrong, which I will find out in future.

closed 2017-12-08 by daniel-leonard-robinson