neu-rah/ArduinoMenu #143: Serial on Arduino M0

MajorMadness, 2017-12-13:
Hello, I try to compile the Test Examples on Arduino M0 with just the Native USB port. Because of this I have to use Serial USB. As sugestet on I wrote . This leads to the Error:
error: no matching function for call to 'Menu::serialIn::serialIn(Serial_&)'
 serialIn serial(Serial);

Full Error Log:
`
SSD1306Ascii:115: error: no matching function for call to 'Menu::serialIn::serialIn(Serial_&)'
 serialIn serial(Serial);
                       ^
C:\...\AppData\Local\Temp\arduino_modified_sketch_855949\SSD1306Ascii.ino:115:23: note: candidates are:
In file included from C:\...\AppData\Local\Temp\arduino_modified_sketch_855949\SSD1306Ascii.ino:12:0:
C:\...\arduino-1.8.5\libraries\ArduinoMenu_library\src/menuIO/serialIn.h:23:9: note: Menu::serialIn::serialIn(HardwareSerial&)
         serialIn(SERIAL_CLASS& serial):in(serial) {}
         ^
C:\...\arduino-1.8.5\libraries\ArduinoMenu_library\src/menuIO/serialIn.h:23:9: note:   no known conversion for argument 1 from 'Serial_' to 'HardwareSerial&'
C:\...\arduino-1.8.5\libraries\ArduinoMenu_library\src/menuIO/serialIn.h:20:11: note: constexpr Menu::serialIn::serialIn(const Menu::serialIn&)
     class serialIn:public menuIn {
           ^
C:\...\arduino-1.8.5\libraries\ArduinoMenu_library\src/menuIO/serialIn.h:20:11: note:   no known conversion for argument 1 from 'Serial_' to 'const Menu::serialIn&'
C:\...\arduino-1.8.5\libraries\ArduinoMenu_library\src/menuIO/serialIn.h:20:11: note: constexpr Menu::serialIn::serialIn(Menu::serialIn&&)
C:\...\arduino-1.8.5\libraries\ArduinoMenu_library\src/menuIO/serialIn.h:20:11: note:   no known conversion for argument 1 from 'Serial_' to 'Menu::serialIn&&'
Bibliothek SSD1306Ascii in Version 2015.7.16 im Ordner: C:\...\arduino-1.8.5\libraries\SSD1306Ascii  wird verwendet
Bibliothek Wire in Version 1.0 im Ordner: C:\...\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.17\libraries\Wire  wird verwendet
Bibliothek ArduinoMenu_library in Version 4.0.13 im Ordner: C:\...\arduino-1.8.5\libraries\ArduinoMenu_library  wird verwendet
exit status 1
no matching function for call to 'Menu::serialIn::serialIn(Serial_&)'
`

neu-rah, 2017-12-13:
hi!
you can try other port (like Serial1) or use the menu without serial output if you have another output device

MajorMadness, 2017-12-13:
Hi Rui,
Yes this is no problem for me, because I never use Serial besides
debugging. I just found this and wanted to let you know SamD21 native USB
is not working with the libary.

Best regards Moritz

neu-rah, 2017-12-13:
Ok, thank you.

MajorMadness, 2017-12-13:
Hi, I have one more thing: Can you please try to compile reuse example? On
Mega2560 and M0 I get:
reuse:117: error: 'Menu::result AlarmPrompt::eventHandler(Menu::eventMask,
Menu::navNode&, Menu::idx_t)' marked override, but does not override
     result eventHandler(eventMask e,navNode& nav,idx_t i) override;
            ^
Bibliothek ArduinoMenu_library in Version 4.0.13 im Ordner:
C:\Users\Stapelfeldt\Dropbox\arduino-1.8.5\libraries\ArduinoMenu_library
wird verwendet
exit status 1
'Menu::result AlarmPrompt::eventHandler(Menu::eventMask, Menu::navNode&,
Menu::idx_t)' marked override, but does not override

Call is "result eventHandler(eventMask e,navNode& nav,idx_t i) override;"

Im looking for a way to set 16 settings for Led (0-16) with time and PWM
(10:00: 50%, 12:00: 100% and so on), that's why I'm looking at reuse
example.

Greeetings Moritz

neu-rah, 2017-12-13:
Hi, I will give it a try next week i think, this sounds a lot like a bug...
on the m0, i'm investigating the Serial stuff
thanks a lot for your feedback, it is very appreciated.
and you are welcome to join us on gitter :D

MajorMadness, 2017-12-13:
I already did. ;) I'm working as software developer too and digging into your lib atm. ;) It's very complex and I slowly understand your way you handle stuff. Specially the extend and overwrite on prompt fucks my brain last night and today...

MajorMadness, 2017-12-18:
Hello Rui,
I've a question but it's to much for gitter, so I hope it's ok to ask you
by mail.
I'm working on the Menu but can get it to work the way I want it. First
menu should shou "Led 1" to "Led 8" an 2nd Menu "Time 1" to "Time 8" I
thought I can rename the Items in a loop by .data[i]->text but it's not
working and sketch crashes. Maybe you can help me to:
1. Rename Led Menu so I have just one sub menu
2. On Enter save index of selected item
3. Rename Time Menu so I just have one "set"-Page
4. Save index of time again
Here is the code I have atm:

byte ledValue,ledHour,ledMinute;
PADMENU(setLedTime,"",doNothing,noEvent,noStyle
   ,FIELD(ledHour,"",":",0,23,1,0,doNothing,noEvent,noStyle)
  ,FIELD(ledMinute,"","",0,59,10,1,doNothing,noEvent,wrapStyle)
);
MENU(setLights,"Zeit",doNothing,noEvent,wrapStyle
  ,SUBMENU(setLedTime)
  ,FIELD(ledValue,"","%",0,100,1,0,doNothing,noEvent,noStyle)
  ,EXIT("<Back")
);
result setLedMenu(eventMask e,navNode& nav,prompt& item);
result setTimeMenu(eventMask e,navNode& nav,prompt& item);

MENU(setLedNum,"Led",setTimeMenu,anyEvent,noStyle
  ,SUBMENU(setLights)
  ,SUBMENU(setLights)
  ,SUBMENU(setLights)
  ,SUBMENU(setLights)
  ,SUBMENU(setLights)
  ,SUBMENU(setLights)
  ,SUBMENU(setLights)
  ,SUBMENU(setLights)
  ,EXIT("<Back")
);
MENU(ledMenu,"Leds",setLedMenu,anyEvent,noStyle
  ,SUBMENU(setLedNum)
  ,SUBMENU(setLedNum)
  ,SUBMENU(setLedNum)
  ,SUBMENU(setLedNum)
  ,SUBMENU(setLedNum)
  ,SUBMENU(setLedNum)
  ,SUBMENU(setLedNum)
  ,SUBMENU(setLedNum)
  ,EXIT("<Back")
);

result setLedMenu(eventMask e,navNode& nav,prompt& item){
  for(byte i = 0; i<8;i++){
    setLedNum.data[i]->text="Led " + i;
  }
  return proceed;
}

result setTimeMenu(eventMask e,navNode& nav,prompt& item){
  for(byte i = 0; i<8;i++){
    setLights.data[i]->text="Time " + i;
  }
  return proceed;
}

MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  ,SUBMENU(ledMenu)
  ,FIELD(standbyTime,"Display
aus","Min",0,10,1,0,doNothing,noEvent,wrapStyle)
  ,OP("SaveSettings",saveSetting,enterEvent)
  ,EXIT("<Back")
);

Thank you a lot for your support!
Best Regarts Moritz

2017-12-13 16:20 GMT+01:00 Moritz Stapelfeldt <mstravendev@gmail.com>:

neu-rah, 2017-12-18:
Hi!
I've not looking into this in depth (but I will), however from the beginning i can say that this is a static menu by default (see bottom not for dynamic), all its structure id statically defined and stored on MCU flash (on AVR's), so, renaming on a (runtime) loop wont work. However you can reuse parts and have variable fields for text and numeric values (either editable or enumerable).

eventually you can make dynamic runtime changes on non AVR MCU, just don't know the details of M0, if they can store on flash or not. For MCU's that use ram as arduino PROGMEM it works on runtime.

The dynamic example is about that.

you can force AVR's to use ram and do runtime changes (not if using arduino IDE) by setting  USE_RAM as a compiler command line option (-DUSE_RAM)

neu-rah, 2017-12-18:
pushed v4.0.15 to master branch to solve this issue (on title)
please open a new about memory or reopen this if Serial on M0 not solved
thanks for your feedback

closed 2017-12-18 by neu-rah