neu-rah/ArduinoMenu #144: Display dynamic values

T94T, 2017-12-16:
Hi,
I want to make an "about" submenu that looks like this:

> IP: 127.0.0.1
> SW: Version 1.0
> <Back

Is there a way to inject the IP address and the SW version dynamically? I didn't find any solution.
First and second menu items do not need to be updateable by the mneu.
```
subMenu[0].disable();
subMenu[1].disable();
```

neu-rah, 2017-12-16:
Hi!
yes, you can use fields (text or numeric) and eventually make them read-only for the user.
code can still change the values and menu system will automatically update changed values.

T94T, 2017-12-16:
Thank you very much for the fast response!
Do you have some simple code snippet for a text field? TextField.ino seems to me not to be the right thing.

neu-rah, 2017-12-16:
most examples include text and numeric fields, text field will need those extra definitions of allowed characters, eventually can be empty, but I've not checked the empty case... let me know if you have problems

and you are welcome :D

T94T, 2017-12-18:
So here is what is working for me:
```
const char* constMEM dummyMask[] MEMMODE={""};
char ipAddress[] = "000.000.000";
```
...
```
MENU(subMenu, "About...", doNothing, noEvent, wrapStyle
 ,EDIT("IP:", ipAddress, dummyMask, doNothing, noEvent, noStyle)
```
And to update the value I just call
```
strcpy(ipAddress, event.ip.toString().c_str());
subMenu.dirty = true;
```
dirty lets the menu update in live view.

neu-rah, 2017-12-18:
thank you for sharing your solution, you are welcome to join us at https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link

closed 2017-12-18 by T94T