neu-rah/ArduinoMenu #146: Use without MENU_INPUTS?

BennehBoy, 2017-12-20:
Hi, I just updated to V4 (think I was using V3 previously) and notice that all the code examples now use a MENU_INPUTS line.

Prior to this I was handling all input like this:
```

	if (inMenu) {
		if (currentMillis - menuMillis > BUT_DELAY) {
			menuMillis = currentMillis;
			if (btn_up.sense() == buttons_debounce) {
				nav.doNav(upCmd); // navigate up
				inptimeoutMillis = currentMillis;
			} else if (btn_down.sense() == buttons_debounce) {
				nav.doNav(downCmd); // navigate down
				inptimeoutMillis = currentMillis;
			} else if (btn_enter.sense() == buttons_debounce) {
				nav.doNav(enterCmd); // do current command
				inptimeoutMillis = currentMillis;
			}
			nav.active().dirty=true;//for a menu
			nav.navFocus->dirty=true;//should invalidate also full screen fields assert(nav.navFocus!=NULL)
			nav.poll();//do serial input
			display1.display();
			display1.clearDisplay();	
		}
	}
```

Now my code fails to compile, complaining about the Serial device (which I don't want to use anyway).  It's just luck that it worked previously.

What do I need to change to make things work, I already am using a buttons library to catch input so was hoping there would be some form of null handler - I need to catch the button presses myself because I'm using a long press to turn off the menus and the buttons are controlling other screens at the same time.

neu-rah, 2017-12-20:
Hi!

`poll` function will do both input and output, if you do not want to use the menu input you can call only `doOutput` instead of `poll`.

yes, on version 4 the menu system is not using Serial or Streams directly, but uses a `serialIn` wrapper object instead, ex:

```c++
serialIn serial(Serial);
NAVROOT(nav,mainMenu,MAX_DEPTH,serial,out);
```

however that object requires a stream, there are some null streams for arduino, but the old Serial will also do because you wont use it anyway

please let me know if this does not work or if you have any other issue

BennehBoy, 2017-12-21:
Thanks, I'll try doOutput instead of poll - I did try wrapping Serial and had further issues but that may be due to the expected stream being null when Input was being processed.

BennehBoy, 2017-12-22:
Great, this works.

Would you consider providing a null input IO handler which simply allows a null stream to be passed to NAVROOT?  Might save a tiny amount of flash usage.

Thanks,

Ben

neu-rah, 2017-12-22:
glad this solution works for you
it's an excellent idea, a deep revision of IO is scheduled for next version.
thanks a lot for your feed back :D

you are welcome to join us at https://gitter.im/ArduinoMenu/Lobby

neu-rah, 2018-02-28:
let me know if it behaves bad :)

closed 2018-02-28 by neu-rah