neu-rah/ArduinoMenu #147: How to simply get a menu implemented?

Jacotheron, 2017-12-22:
Hi. Firstly I want to say that this menu library seems very powerful and seems perfect for my needs (although it may be a little overkill).

Having read through the Wiki and other issues (both closed and open), I was unable to find the information I need:
What I have:

1. Adafruit Feather M0 (uses the same processor as the Arduino M0 aka Arduino Zero, so lots of space and memory; also have interrupts available on every single io pin, possibly making things easier).
1. 128x64 OLED display, which I want to run via U8g2 library. Later this display will be exchanged for an eInk 200x200 display, but for now I have a display using the same pins (SPI + additional wires) as the final one (U8g2 supports it so it should be fine).
1. A 5-way button joystick (so it have 5 buttons internally, in stead of potentiometers; for click, up, down, left and right); each button is connected to its own IO pin (yes, I am using 5 of my 20 available pins for this).

What I need to know is, how do I start?
Obviously I need to create the constructor for the U8g2 (using the appropriate pins), but how do I make the menu use my joystick as an input (letting it know which pins to use)?
Should I initialize the U8g2 in my setup?

Having been through the wiki, I think I can define the menu, as well as use the fields to change the values (my code is very dependent on these global variables, since I am building it quite modular with basically as close to multi-threading as I can get it).

Another question I have:

1. It is possible to have the menu timeout and exit the menu (to return to the home/idling screen) after a set amount of time? Just ensure that the project does not remain in the menu indefinitely.

I have looked through the U8G2 example that came with the library, and it explains my other questions I had.

neu-rah, 2017-12-22:
Hi!

Thank you  for your interest on this library.
The is at least one example using U8g2, the menu does not need to know the oled pins as it is a matter for the u8g2. The menu will however need to know about the u8g2 object. That is what the is done on the u8g2 example.
The same for all other devices, so this is a library that works with other libraries. For some simple devices we have included some simple drivers. That is the case of buttons, encoders and analog joysticks.
There is an example using the analog joystick, and in that case you must provide the pins to the driver and latter provide the driver to the menu as an input device.
This menu can use multiple inputs and multiple outputs in parallel, and most examples work with Serial IO along with the other hardware.
The is an example that works exclusively with serial device, it is a good example to check compatibility with your mcu as it only requires a serial port.

I've used the menu on a arm m0 (nRF52)

For digital joystick you can either define a keyboard with multiple buttons, the current encoder example shows how to setup a single button, has the encoder only has one button, but its a matter of replicating the button setup and associate each one with a different menu command.

You can also use the menu in a code driven navigation reading your own hardware and issuing menu commands for navigation.

If you have any trouble or doubts do not hesitate to ask here or on the gitter channel: https://gitter.im/ArduinoMenu/Lobby

have you tried the serialio example?
https://github.com/neu-rah/ArduinoMenu/tree/master/examples/Serial/serialio

neu-rah, 2017-12-23:
about menu timeout, yes
by several methods, techniques on wiki about suspending menu:
- by not calling menu poll or IO functions whenever you reach a timeout on some external code
- by setting, menu suspended flags again from external timeout

the resume method is related to the method used to suspend the menu, a menu flagged to be in suspended state will ignore input even when poll or io functions called **until enter/select is pressed**, it then auto-resumes.

Jacotheron, 2017-12-23:
Hi, thank you for the quick reply.

I now have a menu displaying, however my inputs are not working. I checked my wiring to ensure that the joystick is connected correctly. The library also recognize all the outputs, since they are getting pulled HIGH (joystick common is connected to ground).
The library also reads the pins, since it will exit the menu and display the splash screen when I press the joystick to give any input for more than 2 seconds (as per the default exit settings).

However the issue I am having is that none of the standard actions (select, up, down, left and right) are registering and changing the menu.

I am including some of my relevant code:

`keyMap joystickBtn_map[]={ {-Joystick_Click, options->getCmdChar(enterCmd)} , {-Joystick_A, options->getCmdChar(upCmd)} , {-Joystick_C, options->getCmdChar(downCmd)}  , {-Joystick_B, options->getCmdChar(leftCmd)} , {-Joystick_D, options->getCmdChar(rightCmd)} };
keyIn<5> joystickBtns(joystickBtn_map);`

`NAVROOT(nav, mainMenu, MENU_MAX_DEPTH, joystickBtns, menu_outputs);`

My Loop is then:
`nav.doInput();
  if(nav.sleepTask){
    Gui_home_screen();
  }else{
    if(nav.changed(0)){
      u8g2.firstPage();
      do nav.doOutput(); while(u8g2.nextPage());
    }
  }`

The markup did not like my line breaks, sorry about that.
Obviously the Joystick_* values are constants defined earlier in my code, for the pins used, and they have the "-" in front of them, to indicate a pull-up needed.

Regarding the timeout, I guess I should have every menu action call a function that will reset the timer, and if it runs out, set the menu to be suspended.

neu-rah, 2017-12-23:
Hi, excellent job
you are correct about the timeout and also about your input construction, and having trouble with that looks like a bug to me, so I will investigate your case.
Meanwhile the codeCtrl example might be of some help to you.

neu-rah, 2017-12-23:
confirmed bug causing wrong code to be returned from `getCmdChar`

you can use characters as workaround
'*' enterCmd
'+' upCmd
'-' downCmd

left/right are using the same codes as down/up

note: up/down are from array index point of view and that happens to be ok for encoders, keys might look reversed, if so just swap them

Jacotheron, 2017-12-23:
Thank you very much for assisting me on a Saturday so close to Christmas, or whatever you call these holidays in your country. 

I was trying to implement the codeCtrl example just as this one came in, so I quickly implemented the change and it works.

Now just have to fine tune my menus.

neu-rah, 2017-12-23:
:D glad it solved, also thanks for your feedback, it allowed me to find that lurking bug... will fix it soon.

you are welcome to join us at https://gitter.im/ArduinoMenu/Lobby

Jacotheron, 2017-12-23:
Being almost done with the menu, I found something that does not make sense.
I am using the FIELD to create a range selector (for percentage, used to dim a light):

`FIELD(zone_signal_spotlight, "Spotlight Dim", "%", 0, 100, 1, 0, zoneUpdated, enterEvent, noStyle)`

So the variable "zone_signal_spotlight" handles a percentage value, which is used for an analogWrite (on my DAC pin, connected to a driver that handles the dimming); this variable's type was a uint8_t, but in debugging this I have changed it to int, unsigned int and others (currently at uint16_t).

The field is rendered, and I can select it, however I am unable to get it to move to a value between 0 and 100%. In stead of remaining within the limits, it overflows to "55667", which then gets converted to 100%.
So the issue is that it does not remain within the bounds. The callback function "zoneUpdated" simply set a bolean value which will, on the next loop, force all zones to have their brightness adjusted, the returns the "proceed" result.

I have used this also for another part of the menu, and this works (it remains within the bounds; this variable type is uint16_t):
`FIELD(sleeptimer_duration_s, "Duration", "s", 10, 900, 10, 0, doNothing, noEvent, noStyle)`

Having added the working one next to the not working one (only changing the variable to use and its bounds), and it again does not work. Can you see something that is wrong?

neu-rah, 2017-12-23:
:D

you should open a new issue for the sake of future readers...
have you tried to make the variable `volatile`?

neu-rah, 2017-12-24:
this was related to C++ initialization order problem, so we can not call that function on startup.

should be ok when replacing this:
```c++
 options->getCmdChar(enterCmd)
```
by this:
```c++
 defaultNavCodes[enterCmd].ch
```

closed 2017-12-24 by neu-rah