neu-rah/ArduinoMenu #148: Using KeyPad with Menu

Ken-Fitz, 2018-01-12:
I'm trying to tie a 4x5 keypad to the menu, I known not all buttons can be mapped, I use a subset of them, or at least I did in v3 of the menu.

I'm trying to adapt to v4 and cant seem to get the keypad to sit in the menu inputs.
Code for the keypad is at http://playground.arduino.cc/Code/Keypad

Could you nudge this noob in the right direction?

My V3 setup uses
// Create the Keypad
Keypad kpd = Keypad( makeKeymap(keys), rowPins, colPins, ROWS, COLS );

// Input Lists
Stream* inputsList[]={NULL};
chainStream<0> in(inputsList);

And I drive the menu using  nav.doNav(enterCmd);  type commands.

neu-rah, 2018-01-12:
Hi!

can you provide more details? this seems to compile... so what it is not doing and expected to do?

you are also welcome to join us at https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link

neu-rah, 2018-01-12:
ops, closed by accident

Ken-Fitz, 2018-01-12:
That code is fine, I wanted to add the keypad into the MENU_INPUTS function rather than drive it from the outside. When I use 
Stream* inputsList[]={kpd}     or &kpd it fails to cast a char to a stream, tried using kpd.getKey() but that also fails.

I suppose the question is with that keypad library, how do I hook it into your menu system (MENU_INPUTS).

neu-rah, 2018-01-12:
probably not, by looking at the examples it sems that there not an unique pin for each key, if so then a derived class from `menuIn` must be created and those key reading got to be turned into stream characters to command the menu.

i can help with this if its the case

Ken-Fitz, 2018-01-13:
the keypad function does not use a single pin for each button, the 5x4 keypad uses 9 pins, as rows and columns.
It can cope with all sorts of keypad sizes and would probably make a good addition to the menu function, allowing direct number entry into a field.

Any help gratefully received neu-rah.

neu-rah, 2018-01-13:
then there must be something similar to this driver

https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuIO/PCF8574KeyIn.h

Ken-Fitz, 2018-01-13:
Yes I can test it. They keypad is this one: https://www.ebay.co.uk/itm/2-New-4x5-Matrix-Extension-Keyboard-20-Key-Membrane-Switch-Keypad-For-Arduino/201552347294?hash=item2eed74c09e:g:yWcAAOSwDk5UDsS0

By 'no hardware' do you mean no keypad? 
If you live in the UK, I'll post you one, I have a spare.

neu-rah, 2018-01-13:
thank you i will try to do a driver for it then :D :+1: 

yea, no keypad
and nop, i live in Azores Islands :D
but i think this will be ok, from what i've read it looks pretty simple

Ken-Fitz, 2018-01-13:
The Azores... wow.. <thinks> May be I could deliver it by hand.. lol.

neu-rah, 2018-01-13:
hehehe
guess you swim a lot then :D

neu-rah, 2018-01-31:
Hi!
i've pushed to the repository a new version that includes experimental keypad driver (first version) and example https://github.com/neu-rah/ArduinoMenu/blob/master/examples/keypad/keypad/keypad.ino

please report errors or doubts.

thanks.

Ken-Fitz, 2018-02-01:
Thanks neu-rah, I'm currently away from home until the weekend. But i'll test it and report back ASAP. 
After reading the code it looks spot on.

neu-rah, 2018-02-01:
I would be surprised if it worked on first attempt... :D

neu-rah, 2018-02-01:
already pushed a correction :D
== when i meant just =

Ken-Fitz, 2018-02-02:
Code not working I'm afraid.
My code is:
[KeyPad5x4Menu.txt](https://github.com/neu-rah/ArduinoMenu/files/1689270/KeyPad5x4Menu.txt)
And the header file i'm using is:
[keypadIn.txt](https://github.com/neu-rah/ArduinoMenu/files/1689278/keypadIn.txt)
I also added a line into keypadin.h to return the key it returns, it always returns a key (a backward question mark) even with no key input.

I've tested the pin assignments with a working code and it outputs the chars.
I also changes 4 of the chars to match your example + - / and * - nothing.

neu-rah, 2018-02-02:
my bad logic...
just by looking at it (again) i can see that:
```c++
int available(void) {return peek()!=0;}
```
wont work, please try with -1 instead of that 0

neu-rah, 2018-02-09:
hi
-1 instead of 0 done any good?

Ken-Fitz, 2018-02-10:
1st I apologise, been working away a lot so not had as much time on this as i'd like
The -1 instead of zero has not done any good.
The code is now
      int peek(void) {return key?key:(key=in.getKey()?key:-1);}
      int read() {
        if (key) {
          char k=key;
		  Serial.println((int)key);
          key=0;
          return k;
        }
        return peek();
      }

It seems the no key value is -1 (which is seen true), so I altered if (key) to if (key != -1) it then prints nothing regardless of what key I press, what key values mean up/down/enter/escape by the way.
I think the issue must lie in the key read area of the code. in.getkey.
I've attached the source, [KeyPad5x4Menu.txt](https://github.com/neu-rah/ArduinoMenu/files/1713563/KeyPad5x4Menu.txt)
The following source works with the keypad, uses the same pins so should eliminate a wiring issue.
[_5x4KeyboardTest.txt](https://github.com/neu-rah/ArduinoMenu/files/1713568/_5x4KeyboardTest.txt)
The only other variable I can see if the keypad.h, the one i'm using according to the library is from here: http://playground.arduino.cc/Code/Keypad

Hope that helps.

neu-rah, 2018-02-10:
Hi!
no problem, we all have something else to do, that's why my code is so rushed... :D
have a matrix keyboard on its way and i have enough buttons here to simulate one (2x2) to be minimal...
the only thing missing is time... having that i will straighten this up.
Thanks for your test patience and for the code.

neu-rah, 2018-05-22:
Hi!
finally got keypad and some time to test it, needless to say that my driver attempt without the hardware was  a complete disaster, as you noticed for sure.

so, I finally finished the driver and published it with an example for a 4x4 keypad matrix.

please let me know if you encounter some problems as this is a fresh new driver.

thanks for you patience

closed 2018-05-22 by neu-rah