neu-rah/ArduinoMenu #150: Display Field in Menu

Ken-Fitz, 2018-01-29:
More of a request/enhancement than an issue.
Would it be possible to have a menu entry that just displays data and changes based on external changes to that data.
Example:
         Set Speed 60                        <= Standard field setting to target speed
         Speed 25 Accel 5                 <= Output field showing the current speed and acceleration

The output could just be a text string set by the app from internal data values.

neu-rah, 2018-01-29:
it is already possible, all fields will still reflect the changes to the original data and refresh.
and you can set the field as read-only to prevent user iteration.

ex:
```c++
mainMenu[0].enable();
mainMenu[1].disable();
```
let me know if you have any doubt

Ken-Fitz, 2018-01-29:
Thanks neu-rah. That should have occurred to me. Still learning.

closed 2018-01-29 by Ken-Fitz