neu-rah/ArduinoMenu #153: Indexed values in menu, compilation fails.

cbpdk, 2018-02-12:
After updating the Arduino Menu Library via the ide, the compiler output window went from almost white to almost red.
The new version being 4.1.7.

The problem started with these error lines:

In file included from /home/cbp/lcdsketchbook/libraries/ArduinoMenu_library/src/menu.h:19:0,
                 from /home/cbp/work/arduino/menu/test/test.ino:4:
/home/cbp/lcdsketchbook/libraries/ArduinoMenu_library/src/shadows.h: In instantiation of 'struct Menu::menuFieldShadowRaw<double&>':
/home/cbp/work/arduino/menu/test/test.ino:45:1:   required from here
/home/cbp/lcdsketchbook/libraries/ArduinoMenu_library/src/shadows.h:113:10: error: forming pointer to reference type 'double&'
       T* value;
          ^
In file included from /home/cbp/lcdsketchbook/libraries/ArduinoMenu_library/src/menuBase.h:28:0,
                 from /home/cbp/lcdsketchbook/libraries/ArduinoMenu_library/src/menu.h:18,
                 from /home/cbp/work/arduino/menu/test/test.ino:4:
/home/cbp/lcdsketchbook/libraries/ArduinoMenu_library/src/macros.h:272:3: error: too many initializers for 'const Menu::menuFieldShadowRaw<double&>'
   };\
   ^
/home/cbp/lcdsketchbook/libraries/ArduinoMenu_library/src/macros.h:51:17: note: in expansion of macro 'DECL_FIELD_'
 #define DECL(x) DECL_##x
                 ^

And the menu in question looks like this:

`double kpid[] = { 0.0,0.0,0.0 };
MENU(subMenu,"PID",doNothing,anyEvent,noStyle
,FIELD(kpid[0],"P."," ",0,2000,1,0.01,doNothing,noEvent,wrapStyle)
,FIELD(kpid[1],"I."," ",0,2000,1,0.01,doNothing,noEvent,wrapStyle)
,FIELD(kpid[2],"D."," ",0,2000,1,0.01,doNothing,noEvent,wrapStyle)
,EXIT("<Back")
);`

After some time it dawned on me that the macros/templete regarded an indexed value as two separate values.
Changing the code, saving a few bytes and cpu cycles in the process, this works:

`double    kpid1, kpid2, kpid3;
MENU(subMenu,"PID",doNothing,anyEvent,noStyle
,FIELD(kpid1,"P."," ",0,2000,1,0.01,doNothing,noEvent,wrapStyle)
,FIELD(kpid2,"I."," ",0,2000,1,0.01,doNothing,noEvent,wrapStyle)
,FIELD(kpid3,"D."," ",0,2000,1,0.01,doNothing,noEvent,wrapStyle)
,EXIT("<Back")
);`

The indexed values was a left over from older code and not needed in this case.
Alas, I only have a very limited experience with templates (or heavy use of macros), hence I have not  pinpointed the cause of the problem.

neu-rah, 2018-02-12:
Interesting case
I can hardly wait for the day that I get rid of those ugly macros...
maybe next version :D

neu-rah, 2018-02-12:
but you mention, it was a change in last versions... strange, right now I'm clueless :D
will dig it and try to track it
thank you!

cbpdk, 2018-02-12:
As it annoys me as well not knowing why the error suddenly showed, I did look at a diff.
The difference making a difference is in macros.h, line 260:
Old file:
``` 
constMEM MEMMODE Menu::menuFieldShadowRaw<typeof(target)> fieldShadowRaw##cnt={\
```

New file:
```
constMEM MEMMODE Menu::menuFieldShadowRaw<decltype(target)> fieldShadowRaw##cnt={\
```

Replacing the decltype  typeof with makes the difference.
On the positive side, it forced me to optimise the code a tiny bit ;)

neu-rah, 2018-02-12:
@cbpdk github should have medals or badges ;), yes it was one of my changes, just to get it more standard (as decltype is c++11)... it looked good, but obviously not :\

neu-rah, 2018-02-12:
I'm giving you access to the repo so that you can push changes without need of a pull request, don't feel obliged to accept it, but it is a pleasure to have your contribution.

cbpdk, 2018-02-12:
Thank you for the opportunity to push changes. 
All I originally wanted was a simple menu system controlled by a rotary switch with the purpose of controlling a surplus hotbed using an Arduino Uno, now Mega2650 due to memory restrictions.
Hence the menu system is a secondary task where somebody else has been doing all the hard work ;-)
Of course I shall inform you before I make any changes/contributions I come across.

closed 2018-02-28 by neu-rah