neu-rah/ArduinoMenu #155: escCmd to return from idle

evfool, 2018-02-26:
Currently only the "enterCmd" seems to be able to navigate back to the menu from an idle function. It would be great if this could happen both with enterCmd or escCmd (exit from idle means open menu mode).

neu-rah, 2018-02-26:
seems ok, missed that myself multiple times. It was an attempt to contain code size.
some devices do not have esc key.

neu-rah, 2018-02-26:
meanwhile, this code can do what you want externally (loop function) before `nav.poll` or `nav.doInput`

```c++
  if (nav.sleepTask&&serial.peek()=='/') {
    serial.read();//remove the escape otherwise we will enter idle again
    nav.idleOff();
  }
```

nav.sleepTask is NULL when menu working
here i'm peeking serial (my menu input object) and look for '/', to be correct i should look for configured character, but this works

evfool, 2018-02-27:
Thanks for the suggestion. I'm using a key-based input, so indeed, some devices might not have an esc key, but in that case esc won't return to the previous screen, as they won't be able to press it. If you want, I could come up with a MR with the relevant changes (based on a quick check the two lines I'd also add escCmd as an option would be https://github.com/neu-rah/ArduinoMenu/blob/44ad5a31fbf2add5ec9d181507dea529cee09251/src/nav.cpp#L114 and https://github.com/neu-rah/ArduinoMenu/blob/44ad5a31fbf2add5ec9d181507dea529cee09251/src/nav.cpp#L128 ).

neu-rah, 2018-02-27:
excellent :D

commit 82e98c73 (2018-02-27): Fixed #155: escCmd also returns to menu from idle task
commit e1ea64ee (2018-02-27): Merge pull request #156 from evfool/master

Fixed #155: escCmd also returns to menu from idle task
closed 2018-02-27 by neu-rah