neu-rah/ArduinoMenu #160: Handling input while idle

evfool, 2018-03-03:
Currently the idle function has a reference which can easily output anything to the "idle" screen.
Some input (enterCmd and escCmd) are handled automatically to leave the idle mode, aka get back to the menu.

I'm trying to find a way to handle input (with callbacks) on the idle screen, and could not find an easy way other than handling the gpio states of the inputs (which are also used by the menu) in the main loop.

Do you have any suggestion how this could be achieved? Maybe having a "navRoot" for the idle task to be able to define callbacks for inputs?

The use-case is having a music player, with a menu, controlled by two buttons and a rotary encoder. On idle, the idletask shows currently playing song, on button press (enter/esc) the menu is shown. Still, on idle I would like to handle the rotary encoder (used for up/down navigation in the menu) as volume up/down.

neu-rah, 2018-03-03:
poll calls both doInput and doOutput, so on idle you can just call doInput provided that the nav focus is your intended field, for that you can either

set nav chain to point to the field... (i can explain the process)
or compile with MENU_ASYNC defined (-DMENU_ASYNC) and use the async api to position your focus

```c++
inline bool async(const char* at);
```
where `at` is a path string, example:
```c++
nav.async("/0/2");//for first submenu third field (zero based index)
```

_this is an experimental API for web access_

neu-rah, 2018-03-03:
can call `parseInput(navNode& nav,menuIn& in);` of target field

https://github.com/neu-rah/ArduinoMenu/blob/master/src/items.h#L100

the element must be a `navTarget` derived

and the navigation path still has to be correct

neu-rah, 2018-03-03:
best way to position focus and still respect the events of the menu structure is to:

1 - escape to minimal common base (or to root)
2 - select index by remaining path

evfool, 2018-03-14:
Marking as closed, thanks for your suggestions.

 I have resolved this the hard way, handling gpio i/o from the relevant pins using interrupts if the menu is not active (is in idle = display mode), and disable the interrupt-based handlers when the menu is active = not idle.

closed 2018-03-14 by evfool