neu-rah/ArduinoMenu #161: Two lines menu on a 128x32 display

T94T, 2018-03-14:
Hi,
We are working with a SSD1306 128x32 pixel display. Everything works fine when using the four line menu. The font is very small though.
No we want to change the menu type to twi lines, but it does not display then contents correctly.
Do you have some advice?

morozgrafix, 2018-03-14:
Don't mean to hijack your issue. Did you get SSD1306Ascii example working? I'm having difficulty running example from this library. It constantly flashing `Hello world!` on my 128x64 oled. Thanks.

T94T, 2018-03-14:
@morozgrafix: This works great for us:
```
#include <Wire.h>
#include <SSD1306Ascii.h>
#include <SSD1306AsciiWire.h>
SSD1306AsciiWire oled;
void setup() {
  Wire.begin();
  oled.begin(&Adafruit128x32, 0x3C);
  oled.set400kHz();
  oled.setFont(TT5x7);
  oled.home();
  oled.set2X();
  oled.print("Line1 Text");
  oled.print("Line2 Text");
}
void loop() {
}
```

morozgrafix, 2018-03-14:
Thanks, I was lookin through closed issues and found this thread with plenty of information and this worked for me https://github.com/neu-rah/ArduinoMenu/issues/134#issuecomment-360387891 I will try your solution as well.

neu-rah, 2018-03-14:
Hi all
you are also welcome to join us on gitter https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link

its easier to chat there, thanks for your help @morozgrafix 

using dual line prompts wont go easy with this menu

can be done by 

- tweaking an output driver
- custom printable items

and cheating the menu system by reporting a font size twice as tall

on either case it will affect all items, mixed sizes are not supported by the current API

let me know if you decide to implement and need help

T94T, 2018-03-20:
I'm willing to implement a new output driver. But for now I don't have enough experience with your library to achieve this.

neu-rah, 2018-03-20:
and what is your favorite driver for the ssd1306 display?
ascii | u8g2 | u8x8 | ... ?

T94T, 2018-03-20:
I'm using the SSD1306Ascii right now.

neu-rah, 2018-03-20:
from the references bellow we can have cursor information from the original SSD1306Ascii driver.

The idea is to derive the SSD1306AsciiOut and tweak its behavior by re-implement some members.

```c++
    class WrapDisplay:public SSD1306AsciiOut {
      public:
        using SSD1306AsciiOut::SSD1306AsciiOut;//use the same constructors

        //change this to wrap lines
        size_t write(uint8_t ch) override {
/*assuming that the original drive can not wrap line by itself 
you have to monitor the col() and increase row() when it is about to 
print out of screen, making col() be zero again, this will wrap lines*/
        }
        void setCursor(idx_t x,idx_t y,idx_t panelNr=0) override {
/*multiply y (row) by 2 and eventually add 1 if needed (x exceeding physical width)*/
        }
    };
```

function like `clearLine` rely on `setCursor` and print => write, so they must be ok by now

however `void clear(idx_t panelNr)` might need that i change the base of SSD1306Ascii to work ok. will see.

_reading the source code i can see that I might need to change the original class to derive from cursorOut instead of menuOut to better address clearing of screen areas, as the original drive is able to position cursor, but that is my part._

I think we can then tweak the device initialization by:

1 - provide WrapDisplay instead of SSD1306AsciiOut (same constructors)
2 - Turn panels twice as wide and half height

hope this works :D

**References:**

Ok, lets see if this makes sense...

the driver is SSD1306AsciiOut (https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuIO/SSD1306AsciiOut.h)

SSD1306AsciiOut derives from menuOut so the base code is here (https://github.com/neu-rah/ArduinoMenu/blob/master/src/io.h#L47)

and there you can see it derives from the Arduino Print class (https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/Print.h)

with only 3 `virtual` functions, but for us it will matter `virtual size_t write(uint8_t) = 0;`

its a pure virtual function and other `write` functions are based on it. So changing it will have an in impact on all others.

SSD1306AsciiOut implements its write here
https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuIO/SSD1306AsciiOut.h#L20

by just doing a `device->write(ch)`

meaning that the original SSD1306Ascii somehow maintains an internal cursor

the idea is to derive the SSD1306AsciiOut into a double line print
for that we need to know the cursor position.

and SSD1306Ascii has a cursor here
https://github.com/greiman/SSD1306Ascii/blob/master/src/SSD1306Ascii.h#L251

they are private but @greiman made some nice `col()` and `row()` functions

col: https://github.com/greiman/SSD1306Ascii/blob/master/src/SSD1306Ascii.h#L117
row: https://github.com/greiman/SSD1306Ascii/blob/master/src/SSD1306Ascii.h#L160

i can see that there is also a `set2X` function there, but I assume that its now what you want.

T94T, 2018-03-20:
Thanks a lot for this great, detailed answer! I will have a look tomorrow.

neu-rah, 2018-03-20:
you're welcome @T94T :+1: 
and feel free to ask any details as you need them

T94T, 2018-03-20:
I made a new class like this:
```
class SSD1306AsciiOutTwoLine: public SSD1306AsciiOut {
  public:
    using SSD1306AsciiOut::SSD1306AsciiOut;//use the same constructors

    void setCursor(idx_t x, idx_t y, idx_t panelNr = 0) override {
      const panel p = panels[panelNr];
      device->setCursor(p.x + x, p.y + y * 2);
    }
};
```
I do not need the word wrap functionality. My goal is to have only _two_ menu items visible at a time.
So I changed this code also:
`constMEM panel menuPanels[] MEMMODE = {{0, 0, 128 / MENU_FONT_WIDTH, 32 / MENU_FONT_HEIGHT * 2}};`

Unfortunately the menu is not working. It seems the logic is still based on a four line menu structure. Do you have any clue?

neu-rah, 2018-03-20:
oh my, yes much more easy :D
you just have to dimension your panel to be `{{0,0,128/MENU_FONT_WIDTH,2}}`
(units are in characters)

T94T, 2018-03-20:
Really cool, it works! Thank you so much for your help!
Summarized I use this now:
```
class SSD1306AsciiOutTwoLine: public SSD1306AsciiOut {
  public:
    using SSD1306AsciiOut::SSD1306AsciiOut;
    void setCursor(idx_t x, idx_t y, idx_t panelNr = 0) override {
      const panel p = panels[panelNr];
      device->setCursor(p.x + x, p.y + y * 2);
    }
};
```

`SSD1306AsciiOutTwoLine outOLED(&oled, tops, pList);`

`constMEM panel menuPanels[] MEMMODE = {{0, 0, 128 / MENU_FONT_WIDTH, 2}};`

closed 2018-03-20 by neu-rah