neu-rah/ArduinoMenu #163: Strange call to Wrire.begin()

d21d3q, 2018-03-19:
I found [this](https://github.com/neu-rah/ArduinoMenu/blob/b7bb71b083c206f5aed5816a962cb22f8a86f0dd/src/menuIO/PCF8574KeyIn.h#L64) call to `Wire.begin(_sda, _scl);`. What is that?

neu-rah, 2018-03-19:
seems to be I2C initialization for use with PCF8574
usually i accept an Wire& but the author preferred otherwise

d21d3q, 2018-03-19:
The point is, that `Wire.begin()` [doesn't take](https://github.com/arduino/Arduino/blob/6b4a3a6d9ecbf36943efbbe65951b4bb17e208fc/hardware/arduino/avr/libraries/Wire/src/Wire.h#L53) such arguments

neu-rah, 2018-03-19:
good point, it might take on some other platform... or some software I2C however the include is `#include <Wire.h>`

guess its better to change and accept just a Wire&

d21d3q, 2018-03-20:
It seems like call to [ESP8266 implementation](https://github.com/esp8266/Arduino/blob/6d3f0b40239b61472a1a0777b402f00d1165beec/libraries/Wire/Wire.h#L53).

neu-rah, 2018-05-20:
changed the code and removed _sda and _scl as you suggested, and accepting a TwoWire& defaulting to Wire. As I don't have the hardware can someone test this?

closed 2018-08-18 by neu-rah