neu-rah/ArduinoMenu #164: Hard time getting U8G2OUT working

thucar, 2018-03-22:
I'm having a hard time getting the U8G2 example working. It works perfectly with the U8G2's own GraphicTest example.

The settings for my LCD are:
`U8G2_UC1701_MINI12864_F_4W_SW_SPI u8g2(U8G2_R2, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);`

So in ArduinoMeny U8G2 example I have it set up like this:
```
#include <SPI.h>
  #define fontName u8g2_font_5x7_tf
  #define fontX 5
  #define fontY 9
  #define offsetX 0
  #define offsetY 0
  #define U8_Width 128
  #define U8_Height 64
  U8G2_UC1701_MINI12864_F_4W_SW_SPI u8g2(U8G2_R2, /* clock(D7)=*/ 11, /* data(D6)=*/ 13, /* cs(CS)=*/ 10, /* dc(RS)=*/ 9, /* reset(RST)=*/ 8);
```
Unfortunately, I'm not getting anything to show up on the display. What am I missing here?

thucar, 2018-03-22:
An update on the issue: I've enabled both serialIn and serialOut in addition to the display so I could confirm the sketch itself is working. 

I also did a separate test using LCDMenuLib2 which also supports U8G2, and the example sketch works correctly with the settings I've posted above.

I'm running out of ideas to debug this further..

neu-rah, 2018-03-22:
can you print something on setup before the menu? (just to confirm the u8g2 setup is ok)

also, can you show some more code of u8g2out initialization, menu IO setup and the loop (poll, doInput, doOutput)?

if you prefer you can join gitter and open a private chat to paste code.

https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link

neu-rah, 2018-03-22:
after some discussion on gitter we concluded that it was a problem with LED_BUILTIN, as the SPI also uses pin13 on arduino mega the setting of led state was interfering with the display.

closed 2018-03-22 by neu-rah