neu-rah/ArduinoMenu #165: Split menu among modules

d21d3q, 2018-03-22:
Is it possible to split menu among modules? I was trying to have parts of menu inside of modules to organise it somehow, and linker started to yell errors about multiple definitions. It seems (I didn't dive in source), that during expanding macros, there are some labels created that are unique among single file, but on linking time it becomes a problem.

I found working solution with using namespaces:
module.h
```c++
#include <menu.h>

namespace ModuleMenu {
  extern Menu::menu someMenu;
}
```
module.cpp
```c++
#include "module.h"
int choose;
namespace ModuleMenu {
CHOOSE(some_choose,choose,"Choose me",doNothing,noEvent,noStyle
  ,VALUE("Foo",0,doNothing,noEvent)
  ,VALUE("bar",25,doNothing,noEvent)
);

MENU(someMenu, "Some menu", doNothing, noEvent, wrapStyle
  ,SUBMENU(some_choose)
  ,EXIT("<Back")
);
}
```

main.ino
```c++
#include <menu.h>
#include "module.h"

MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  ,SUBMENU(ModuleMenu::someMenu)
  ,EXIT("<Back")
);
NAVROOT(...)
```
Some alternative way of doing this in more elegant way?

neu-rah, 2018-03-23:
if you want to use macros then using namespaces is the most elegant way i can think of.
what about not using macros? that way you can name your own elements, declare them on headers and implement on cpp...
or you can use .ino files, i guess they are all concatenated before compile, this does not seem elegant for me... but it seems to work for macros.

neu-rah, 2018-05-20:
i was expecting duplicate symbol errors, because macros generate unique labels as you pointed out but my test compiled ok, just put part of my menu on **.h** files and included them (only once) on .ino file.
no errors...
.h files had `#pragma once` as usual, but as I included only once it has no effect.

closed 2018-05-20 by neu-rah