neu-rah/ArduinoMenu #166: Menu change State,

kamilszczepan, 2018-03-25:
Hi, i have problem with menu .
I understand most things, but I can not do it to change the submenu in the main menu after selecting one item in the menu

For example, sketch, choose a motor in main menu select function. in the next position I have a submenu only to the selected motor and in the next to start this one

MAIN MENU

|| MOROR ||   : MOTOR 1, MOTOR 2, MOTOR 3 ( select function, chose motor )
|| SETTINGS ||  ( sub menu, for before chosing motor, settings submenu must be changed when select motor )
|| START || ( toggle  start stop, must be changed for start seleceted motor )

any idea?

neu-rah, 2018-03-25:
Hi!

for a static menu, if settings depend on motor then they should be a sub-menu or motor, and of course motor wont be a selection but only sub-menu entries. and start stop be on each motor sub-menu.

still if you want to keep this organization and as static menu you can change menu root, because the navigation object is not static and you can switch the menu its attached to it. And have a collection of menus for each case...

or a dynamic menu (default for non-avrs), then you can change the menu at runtime.

not sure what you are trying to do, but i can provide more details on each case...

kamilszczepan, 2018-03-25:
Thank you for the response, 
I can do to make every motor in a separate submenu,

|| motor1 || -> settings 
                   -> start 
|| motor2 || -> settings 
                   -> start 
|| motor3 || -> settings 
                   -> start 
 but that's not what I mean
I would like to get something like dynamic menu, if select motor then i got i got different settings. 

Like that 

int chooseMotor=1;
SELECT(choseMotor,chooseMenu," Mode ~",doNothing,noEvent,noStyle
  ,VALUE("Motor1",1,doNothing,noEvent)
  ,VALUE("Motor2",2,doNothing,noEvent)
  ,VALUE("EMotor 3",3,doNothing,noEvent)
);

MENU(Settings1," motor1settings",showEvent,anyEvent,noStyle
 ,FIELD(speed_es,"Speed","%",0,100,10,2,doNothing,noEvent,wrapStyle)
  ,EXIT("< Back")
)
MENU(Settings2," motor2settings",showEvent,anyEvent,noStyle
 ,FIELD(speed_es,"Speed","%",0,100,10,2,doNothing,noEvent,wrapStyle)
  ,EXIT("< Back")
)
MENU(Settings3," motor3settings",showEvent,anyEvent,noStyle
 ,FIELD(speed_es,"Speed","%",0,100,10,2,doNothing,noEvent,wrapStyle)
  ,EXIT("< Back")
)

MENU(mainMenu,"Motor main menu '",doNothing,noEvent,wrapStyle
,SUBMENU(ChoseMotor)
 ,SUBMENU(Settings_motor ) ->> if chooseMotor=1 then Settings_motor=motor1settings else /motor2, motor3  
,SUBMENU(Motor_stop)      the same as settings_motor if select motor then i got start stop only for this one 
  );

I have no idea how  .. 

thank you for your time

neu-rah, 2018-03-25:
Hi! you are welcome.

I think that your case is similar to the `reuse` example
I would use a single settings menu targeting some temp variables and use the **events** to:

- set the temp. vars to the adequate values when choosing a motor.
- when changing the temp vars on settings, actively reflect the changes to the adequate motor

_still this can be done by swapping the menu, but that is unnecessarily more complicated_

you are also welcome to join us at gitter:
https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link

kamilszczepan, 2018-03-25:
in reuse example i got error

Arduino:1.6.9 (Windows 10), Board:"Arduino Pro or Pro Mini, ATmega328 (5V, 16 MHz)"

reuse:124: error: 'bool AlarmMenu::changed(const Menu::navNode&, const Menu::menuOut&, bool)' marked override, but does not override
 bool changed(const navNode &nav,const menuOut& out,bool sub=true) override {

neu-rah, 2018-03-25:
yes, `reuse` is dynamic, and dynamic is only available for AVR's if compiling with `MENU_USERAM` defined on the compiler command line (not available on arduino IDE AFAIK), but because you don't need to change the menu structure only have to catch values with the event handler this is ok for static menus.

neu-rah, 2018-03-25:
ok, here's an example that works for static menu

```c++
/********************
Arduino generic menu system

Rui Azevedo - ruihfazevedo(@rrob@)gmail.com

output: Serial
input: Serial
*/

#include <menu.h>
#include <menuIO/serialOut.h>
#include <menuIO/serialIn.h>

using namespace Menu;

int chooseMotor=1;
SELECT(chooseMotor,chooseMenu," Mode ~",doNothing,noEvent,noStyle
  ,VALUE("Motor 1",1,doNothing,noEvent)
  ,VALUE("Motor 2",2,doNothing,noEvent)
  ,VALUE("Motor 3",3,doNothing,noEvent)
);

int speed_es=0;

result setSpeed() {
  Serial.print("\n\nsettting speed of motor ");
  Serial.print(chooseMotor);
  Serial.print(" to ");
  Serial.print(speed_es);
  Serial.println("%%");
  switch(chooseMotor) {
    case 1:
      //set motor 1 to speed_es
      break;
    case 2:
      //set motor 1 to speed_es
      break;
    //....
  }
  //or use something like motors[chooseMotor]
  return proceed;
}

//if you want this prompt to show the motor number you can customize the print
MENU(Settings_motor," motor settings",doNothing,noEvent,noStyle
  ,FIELD(speed_es,"Speed","%",0,100,10,2,setSpeed,updateEvent|enterEvent,wrapStyle)
  ,EXIT("< Back")
)

result Motor_stop() {
  Serial.print("\n\nStoping motor ");
  Serial.println(chooseMotor);
  switch(chooseMotor) {
    case 1: break;
    case 2: break;
    //....
  }
  //or use something like motors[chooseMotor]
  return proceed;
}

MENU(mainMenu,"Motor main menu ",doNothing,noEvent,wrapStyle
  ,SUBMENU(chooseMenu)
  ,SUBMENU(Settings_motor ) //->> if chooseMotor=1 then Settings_motor=motor1settings else /motor2, motor3
  // ,SUBMENU(Motor_stop) //the same as settings_motor if select motor then i got start stop only for this one
  ,EXIT("Exit")
);

#define MAX_DEPTH 2

MENU_OUTPUTS(out,MAX_DEPTH
  ,SERIAL_OUT(Serial)
  ,NONE//must have 2 items at least
);

serialIn serial(Serial);
NAVROOT(nav,mainMenu,MAX_DEPTH,serial,out);

void setup() {
  Serial.begin(115200);
  while(!Serial);
  Serial.println("menu 4.x test");Serial.flush();
}

void loop() {
  nav.poll();
  delay(100);//simulate a delay when other tasks are done
}
```

kamilszczepan, 2018-03-26:
Thank you so much, 
I got Error compiling - - updateEvent|enterEvent,wrapStyle is a problem 
I change to FIELD(speed_es,"Speed","%",0,100,10,2,setSpeed,enterEvent,wrapStyle) 
Then works, but when i change speed in one motor then Of course, it changes for all , defined one int speed_es. 
Is it an option   to have different submenus after choosing different motors ? 

example motor 1 have speed. motor 2 have acceleration , speed, delay, bla bla bla and motor 3 have speed and somenthing else . 

unfortunately I think I'm too stupid for it

neu-rah, 2018-03-26:
@kamilszczepan do not depreciate yourself, we all had our first day on c++ :D and then you should see me trying to solder hehehe.

Its because somewhere in your code you are setting the motors speed by that variable, you should either have different variables and use the event to copy the value to the adequate variable or change the correct motor speed inside the event handler and use only that variable.

you can have a menu of motors and a sub-menu to control each one also.

You can also send me the code and i can fix it or join use on gitter chat, we can have public and private rooms if you don't want your code to be public.

kamilszczepan, 2018-03-26:
the only I need is to change the whole submenu in the main menu. 
If change state 'ChoseMotor" 1,2,3 case 1 ; submenu1 , case2, submenu2

someting like 

 MENU(set1,"settings1",doNothing,noEvent,wrapStyle
  ,FIELD(speed1,"Speed","%",0,100,10,2,setSpeed,updateEvent,wrapStyle)
  ,EXIT("< Back")
));
 MENU(set2,"settings2",doNothing,noEvent,wrapStyle
  ,FIELD(speed2,"Speed","%",0,100,10,2,setSpeed,updateEvent,wrapStyle)
  ,EXIT("< Back")
));
 MENU(set2,"settings3",doNothing,noEvent,wrapStyle
  ,FIELD(speed3,"Speed","%",0,100,10,2,setSpeed,updateEvent,wrapStyle)
  ,EXIT("< Back")
));

In main menu, my settings submenu is VARIABLE to chose current menu when  chose motor 1,2,3

MENU(mainMenu,"Motor main menu ",doNothing,noEvent,wrapStyle
  ,SUBMENU(chooseMenu)
  ,SUBMENU(VARIABLE )

Then in switch change the menu 

switch(chooseMotor) {
    case 1:
   variable=set1;
    case 2:
       variable=set2;
      break;
     case 3:
       variable=set3;
      break;
  }
 
everything is simple in terori, or it seems so . but but I have too many deficiencies in language 

I do not have any code  because I'm stuck on your example, if I have a variable submenu,  rest is 15 min, copy paste, change names and done

neu-rah, 2018-03-26:
I was pointing you to a static menu and the variability is done inside the event handler, where we check the selected motor and do the changes only to that motor there.

do you have any code of how to control your motors?

kamilszczepan, 2018-03-26:
I got only one motor, the choice menu motor 1, motor 2, motor3  this is only example . 
final code will be something like motor option one, motor option 2 ... 
I will control the stepper motor with a library where I will put it map value from FIELD. 
I have an idea like that
in every each option I want different functions so i can not relate to only one sub menu i need 3 for 3 choice and  different options . 

if you have a moment and you could show how to change the sub menu after choosing a motor 
I will build the rest of my vision on this.

neu-rah, 2018-03-26:
no problem, i will post some code here :D

kamilszczepan, 2018-03-26:
If i have 3 submenu's changing  in main menu when I change SELECT - > value, 
all the rest would be easy. 
I got different FIELD, differend int...  and I could very easily map to a stepper motor value 

I'm sorry but I'm not a programmer, but I'm soldering very well :)

neu-rah, 2018-03-28:
hi!

here is some code that uses a single interface to control 3 different leds

it is not necessary to have multiple menus to control multiple devices, neither is needed to change the menu.

```c++
/********************
Arduino generic menu system

Rui Azevedo - ruihfazevedo(@rrob@)gmail.com

output: Serial
input: Serial
*/

#include <menu.h>
#include <menuIO/serialOut.h>
#include <menuIO/serialIn.h>

using namespace Menu;

//will output pwm to controll led bright
#define LED1 9
#define LED2 10
#define LED3 11

//this can be done into a single array of object.. but i leave it as separate arrays for clarity
int ledPins[3]{LED1,LED2,LED3};
int ledBright[3]{0};
bool ledState[3]{true};

int selLed=0;
int led_bright=0;
bool state=true;

//when changing led selection we copy the original value
result getLedValues() {
  Serial.print("\n\nget value for led ");
  Serial.println(selLed);
  led_bright=ledBright[selLed];
  state=ledState[selLed];
  return proceed;
};

SELECT(selLed,chooseMenu,"Control",doNothing,noEvent,noStyle
  ,VALUE("Led 1",0,getLedValues,updateEvent|enterEvent)
  ,VALUE("Led 2",1,getLedValues,updateEvent|enterEvent)
  ,VALUE("Led 3",2,getLedValues,updateEvent|enterEvent)
);

result setBright() {
  ledBright[selLed]=led_bright;//store value for led bright
  if (ledState[selLed]) {//only change led if its turned on
    Serial.print("\n\nsettting brightness of led ");
    Serial.print(selLed);
    Serial.print(" to ");
    Serial.print(((float)led_bright));
    Serial.println("%%");
    analogWrite(ledPins[selLed],(float)led_bright);
  }
  return proceed;
}

result togLed() {
  Serial.print("\n\ntoggling led ");
  Serial.println(selLed);
  ledState[selLed]=state;
  //restore brightness of turn off
  if (state) analogWrite(ledPins[selLed],((float)led_bright));
  else analogWrite(ledPins[selLed],0);
  return proceed;
}

TOGGLE(state,stateTog,"State:",togLed,updateEvent|enterEvent,noStyle
  ,VALUE("Off",false,doNothing,noEvent)
  ,VALUE("On",true,doNothing,noEvent)
);

MENU(mainMenu,"LED main menu ",doNothing,noEvent,wrapStyle
  ,SUBMENU(chooseMenu)
  ,FIELD(led_bright,"Bright","%",0,100,10,2,setBright,updateEvent|enterEvent,wrapStyle)
  ,SUBMENU(stateTog)
  // ,EXIT("Exit")
);

#define MAX_DEPTH 2

MENU_OUTPUTS(out,MAX_DEPTH
  ,SERIAL_OUT(Serial)
  ,NONE//must have 2 items at least
);

serialIn serial(Serial);
NAVROOT(nav,mainMenu,MAX_DEPTH,serial,out);

void setup() {
  Serial.begin(115200);
  while(!Serial);
  Serial.println("menu 4.x test");Serial.flush();
  pinMode(LED1,OUTPUT);
  pinMode(LED2,OUTPUT);
  pinMode(LED3,OUTPUT);
}

void loop() {
  nav.poll();
  delay(100);//simulate a delay when other tasks are done
}
```

kamilszczepan, 2018-03-28:
First of all, I would like to thank you for your time. 
Your example not compile in arduino 1.6.9 But it is not important at this point. 

I told you I need the whole submenu to change into main. after choosing "led" in this case.
I want to have different functions in it, and a different number of these functions.

  I see that it is not possible

all the best ;)

neu-rah, 2018-03-28:
yes, it is possible. Have to compile with MENU_USERAM globally defined (compiler command line), ir use a non-avr platform (defaults to dynamic). ex: `-DMENU_USERAM`

sorry I had not perceived from your code that you need sub-menus to be different, the sub-menus seemed all equal so i deduced we can solve this by just using the handlers.

when using dynamic menus you can access and change the menu data at runtime.

```c++
mainMenu[1]=&set1;
```

dont know if arduino ide 1.6.9 already allows command line options, I've compiled my example using platformio (because it allows me more advanced customization), thanks for alerting me of problems compiling with 1.6.9, gota check it.

kamilszczepan, 2018-03-29:
Yourcode is very economical , but it will not work for me.
For now I am at the stage when I change the entire main menu, the encoder and the screen does not work. Maybe  "when using dynamic menus you can access and change the menu data at runtime."  , or I'm stupid.  

I came up with an idea, is there any way to mask / unmask FIELD's ?
we have all the options in the submenu, and when choosing in  SELECT  mask unmask /true false field or something . maybe it's genius in its simplicity, I do not know ...

neu-rah, 2018-03-29:
yes the last code is damn economical and wont work on an AVR unless you do some defines and is kind of pseudo as it lacks the remaining definitions as you noticed... but you cant do the defines using arduino IDE, and you use an AVR and use an old arduino IDE :D

here are some options,

**on static menus:**
- options can be enabled disabled at runtime (disabled field still reflect values changes, and they still appear)
- print can be customized, changing format or including info from other sources
- menu root can be changed changing the whole menu
- can NOT hide/add/remove options at runtime

**dynamic menus**
- do all the above of course and can add/remove/replace options at runtime

for dynamic code to work for you, you have to either **use another ide** or an **non-avr board** because at least one is required for you to use dynamic.

there is still the chance of making the motor selection a menu (not a select/choose/toggle field) an the each motor can be a sub-menu and each with its own options, no extra tricks needed.

kamilszczepan, 2018-03-29:
I have some silly concepts
[image]

neu-rah, 2018-03-29:
just curious, what are you trying to control? what kind of machine/system?

kamilszczepan, 2018-03-29:
I will not say...
guess  :)
[image]

neu-rah, 2018-03-29:
ah-ah! excellent
congratulations very nice construct!
luv it! and its the kind of thing (mechanical construct and electronic circuits) that i really suck! hehe
so we can complement each other
i see that you are using an LCD, that good because this menu library is heavy and using a gfx display would leave almost no space for the menu structure on a regular avr like arduino nano/uno (atmega328).
do you want to have pre-programmed sequences?

kamilszczepan, 2018-03-29:
Everything seemed beautiful, there was a concept, there was a desire and it went out as usual .. 
My goofy, very ineffective program does not work, says something has not been declared.
 can you look 

#include <Wire.h>
#include <LiquidCrystal_I2C.h>//F. Malpartida LCD's driver
#include <menu.h>//menu macros and objects
#include <menuIO/lcdOut.h>//malpartidas lcd menu output
#include <TimerOne.h>
#include <ClickEncoder.h>
#include <menuIO/clickEncoderIn.h>
 
#include <menuIO/chainStream.h>

using namespace Menu;

LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3,POSITIVE);  // Set the LCD I2C address and pinout

// Encoder /////////////////////////////////////
#define encA 3
#define encB 2
//this encoder has a button here
#define encBtn 4

ClickEncoder clickEncoder(encA,encB,encBtn,2);
ClickEncoderStream encStream(clickEncoder,1);
MENU_INPUTS(in,&encStream);
void timerIsr() {clickEncoder.service();}

int speed_1=0;
int accel_1=0;
int dec_1=0;
int rotation_1=0;

int speed_2=0;
int accel_2=0;
int dec_2=0;
int rotation_2=0;

int speed_3=0;
int accel_3=0;
int dec_3=0;
int rotation_3=0;

MENU(choose_1,"Mode ~ Program 1",doNothing,noEvent,wrapStyle
 ,EXIT("Mode ~ Program 1")
,SUBMENU(mainMenu2)
 ,SUBMENU(mainMenu3)
);

MENU(choose_2,"Mode ~ Program 2",doNothing,noEvent,wrapStyle
,SUBMENU(mainMenu1)
 ,EXIT("Mode ~ Program 2")
 ,SUBMENU(mainMenu3)
);
MENU(choose_3,"Mode ~ Program 3",doNothing,noEvent,wrapStyle
,SUBMENU(mainMenu1)
 ,SUBMENU(mainMenu3)
  ,EXIT("Mode ~ Program 3")
);

//if you want this prompt to show the motor number you can customize the print
MENU(Settings_motor1," motor settings",doNothing,noEvent,noStyle
  ,FIELD(speed_1,"Speed","%",0,100,10,2,doNothing,noEvent,wrapStyle)
   ,FIELD(accel_1,"Accel","%",0,100,10,2,doNothing,noEvent,wrapStyle)
    ,FIELD(dec_1,"Dec","%",0,100,10,2,doNothing,noEvent,wrapStyle)
     ,FIELD(rotation_1,"Rotation","%",0,100,10,2,doNothing,noEvent,wrapStyle)
  ,EXIT("< Back")
)

MENU(Settings_motor2," motor settings",doNothing,noEvent,noStyle
  ,FIELD(speed_2,"Speed","%",0,100,10,2,doNothing,noEvent,wrapStyle)
   ,FIELD(accel_2,"Accel","%",0,100,10,2,doNothing,noEvent,wrapStyle)
    ,FIELD(dec_2,"Dec","%",0,100,10,2,doNothing,noEvent,wrapStyle)
     ,FIELD(rotation_2,"Rotation","%",0,100,10,2,doNothing,noEvent,wrapStyle)
  ,EXIT("< Back")
)

MENU(Settings_motor3," motor settings",doNothing,noEvent,noStyle
  ,FIELD(speed_3,"Speed","%",0,100,10,2,doNothing,noEvent,wrapStyle)
   ,FIELD(accel_3,"Accel","%",0,100,10,2,doNothing,noEvent,wrapStyle)
    ,FIELD(dec_3,"Dec","%",0,100,10,2,doNothing,noEvent,wrapStyle)
     ,FIELD(rotation_3,"Rotation","%",0,100,10,2,doNothing,noEvent,wrapStyle)
  ,EXIT("< Back")
)

MENU(mainMenu1,"Motor main menu ",doNothing,noEvent,wrapStyle
  ,SUBMENU(choose_1)
  ,SUBMENU(Settings_motor1) //->> if chooseMotor=1 then Settings_motor=motor1settings else /motor2, motor3
//   ,SUBMENU(Motor_stop) //the same as settings_motor if select motor then i got start stop only for this one
  ,EXIT("< Back")
);
MENU(mainMenu2,"Motor main menu ",doNothing,noEvent,wrapStyle
  ,SUBMENU(choose_2)
  ,SUBMENU(Settings_motor2) //->> if chooseMotor=1 then Settings_motor=motor1settings else /motor2, motor3
//   ,SUBMENU(Motor_stop) //the same as settings_motor if select motor then i got start stop only for this one
  ,EXIT("< Back")
);
 
MENU(mainMenu3,"Motor main menu ",doNothing,noEvent,wrapStyle
  ,SUBMENU(choose_3)
  ,SUBMENU(Settings_motor3) //->> if chooseMotor=1 then Settings_motor=motor1settings else /motor2, motor3
//   ,SUBMENU(Motor_stop) //the same as settings_motor if select motor then i got start stop only for this one
  ,EXIT("< Back")
);

 

#define MAX_DEPTH 4

MENU_OUTPUTS(out,MAX_DEPTH
  ,LCD_OUT(lcd,{0,0,20,4})
  ,NONE
);

 
NAVROOT(nav,mainMenu1,MAX_DEPTH,in,out);

void setup() {
  pinMode(encBtn,INPUT_PULLUP);
    lcd.begin(20,4);
  Timer1.initialize(1000);
  Timer1.attachInterrupt(timerIsr);
}

void loop() {
  nav.poll();
  delay(100);//simulate a delay when other tasks are done
}

neu-rah, 2018-03-29:
still trying to understand your menu schema, it seems that you want ti execute some procedures like "program2" and have access to other menu after (MAIN MENU 2)... or is it a replacement?

yeah i will look at the program :D

neu-rah, 2018-03-29:
add
`extern menu mainMenu1,mainMenu2,mainMenu3;`
before the menu defintion because you need to forward definitions has you are making forward references

this can play bad with the navigation if you close a loop, the navigation depth is potentially infinite on that case :D

neu-rah, 2018-03-29:
have you considered using some other MCU? like esp8266 or some other ARM

kamilszczepan, 2018-03-29:
i got esp, but I have never used it.

Menu work like a want but i have error:maxDepth reac. :)
i have max depth on 20 :)

neu-rah, 2018-03-29:
so you closed the loop... hmm
cant, should use a "Home" option instead, one that goes to the start but does not include the start as a submenu, otherwise there will be no enough depth....

are you sure you don't want to use gitter? we have provate and public chat rooms there and we can also format code as in here.

neu-rah, 2018-03-29:
ok, but now you can elaborate specific menus and sub-menus for each device.. i will check the hanging

kamilszczepan, 2018-04-08:
if i have no exit to root will be nice . 
I need to turn off the encoder after pressing something in my menu.

int encA = 3;
int encB = 2;

 ClickEncoder clickEncoder(encA,encB,encBtn,2);

 switch(qwerty) {
    case 1: 
encA = 0;
encB = 0;
    break;
    case 2: 
encA = 3;
encB = 2;
  break;
}
return proceed;
}
of course it does not work like that :)

Any idea ?

kamilszczepan, 2018-04-08:
I need disable encoder move when i press start case;1
of course only move, not encBtn :)
https://www.youtube.com/watch?v=uddWzXYj040

neu-rah, 2018-04-08:
I had a similar case somewhere, searching the code... made a custom input handler element i think.
to facilitate cases like this i think i can allow an input filter to be set on navRoot, just thinking.
Either way you should command your stepper based on a timer, of course it does not solve the menu issue, but gives your movement reliability and smoothness.

neu-rah, 2018-04-08:
ok, found the code and cleaned it up a bit

```c++
  // ...
int started=false;

template<typename T>
class Waiter:public menuField<T> {
public:
  using menuField<T>::menuField;
  Used printTo(navRoot &root,bool sel,menuOut& out, idx_t idx,idx_t len,idx_t pn) override {
    return out.printRaw(started?"[Stop]":"[Start]",len);
  }
};

result clicked(eventMask e, prompt &item) {
  started^=1;
  item.dirty=true;
  return proceed;
}

MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  ,OP("Op1",doNothing,noEvent)
  ,altFIELD(Waiter,started,"","",0,1,0,0,clicked,(eventMask)(enterEvent|exitEvent),wrapStyle)
  ,EXIT("<Back")
);

// ....

void setup() {
  // ...
  nav.useUpdateEvent=true;
}

```

hacking this with a field as it naturally captures the input, but making fields steps=0 so that user up/down keys have no effect.

using also an alternative field definition to customize the print

kamilszczepan, 2018-04-08:
OK ! ....
it locks the encoder, but the menu always sees the impulse ,
and it affects the work of the stepper // acceleration number of steps //

maybe I will add a relay and I will include or not the pins of encA and encB BUAHAHHAHAHAHAHA
it's just an idiot idea :)

https://www.youtube.com/watch?v=5EQNJzT3qnQ

neu-rah, 2018-04-08:
yes its does, as expected. you can minimize that by using the efficient way explained here (ask for details if needed), but it does not invalidate the fact that you should drive your steppers with a timer.

you can read the encoder button state at any time, so it will provide the method for [stop] and return to menu.

kamilszczepan, 2018-04-08:
Listen, my joke with the relay was funny, but he made me think :)
I will have a ready solution in 5 minutes. We have three pins encA gnd encB, I know this should not be done, and that's not the best way .But one transistor does everything :)
Base -> arduino pin, Emitter  -> gnd , collector -> encoder gnd . 
case:1 pin=HIGH;  case:2 pin=LOW 

I know you will not approve, but it must work :)

kamilszczepan, 2018-04-08:
https://www.youtube.com/watch?v=fXu0pX2MtbA

okay, there was one more thing left ...  In Field i do not want to enter x 10 digits , i want to enter only 
one digit  like when you click a second time, can be ??

neu-rah, 2018-04-09:
not sure i understand this one, can you explain more pls?

kamilszczepan, 2018-04-09:
ok, from the beginning :)
when I change the value in FIELD first i got : 10, 20 , 30 ,40 if i click a got > 11,12,13,14 .... 

I want to have only the second way to intput numbers

neu-rah, 2018-04-09:
easy, make first step 1 instead of 10 and the second zero
ex: `FIELD(test,"Test","%",0,100,1,0,doNothing,noEvent,wrapStyle)`

kamilszczepan, 2018-04-09:
🥇

neu-rah, 2018-04-09:
i prefer github stars ;) but a medal is ok
and please open new issues for new stuff for the sake of future reference
so, you using that transistor thing to suspend encoder rotation?

kamilszczepan, 2018-04-09:
yes,  because for me it was the fastest reliable solution.  I have another problem, this time with a stepper motor and the number of steps you can ask him in one package :)  
if you want to have a smooth movement you need to have more steps

closed 2018-04-09 by kamilszczepan

neu-rah, 2018-04-09:
that's the part of the timer