neu-rah/ArduinoMenu #167: Outputting a variable as text

Jebuar, 2018-03-26:
Hello.
The following problem occurs when creating a menu
`int countGlobal = EEPROM.read(0);;          //conditionally

MENU(aboutMenu, "About", doNothing, anyEvent, noStyle          
         , EXIT("<Back")   
         , OP("version v0.1", showEvent, anyEvent)
         , OP("ID 00000AA", showEvent, anyEvent)         
         , FIELD(countGlobal, "Mileage: ", "", 0, 0, 0, 0, doNothing, noEvent, wrapStyle)
);
 `
The problem in the output of the line SUBMAN on the LСD:
` FIELD(countGlobal, "Mileage: ", "", 0, 0, 0, 0, doNothing, noEvent, wrapStyle)`

When using the encoder, this line is active for me and when I press it it automatically skips to 0, everything reboots normally when rebooting.

How can I make it so that I can just output the variable for clarity?
Example:
` OP("Mileage: " + countGlobal, showEvent, anyEvent) `

Jebuar, 2018-03-26:
```javascript
function fancyAlert(arg) {
  if(arg) {
    $.facebox({div:'#foo'})
  }
}
```

neu-rah, 2018-03-26:
Hi!
Variable with values outside range can behave bad, you should set the range to valid values and disable the option, that way it will update but wont allow edit.

```c++
//use a value range, ex: 0~10000 or whatever range you find adequate
FIELD(countGlobal, "Mileage: ", "", 0, 10000, 0, 0, doNothing, noEvent, wrapStyle)
//...
//somewhere on setup
aboutMenu[3].disable();
```

Jebuar, 2018-03-26:
Thank you very much and sorry for the stupid question, later I read the wiki again and in the Menu definition section I saw everything

closed 2018-03-26 by Jebuar

neu-rah, 2018-03-26:
you are welcome :D