neu-rah/ArduinoMenu #169: ESP8266 (nodemcu) + 3 buttons + ClickEncoder(lib)

sdurnev, 2018-03-26:
Hi
I have the following system:
ESP8266 (nodemcu) + 3 buttons + ClickEncoder(lib) + Serial (debug) + Serial1 (only TX "Menu Out")

Serial1 output
```
[1]>Test 55% !!!!!!!!!!!
[2] Sub-Menu
[3] OpA
[4] OpB
[5] Led: Off
[6] LED On
[7] LED Off
[8] Select Zero
[9] Choose Last
[-] Alert test
```
If i press the button encB:
Serial1 output
```
[1] Test 55% !!!!!!!!!!!
[2]-Sub-Menu !!!!!!!!!!!
[3] OpA
[4] OpB
[5] Led: Off
[6] LED On
[7] LED Off
[8] Select Zero
[9] Choose Last
[-] Alert test
```
Serial output
```
event: 16
```
If I release the button encB:
Serial1 output
```
[1]>Test 55% !!!!!!!!!!!
[2] Sub-Menu !!!!!!!!!!!
[3] OpA
[4] OpB
[5] Led: Off
[6] LED On
[7] LED Off
[8] Select Zero
[9] Choose Last
[-] Alert test
```
Serial output
```
event: 16
event: 32 !!!!????
```
Do not fixed cursor in the menu item.
What is the problem?
Help me please.

Code below..

```
//*******************
//Arduino generic menu system
//Arduino menu using clickEncoder and I2C LCD
//
//Sep.2014 Rui Azevedo - ruihfazevedo(@rrob@)gmail.com
//Feb.2018 Ken-Fitz - https://github.com/Ken-Fitz
//
//LCD library:
//https://bitbucket.org/fmalpartida/new-liquidcrystal/wiki/Home
//http://playground.arduino.cc/Code/LCD3wires

#include <Arduino.h>
#include <Ticker.h>
#include <ClickEncoder.h>
#include <menuIO/clickEncoderIn.h>
#include <menuIO/keyIn.h>
#include <menuIO/chainStream.h>
#include <menuIO/serialOut.h>
#include <menuIO/serialIn.h>

using namespace Menu;

// Encoder /////////////////////////////////////
#define encA     4 // 2 - D4 Nodemcu
#define encB     0 // 0 - D3 Nodemcu
#define encBtn   13 // 13 - D7 Nodemcu

ClickEncoder clickEncoder(encA, encB, encBtn, 1);
ClickEncoderStream encStream(clickEncoder, 1);
MENU_INPUTS(in, &encStream);

Ticker ticker;

void timerIsr() { clickEncoder.service(); }

result doAlert(eventMask e, prompt &item);

result showEvent(eventMask e, navNode &nav, prompt &item) {
    Serial.print("event: ");
    Serial.println(e);
    return proceed;
}

int test = 55;

result action1(eventMask e, navNode &nav, prompt &item) {
    Serial.print("action1 event: ");
    Serial.print(e);
    Serial.println(", proceed menu");
    Serial.flush();
    return proceed;
}

result action2(eventMask e, navNode &nav, prompt &item) {
    Serial.print("action2 event: ");
    Serial.print(e);
    Serial.println(", quiting menu.");
    Serial.flush();
    return quit;
}

int ledCtrl = LOW;

result myLedOn() {
    ledCtrl = HIGH;
    return proceed;
}

result myLedOff() {
    ledCtrl = LOW;
    return proceed;
}

TOGGLE(ledCtrl, setLed, "Led: ", doNothing, noEvent, noStyle//,doExit,enterEvent,noStyle
, VALUE("On", HIGH, doNothing, noEvent), VALUE("Off", LOW, doNothing, noEvent)
);

int selTest = 0;
SELECT(selTest, selMenu, "Select", doNothing, noEvent, noStyle, VALUE("Zero", 0, doNothing, noEvent),
       VALUE("One", 1, doNothing, noEvent), VALUE("Two", 2, doNothing, noEvent)
);

int chooseTest = -1;
CHOOSE(chooseTest, chooseMenu, "Choose", doNothing, noEvent, noStyle, VALUE("First", 1, doNothing, noEvent),
       VALUE("Second", 2, doNothing, noEvent), VALUE("Third", 3, doNothing, noEvent),
       VALUE("Last", -1, doNothing, noEvent)
);

class altPrompt : public prompt {
public:
    altPrompt(constMEM promptShadow &p) : prompt(p) {}

    Used printTo(navRoot &root, bool sel, menuOut &out, idx_t idx, idx_t len, idx_t) override {
        return out.printRaw("special prompt!", len);;
    }
};

MENU(subMenu, "Sub-Menu", showEvent, anyEvent, noStyle, OP("Sub1", showEvent, anyEvent),
     OP("Sub2", showEvent, anyEvent), OP("Sub3", showEvent, anyEvent), altOP(altPrompt, "", showEvent, anyEvent),
     EXIT("<Back")
);

char *constMEM hexDigit MEMMODE = "0123456789ABCDEF";
char *constMEM hexNr[] MEMMODE = {"0", "x", hexDigit, hexDigit};
char buf1[] = "0x11";

MENU(mainMenu, "Main menu", doNothing, noEvent, wrapStyle,
     FIELD(test, "Test", "%", 0, 100, 10, 1, doNothing, noEvent, wrapStyle), SUBMENU(subMenu),
     OP("OpA", action1, anyEvent), OP("OpB", action2, enterEvent),
     SUBMENU(setLed),
     OP("LED On", myLedOn, enterEvent), OP("LED Off", myLedOff, enterEvent), SUBMENU(selMenu), SUBMENU(chooseMenu),
     OP("Alert test", doAlert, enterEvent), EDIT("Hex", buf1, hexNr, doNothing, noEvent, noStyle), EXIT("<Back")
);

#define MAX_DEPTH 2

MENU_OUTPUTS(out, MAX_DEPTH, SERIAL_OUT(Serial1), NONE);

NAVROOT(nav, mainMenu, MAX_DEPTH, in, out);//the navigation root object

result alert(menuOut &o, idleEvent e) {
    if (e == idling) {
        o.setCursor(0, 0);
        o.print("alert test");
        o.setCursor(0, 1);
        o.print("[select] to continue...");
    }
    return proceed;
}

result doAlert(eventMask e, prompt &item) {
    nav.idleOn(alert);
    return proceed;
}

result idle(menuOut &o, idleEvent e) {
    switch (e) {
        case idleStart:
            o.print("suspending menu!");
            break;
        case idling:
            o.print("suspended...");
            break;
        case idleEnd:
            o.print("resuming menu.");
            break;
    }
    return proceed;
}

void setup() {
    pinMode(encBtn, INPUT_PULLUP);
    pinMode(LED_BUILTIN, OUTPUT);

    clickEncoder.setButtonOnPinZeroEnabled(true);

    Serial.begin(115200);
    Serial1.begin(115200);
    while (!Serial);
    Serial.println("Arduino Menu Library");
    Serial.flush();
    nav.idleTask = idle;//point a function to be used when menu is suspended
    mainMenu[1].enabled = disabledStatus;
    nav.showTitle = false;
    ticker.attach(0.001, timerIsr);
    delay(2000);
}

void loop() {
    nav.poll();
    digitalWrite(LED_BUILTIN, ledCtrl);
    delay(100);//simulate a delay as if other tasks are running
}
```

neu-rah, 2018-03-26:
Hi!
the events list is here https://github.com/neu-rah/ArduinoMenu/blob/master/src/menuBase.h#L97

32 is binary 1<<5 = 0b00100000 = blurEvent (sent to an element that is about to loose focus)

encA and encB pins should be the encoder channels A and B

encoder button should be on pin encBtn

the second option is disabled, it was disable on setup by this line:
```c++
mainMenu[1].enabled = disabledStatus;
```

does it help?

sdurnev, 2018-03-26:
Hi, Rui!
My question was different.
The situation looks like this.
1. The cursor is on the first position
2. I press the button "encB"
3. The cursor jumps to the second position
4. I release the button "encB"
5. The cursor returns to the first position
Why does the cursor come back?

With the events, I understand everything, thank you!

neu-rah, 2018-03-26:
encB is not the button, it should be one of the encoder pins, ant it will have that effect when set on/off as if encoder was moving back and forth... please check you encoder pins

sdurnev, 2018-03-26:
Oh, I did not correctly understand the logical work.
I do not have an encoder.
I have 3 buttons - up, down, select.
For this I need - keyIn.h
Is this true?

neu-rah, 2018-03-26:
yes, keys are adequate then

sdurnev, 2018-03-26:
Thank you, Rui!
I will try....

sdurnev, 2018-03-26:
Thanks again!
Working...

closed 2018-03-26 by sdurnev