neu-rah/ArduinoMenu #171: ClickEncoder issues counter-clockwise

ibishop100, 2018-04-05:
Hi, I've been fighting with an issue with ClickEncoder for a while now.  I am able to get perfect results with the ESP8266 demo from the ClickEncoder repository.  Up, down, click, double click, etc all work fine.

When I wire it up with ArduinoMenu combining a few examples together (since I am using a different screen, and timer method), I get reliable button presses and 
clockwise events, but not counter-clockwise.

Code attached: [main.txt](https://github.com/neu-rah/ArduinoMenu/files/1878757/main.txt)

Any ideas?

ibishop100, 2018-04-05:
Was just switching my display driver from SSD1306Ascii to U8G2 and because I have a bug somewhere that's not working, but I found that the encoder works perfectly when the Serial output is the only output.

neu-rah, 2018-04-05:
checking... thanks for reporting it

neu-rah, 2018-04-06:
Hi!

i've used this code on esp12e + U8G2 + ssd8266 + clickEncoder.

console outputs encoder pin states
please note that i'm using GPIO pin numbers (https://arduining.com/2015/08/15/nodemcu-esp-12-development-kit-v1-0/)

a note about your encoder timer, please see esp8266 timer usage on my code

```c++
extern "C" {//esp8266 stuff
  #include "user_interface.h"
}

#include <Arduino.h>
#include <ESP8266WiFi.h>

#include <Wire.h>//oled+? bus
#include <U8g2lib.h>//oled driver
// #include <ClickEncoder.h>//from https://github.com/0xPIT/encoder

#define I2C_ADDRESS 0x3C
// #include "SSD1306Ascii.h"
// #include "SSD1306AsciiWire.h"
#include <menu.h>
// #include <menuIO/SSD1306AsciiOut.h>

#include <ClickEncoder.h>
#include <menuIO/clickEncoderIn.h>
#include <menuIO/keyIn.h>
#include <menuIO/chainStream.h>
#include <menuIO/u8g2Out.h>

using namespace Menu;

//monocrome colors
const colorDef<uint8_t> colors[]={
  {{0,0},{0,1,1}},//bgColor
  {{1,1},{1,0,0}},//fgColor
  {{1,1},{1,0,0}},//valColor
  {{1,1},{1,0,0}},//unitColor
  {{0,1},{0,0,1}},//cursorColor
  {{1,1},{1,0,0}},//titleColor
};
#define menuFont u8g2_font_7x13_mf
constexpr int fontX=7;
constexpr int fontY=16;

// SSD1306AsciiWire oled;
#define OLED_ROT U8G2_R2
constexpr int OLED_SDA=0;
constexpr int OLED_SDC=2;
U8G2_SSD1306_128X64_NONAME_F_HW_I2C oled(OLED_ROT, U8X8_PIN_NONE, OLED_SDC, OLED_SDA);

#define encA 5
#define encB 4
#define encBtn 10
#define encSteps 4

ClickEncoder clickEncoder = ClickEncoder(encA,encB,encBtn,encSteps);
ClickEncoderStream encStream(clickEncoder,1);

//----------------------------------------------------------

result doAlert(eventMask e, prompt &item);

result showEvent(eventMask e,navNode& nav,prompt& item) {
  Serial.print("event: ");
  Serial.println(e);
  return proceed;
[... 165 lines cut ...]
  //   clickEncoder.service();
  //
  // }

  //--------------------Menu:------------------------
  constexpr int menuFPS=1000/30;
  static unsigned long lastMenuFrame=-menuFPS;
  unsigned long now=millis();
  if (now-lastMenuFrame>=menuFPS) {
    lastMenuFrame=millis();
    // Serial.write('.');
    Serial.print(digitalRead(encA));
    Serial.print(digitalRead(encB));
    Serial.println(digitalRead(encBtn));
    nav.doInput();
    if (nav.changed(0)||(nav.sleepTask&&nav.idleChanged)) {//gfx device [0] changed?
      oled.setFont(menuFont);
      oled.firstPage();
      do nav.doOutput(); while(oled.nextPage());
    }
  }
}
```

please let me know the outcome, thanks

ibishop100, 2018-04-06:
Wow, thanks for your efforts!

Same issue, CW works perfectly, CCW is unreliable.  Running the menu with serial output shows the encoder is working with the menu code perfectly though.  Very very odd.

I switched to the encoder library you linked, to no difference.  I have been using the soligen2010 fork as that's what PlatformIO finds for me (and it works out of the box with ESP).

Perhaps it's a hardware issue with the NodeMCUs I'm using?  I seriously doubt they are genuine, perhaps they are causing the issue somehow?

I have some different ESP boards (D1 Mini, and something called a NodeMCU v3).  Will try them this weekend.  Too bad I don't have any different encoders, though they do seem ok when used without the LCD.

[This](https://instudio.mabangapp.com/img/201809/BBM6732/BBM6732g01.jpg) is what the encoder looks like.

Thanks again!

ibishop100, 2018-04-07:
Well, just tried it with a Wemos D1 Mini, same result.  I have to assume it's some kind of timing issue combined with the encoder I am using.  It's hard to put my finger on, but it doesn't seem consistent either, sometimes works better than others.  Possibly it works better if left alone for a minute?

What encoder are you testing with?  Does it have the pcb or is it just the 5 pin encoder?  I have a couple other encoders on their way, but it will be weeks before they show up.

I'll try the other ESP board I have just in case.

neu-rah, 2018-04-07:
have and encoder like yours on its way... my current is an old encoder that probably has some small capacitor and i usually add a 330 resistor on the common ground as a precaution. In fact i have a bad encoder and it flicks sometimes, but it flick on both directions.

there is an alternative encoder driver `menuIO/encoderIn.h`

yes, timing is very important for clickEncoder, be sure to use the esp8266 timer.

also check that none of the pins has a led on it and also that the pins are not used by internal hardware such as SPI_CS... well you trying a different board so i think its not the case.

a good test to put your finger on it would be not using the encoder for menu, but use the encoder test code mixed with the menu so that we can tell if there is something activated while running on the menu environment that affects the encoder. Because the same encoder on the same pins works ok while using the encoder test code.

hmm... i do really want to know the cause of this

thanks for your feedback

neu-rah, 2018-04-13:
any news on this?

ibishop100, 2018-04-13:
No, sorry.  Has been a very busy week.  Really good suggestions though.

I should have some time this weekend.

neu-rah, 2018-04-13:
no rush, i was just curious
take your time
and thanks for the quick response :D

ibishop100, 2018-05-05:
Hi, just a quick note.  I've been waiting for my new encoders to arrive and they did today.  Your code works fine with with the Keyes clone encoder I just received.  This encoder has just two 10k resistors, while the encoder I was having trouble with has 5 resistors and two caps.  I also have some bare encoders I will play with later.

I'm ok with using these new encoders, maybe if you want some of the 'bad' ones, I can send you a couple.  I might try desoldering the extraneous bits from one to make it like the keyes one...  Caps off, no effect.  I'll trace out the resistors later and pull off the ones that don't match.

neu-rah, 2018-05-05:
Thanks a lot, I do live on a very remote location but if you have a picture of them or a web reference...

living in Azores islands, Portugal :D
guess its cheaper to order a new one from china

ibishop100, 2018-05-05:
Maybe I should deliver them, looks like an amazing place!  They're not even available on Aliexpress anymore, so possibly a broken product, or something made for a different purpose.

[image]
[image]
[image]

Oh, here they are:  [encoders](http://www.icstation.com/rotary-encoder-module-brick-sensor-development-round-audio-rotating-potentiometer-knob-arduino-p-12273.html)

Still I wouldn't bother.  Worst case I'd just desolder the rotary encoder and use it without the pcb.  I have 10 other ones to use first though..  ;)

neu-rah, 2018-05-08:
it looks just like the ones that just arrived here, not checked the resistor values or connect them yet

neu-rah, 2018-05-20:
[image]
[image]

have this ones, done a quick check with clickEncoder and it worked pretty well, guess cliencoder is using internal pull-ups, but not checked to that detail. here's the link: https://www.ebay.com/itm/Rotary-Encoder-Module-Brick-Sensor-Development-Board-Potentiometer-5V/382178758228?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

closed 2018-05-20 by neu-rah