neu-rah/ArduinoMenu #172: Best way use input in function

meditant, 2018-04-13:
Hello,

I need to use input in a loop in a function to define timers action start/pause/reset...
what is the best way to use, the chainStream or something else ?

I start to write the program and i test the functionality of the library

The program is something like :

- I am in the menu
- I chose a entry and the library start a function
- In the function a have a loop until the "back" button is pressed
- If i press the "start" button a start to print on screen the time and the sensor information until the "stop" button is pressed
- I press the "back" button to return in the menu

Berst regards

neu-rah, 2018-04-13:
Hi!

the easiest way is as you described, you start a function and it will block until return. This is ok if you do not have other tasks on the main cycle that need attention. Please not that the menu itself will be also inoperative until your function returns (unless you call the menu system inside your function), but as you mention that you will be using the screen so i assume not.

There are other possibilities but if the blocking is ok with you, then its the easiest way of doing it.

about the inputs, chainStream is just an input wrapper that can host multiple input methods (ex: encoder and button) it allows you to use multiple inputs in `parallel`.

the input source should stay available for you, for example if you are using button you can still read the buttons state inside your function. You can also read from the menu input object if you wish to do so, but usually its preferable to read from the source (the buttons).

does that answer you question?
if not feel free to ask for details

thanks for using this lib.

neu-rah, 2018-04-13:
well because they give the information you need with less intermediaries, so less code, better performance.

meditant, 2018-04-14:
OK Thanks

Best regards and thank you for your library.

closed 2018-04-14 by meditant

neu-rah, 2018-04-14:
you are also welcome to join us at https://gitter.im/ArduinoMenu/Lobby?utm_source=share-link&utm_medium=link&utm_campaign=share-link