neu-rah/ArduinoMenu #173: <menu_id>_text was not declared in this scope

lucky62, 2018-04-29:
I got this error when trying to use menu library:
`'menuSec1_text' was not declared in this scope`

Menu declaration:

```
//Menu
TOGGLE(lcdBacklight,menuLcdBacklight,"Backlight: ",doNothing,noEvent,noStyle//,doExit,enterEvent,noStyle
  ,VALUE("On",HIGH,doNothing,noEvent)
  ,VALUE("Off",LOW,doNothing,noEvent)
);

MENU(menuSec1,"Sekcia1",doNothing,anyEvent,noStyle
  ,EXIT("Exit")
);

MENU(menuSec2,"Sekcia 2",doNothing,anyEvent,noStyle
  ,EXIT("Exit")
);

MENU(menuSec3,"Sekcia 3",doNothing,anyEvent,noStyle
  ,EXIT("Exit")
);

MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  ,SUBMENU(menuLcdBacklight)
  ,SUBMENU(menuSec1)
  ,SUBMENU(menuSec2)
  ,SUBMENU(menuSec3)
  ,EXIT("EXIT")
);
```

But first error in Arduino log is this:
```

"/home/lucky62/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os  -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR   "-I/home/lucky62/.arduino15/packages/arduino/hardware/avr/1.6.21/cores/arduino" "-I/home/lucky62/.arduino15/packages/arduino/hardware/avr/1.6.21/variants/eightanaloginputs" "-I/SHARED/SSD_DATA/00_MyProjects/ArduinoProjects/libraries/Newliquidcrystal_1.3.5" "-I/home/lucky62/.arduino15/packages/arduino/hardware/avr/1.6.21/libraries/EEPROM/src" "-I/SHARED/SSD_DATA/00_MyProjects/ArduinoProjects/libraries/ArduinoMenu_library/src" "-I/SHARED/SSD_DATA/00_MyProjects/ArduinoProjects/libraries/encoder-master" "-I/home/lucky62/.arduino15/packages/arduino/hardware/avr/1.6.21/libraries/Wire/src" "/tmp/arduino_build_20606/sketch/Zavlazovanie_1.ino.cpp" -o "/tmp/arduino_build_20606/sketch/Zavlazovanie_1.ino.cpp.o"
In file included from /SHARED/SSD_DATA/00_MyProjects/ArduinoProjects/libraries/ArduinoMenu_library/src/menuBase.h:28:0,
                 from /SHARED/SSD_DATA/00_MyProjects/ArduinoProjects/libraries/ArduinoMenu_library/src/menu.h:18,
                 from /SHARED/SSD_DATA/00_MyProjects/ArduinoProjects/Zavlazovanie_1/Zavlazovanie_1.ino:3:
/SHARED/SSD_DATA/00_MyProjects/ArduinoProjects/libraries/ArduinoMenu_library/src/macros.h:51:17: error: 'DECL_' does not name a type
 #define DECL(x) DECL_##x
                 ^

```

The complete errors log is attached.

lucky62, 2018-04-29:
Just I realized, that this error occurs when only EXIT item is in the menu.
When additional items were added, compiling is OK.

neu-rah, 2018-04-29:
correct, its a macro limitation, menus must have more than one element.
you can build without macros, but i guess your objective is not to build single option menus. so this was circumstantial.

closed 2018-04-29 by neu-rah